zephyrproject-rtos/zephyr #54424: Add btmon BT_DEBUG backend for bsim targets

aescolar, 2023-02-03:
**Is your enhancement proposal related to a problem? Please describe.**

When developing/debugging using bsim targets, it would be nice to be able to monitor the HCI traffic in btmon.

**Describe the solution you'd like**

Today there is a BT_DEBUG_MONITOR_UART backend for real targets but not for bsim targets.
Either getting that backend to work for a bsim target or adding one new backend for bsim targets would be nice.

**Describe alternatives you've considered**

Increasing the amount of logging in the BT_DEBUG_LOG backend, but that would be more tedious that using btmon.

**Additional context**
Requested by @Thalley

aescolar, 2023-02-03:
CC: @jhedberg @cvinayak @Vudentz @thoh-ot

PavelVPV, 2025-03-10:
I've managed to use `BT_DEBUG_MONITOR_UART` together with [UART driver for native simulator](https://docs.zephyrproject.org/latest/boards/native/native_sim/doc/index.html#ptty-uart) in BabbleSim tests. Here is how I did this

1. Add this to your prj.conf:

```
CONFIG_BT_DEBUG_MONITOR_UART=y
CONFIG_UART_CONSOLE=n
CONFIG_SERIAL=y
CONFIG_UART_NATIVE_POSIX=y
CONFIG_UART_NATIVE_WAIT_PTS_READY_ENABLE=y
```

2. Create an overlay file (e.g. bt_mon.overlay) and add this:

```
/delete-node/ &uart0;
/ {
       chosen {
               zephyr,bt-mon-uart = &uart0;
       };
       uart0: uart {
               status = "okay";
               compatible = "zephyr,native-posix-uart";
               /* Dummy current-speed entry to comply with serial
                * DTS binding
                */
               current-speed = <0>;
       };
};
```

3. Include this overlay through `-DEXTRA_DTC_OVERLAY_FILE` option passed to CMake. When using compile function from `sh_common.source` from `zephyr/tests/bsim`, add this before compile:

```
cmake_extra_args=-DEXTRA_DTC_OVERLAY_FILE=bt_mon.overlay
```

For example:

```
app=tests/bsim/bluetooth/host/l2cap/stress sysbuild=1 \
 cmake_extra_args=-DEXTRA_DTC_OVERLAY_FILE=bt_mon.overlay compile
```

If building with twister, add this to testcase.yaml to the test:

```
    extra_args:
      - EXTRA_DTC_OVERLAY_FILE="bt_mon.overlay"
```

5. Because `CONFIG_UART_NATIVE_WAIT_PTS_READY_ENABLE` is enabled, you can add `-wait-uart` option to hold test execution until terminal is attached:

```
Execute "${bsim_exe}" -v=${verbosity_level} -s=${simulation_id} -d=0 -testid=central -rs=43 \
    -wait_uart
```

And run your own terminal in a separate window.

6. Alternatively, you can pass your command to executable through `-attach_uart_cmd` option:

```
ATTACH_CMD="btmon --tty=%s -w hci_log_central.btsnoop &"
Execute "${bsim_exe}" -v=${verbosity_level} -s=${simulation_id} -d=0 -testid=central -rs=43 \
    -attach_uart_cmd="${ATTACH_CMD}"
```

Commit with the code: https://github.com/zephyrproject-rtos/zephyr/commit/8417de41c6cdb6af62bb6017c93c9769ed762db7

It is also possible to use `NRFX_UARTE` with BabbleSim HW modules, however, this (for unclear reasons) was affecting the simulation quite significantly. So if one only needs to dump HCI logs without affecting the simulation environment, use native simulator UART driver.

Also, rebase on top of https://github.com/zephyrproject-rtos/zephyr/pull/86638 PR is required for passing string arguments to a BabbleSim test.

Thalley, 2026-08-21:
I did try to have an LLM create a PR that will automatically output HCI traffic for BSIM tests: https://github.com/Thalley/zephyr/pull/11

It's far from perfect, but it does the job locally. It effectively defines a new BT_DEBUG_MONITOR_NATIVE and writes to a file, but I feel it's not reusing enough existing code, nor have I found a way to make it output the file in the right directory :D