zephyrproject-rtos/zephyr #54428: esp32 invalid flash dependencies

nordicjm, 2023-02-03:
**Describe the bug**
The following was seen in a MCUmgr test which enables flash for the `icev_wireless` board:
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /opt/toolchains/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/__w/zephyr/zephyr/twister-out/icev_wireless/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.crc32/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/subsys/fs/libsubsys__fs.a  zephyr/subsys/mgmt/mcumgr/mgmt/libsubsys__mgmt__mcumgr__mgmt.a  zephyr/subsys/mgmt/mcumgr/smp/libsubsys__mgmt__mcumgr__smp.a  zephyr/subsys/mgmt/mcumgr/util/libsubsys__mgmt__mcumgr__util.a  zephyr/subsys/mgmt/mcumgr/grp/fs_mgmt/libsubsys__mgmt__mcumgr__grp__fs_mgmt.a  zephyr/subsys/mgmt/mcumgr/transport/libsubsys__mgmt__mcumgr__transport.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/zcbor/libmodules__zcbor.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -L"/opt/toolchains/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0"  -L/__w/zephyr/zephyr/twister-out/icev_wireless/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.crc32/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -lgcc  -no-pie  -mabi=ilp32  -march=rv32ima_zicsr_zifencei  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,--fatal-warnings  -T/__w/zephyr/modules/hal/espressif/zephyr/esp32c3/src/linker/esp32c3.rom.alias.ld  -T/__w/zephyr/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.ld  -T/__w/zephyr/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld  -T/__w/zephyr/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld  -T/__w/zephyr/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld  -T/__w/zephyr/modules/hal/espressif/zephyr/esp32c3/../../components/soc/esp32c3/ld/esp32c3.peripherals.ld && cd /__w/zephyr/zephyr/twister-out/icev_wireless/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.crc32/zephyr && /usr/local/bin/cmake -E echo
/opt/toolchains/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(flash_mmap.c.obj): in function `spi_flash_mmap_pages':
/__w/zephyr/modules/hal/espressif/components/spi_flash/flash_mmap.c:185: undefined reference to `k_malloc'
/opt/toolchains/zephyr-sdk-0.15.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(flash_mmap.c.obj): in function `spi_flash_mmap':
/__w/zephyr/modules/hal/espressif/components/spi_flash/flash_mmap.c:151: undefined reference to `k_malloc'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

The Kconfig dependencies for this are therefore invalid

**Expected behavior**
Dependencies to be properly set

**Impact**
Showstopper, holding up and blocking completely unrelated PRs

nordicjm, 2023-02-03:
Oh, and on esp32_net, and on esp32c3_devkitm

sylvioalves, 2023-02-03:
@nordicjm which test case?

sylvioalves, 2023-02-03:
`samples/subsys/mgmt/mcumgr/smp_svr/` I guess..

sylvioalves, 2023-02-03:
Not sure this is the correct approach, but that will bring HEAP for all ESP32xx boards (enabling k_malloc). Another option would keep everything as is and only add the `boards/esp32x.conf` into MCUmgr sample code.

sylvioalves, 2023-02-03:
Anyway, CI is failing in this PR only (https://github.com/zephyrproject-rtos/zephyr/pull/54416), right?

nordicjm, 2023-02-03:
Yes, just tried it locally here with your patch and it builds fine

kartben, 2023-02-14:
> Yes, just tried it locally here with your patch and it builds fine

+1, I can confirm that this did the trick for me on esp32c3_devkitm, where spi_esp32_spim driver wouldn't work otherwise (undefined reference to k_calloc)

sylvioalves, 2023-02-14:
@nordicjm Perhaps this PR should be merged as bug..

PR #54434 (2023-02-14): boards: esp32: add default HEAP for all ESP32 boards
In order to avoid missing k_malloc options, add default HEAP value for all boards.

Fixes #54428
commit 839050b9 (2023-02-14): boards: esp32: add default HEAP for all ESP32 boards

In order to avoid missing k_malloc options, add default
HEAP value for all boards.

Fixes #54428
closed 2023-02-14 by stephanosio