zephyrproject-rtos/zephyr #54435: mec172xevb: sample.drivers.sample.drivers.peci failing

pkunieck, 2023-02-03:
**Describe the bug**
Only output from device is: 
Booting Zephyr OS build v3.3.0-rc1-62-g60a20471b561 (delayed boot 500ms)
PECI sample test
Err: PECI device is not ready

Test wasn't running before this commit: 2a8f718 and after that it only fails with no console output error massage. 

**To Reproduce**

twister -p mec172xevb_assy6906 --device-testing --device-serial /dev/ttyUSB0 -T samples/drivers/peci/ -vv

**Expected behavior**
Test pass

**Impact**
annoyance

**Logs and console output**
See above.

**Environment**
 - OS: Linux
 - Zephyr SDK
 - 2a8f718

pkunieck, 2023-05-18:
Can we have an update on this? What is a plan for that issue?

albertofloyd, 2023-06-05:
@pkunieck PECI samples example should not be run in MEC172x EVB, since it performs no operation. Only valid HW is MEC172xmodular. 

This because PECI bus sample requires either a controller and a target e.g. Intel system (PECI controller) + Embedded controller (PECI target) where EC HW can be add-on card (mec172xmodular) or mother board in said system. (no zephyr supported HW)