zephyrproject-rtos/zephyr #54444: samples/modules/chre/sample.modules.chre should not attempt to build on toolchains w/o newlib

galak, 2023-02-03:
when building with a toolchain like llvm, we attempt to build sample.modules.chre as it needs newlib which isn't provided as part of llvm.

jgl-meta, 2023-06-20:
@yperess this was marked as stale. If this is no longer an issue can you close it?

yperess, 2023-06-20:
> @yperess this was marked as stale. If this is no longer an issue can you close it?

Still on my roadmap, I'll probably get to it in Q3

tejlmand, 2023-10-30:
@yperess is this still an issue ?
Shouldn't this be working with picolibc ?

nashif, 2023-10-30:
right now we have: 

```
diff --git a/samples/modules/chre/sample.yaml b/samples/modules/chre/sample.yaml
index cd624b92d8..376839d35a 100644
--- a/samples/modules/chre/sample.yaml
+++ b/samples/modules/chre/sample.yaml
@@ -24,4 +24,4 @@ tests:
         - "I: Exiting EventLoop.*"
     integration_platforms:
       - native_posix
-    filter: not CONFIG_MINIMAL_LIBC
+    filter: CONFIG_FULL_LIBC_SUPPORTED

```,

so no dependency on newlib.

closed 2023-10-30 by nashif