zephyrproject-rtos/zephyr #54453: ESP32-C6: Add support for the ESP32-C6 and ESP32-C6-DevKitC

ajf58, 2023-02-04:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
_Describe in details the hardware support being requested and why this support benefits Zephyr._

From Espressif's website at https://www.espressif.com/en/products/socs/esp32-c6
> ESP32-C6 is Espressif’s first Wi-Fi 6 SoC integrating 2.4 GHz Wi-Fi 6, Bluetooth 5 (LE) and the 802.15.4 protocol.

As with other SoCs from Espressif, there are modules built around the SoC, as well as development kits that support the module.

This is a proposal to:

1. Add support for the ESP32-C6 SoC.
2. Add support for the ESP32-C6-DevKitC development kit as a new board in the Zephyr project.

**Describe why you are asking for this support?**
_Describe why you are asking for this support instead of contributing it directly to the tree_

I'm raising this issue to give visibility upstream of what I'm working on in my fork at https://github.com/ajf58/zephyr/tree/esp32c6 , so that others may contribute either directly and/or avoid duplication of effort.

_If this is a new board or SoC, please state whether you are willing to maintain the Zephyr support for it if it is included in the main tree_

Yes.

**Additional context**
_Add any other context or graphics (drag-and-drop an image) about the hardware here._

Press releases and general information can be found on Espressif's website at https://www.espressif.com/en/products/socs/esp32-c6

rftafas, 2023-02-06:
@ajf58 , is your plan to support the DevKitC board once C6 is supported or you plan to go all for C6?

ajf58, 2023-02-06:
@rftafas, I'm planning on supporting the DevKitC, and this in turn means adding support for the ESP32-C6.

rftafas, 2023-02-07:
@ajf58, are you also planing to use hal_espressif? It's not available for C6 yet.

ajf58, 2023-02-07:
@rftafas I am, yes. AIUI https://github.com/zephyrproject-rtos/hal_espressif is a fork of https://github.com/espressif/esp-idf with some Zephyr Project modifications. Whilst the former doesn't have support for the ESP32-C6, the latter does (in v5.1), so we can update hal_espressif as required.

rftafas, 2023-02-07:
@ajf58 tks for explaining. As we released documentation for ESP32-C6, I wanted to make sure you wouldn't go down to register level. As for hal_espressif and IDF, we have some ongoing internal work to make it cleaner.

If you find shortcomings, let us know.

ajf58, 2023-02-07:
> If you find shortcomings, let us know.

Thanks. In the C6's TRM there's a few sections that are in progress, e.g. "Low-power Management (RTC_CNTL)" is currently 11% complete. Do you know if there's a planned update to the TRM?

rftafas, 2023-02-08:
Unfortunately, I can't state an specific date. It will come, eventually.

siddharthgpta, 2023-04-12:
@ajf58 / @rftafas was there any progress made on this?

rftafas, 2023-04-12:
We are working on S3 and there is a few PRs to be timely submitted.  So, unfortunately due to that, no C6 yet. We have started planning for it, though.

rftafas, 2023-06-21:
FWIW, we started C6. Major hal_espressif improvements to come in order to support it. :)

Dids, 2023-08-25:
Could this issue be kept open until support is added/implemented, or is there another issue tracking this?

ajf58, 2023-08-30:
This became a lower priority activity for me, but I've got a more time to look at this now, and it seems that in ESP-IDF 5.1 there's support for the C6, so I'll take another look.

rftafas, 2023-08-30:
In that case, I don't suggest using plain ESP-IDF. We are moving away from it and have been using something that extracts only the HAL from IDF repository. Next hal_espressif will look a lot like this repository: https://github.com/espressif/esp-hal-3rdparty/tree/release/v5.1.c .

It probably will jump from release to release. We are doing internal tests and when all regressions are fixed, it will replace current hal_espressif. Only then C6 work will start.

Finally, a WARNING: end users still are not encouraged to use any form of hal (be it either the hal_espressi for esp-hal-3rdparty) for applications. These repos are intended for system design only, hal is not a public API and it will change without notice.
 
If there is real interest in engaging in pre-alfa, let me know and I'll see what I can do to help.