zephyrproject-rtos/zephyr #54468: Zephyr should support the USB MTP feature.

hvbon2010, 2023-02-06:
MTP (Media Transfer Protocol) is indeed a protocol designed for transferring media files between devices, and it does offer some advantages over MSC (Mass Storage Class).

Having MTP support in the Zephyr USB device class could be useful for those who want to transfer media files between devices. However, it is important to consider the specific requirements and use cases of the project before making a decision to implement this feature.

fwittenfeld, 2023-08-03:
MTP support would be great. 
It would enable us to native access our LittleFS flash from Windows Computers.

pnchatterji, 2023-11-07:
Hello, is there any progress on this issue? Is it in the works? I would greatly appreciate a ballpark figure of when it might be available. Reason is that we need this feature in a product, and we are debating if it makes sense to implement it at the application level, or wait for it to become a part of Zephyr. If it becomes a part of Zephyr in, say, 4 months, it would make sense for us to wait.

Finomnis, 2024-07-15:
@pnchatterji I don't think anybody is working on it, otherwise there would be something linked here. I would love this, but I don't think it's available right now.

Finomnis, 2024-09-13:
Is there a way to bump/bounty this?

ExtremeGTX, 2024-11-03:
I would like to share an update here about USB MTP Class, I have implemented the read path (Enumerate/Copy/Read) Files/folders , tested on Windows 10.

I'm working now to implement write path features, initially just paste a file to a device then may be `DELETE_OBJECT` command then i will open a PR and call for testers.

Initially i can test on Windows and Linux, but don't have a mac machine to test it.

Finomnis, 2024-11-04:
> I would like to share an update here about USB MTP Class, I have implemented the read path (Enumerate/Copy/Read) Files/folders , tested on Windows 10.
> 
> I'm working now to implement write path features, initially just paste a file to a device then may be `DELETE_OBJECT` command then i will open a PR and call for testers.
> 
> Initially i can test on Windows and Linux, but don't have a mac machine to test it.
> 
> My aim is to work mainly on file transfer, skipping all media related features like (Capture, media playback, photo import, etc.). I may share later which exact features are supported.

That is great news! How are the memory requirements for it?

ExtremeGTX, 2024-11-05:
> That is great news! How are the memory requirements for it?

didn't calculate it yet, but i'm aiming to have like a minimal version of MTP sufficient for file transfer, sometimes i face trade offs CPU Cycles vs RAM usage.

Anyway, could you please tell me what is your target storage (RAM/flash) specs  ? or how much space would be suitable for your project.

The goal here is to check if i can aim for a reasonable space goal during development.

Till now the most thing I need to optimize is handles management for files/folders.

I will post here the memory usage when i get back to my desk.

Finomnis, 2024-11-05:
For our specific project, we have about 50MB of storage and about 50KB of RAM left for usb related stuff. I know that's very tight, that's why I ask :D

Finomnis, 2024-11-05:
And yes, absolutely only the file transfer part is necessary for us. Existing internal littlefs to USB host. And as an alternative to FAT block device, which requires annoying synchronization between host and device and does not have the advantages of littlefs

ExtremeGTX, 2024-11-05:
Memory usage now is quite high I would say:
Flash usage: ~27KB
RAM usage: ~22KB

but since I'm still not in optimization phase.

Finomnis, 2024-11-06:
Any chance this will be backportable to 3.7? Or do you use new features?

Finomnis, 2024-11-15:
@ExtremeGTX Any news?

ExtremeGTX, 2024-11-16:
Was on a business trip.

I don't think i'm using anything special from post v3.7 features, it is mainly usb `device_next` and `net_buf`

ExtremeGTX, 2024-12-26:
Sharing progress of MTP

- MTP is only supportd in `device_next` stack.
- Supported commands:
  - [x] GET_DEVICE_INFO
  - [x] OPEN_SESSION
  - [x] CLOSE_SESSION
  - [x] GET_STORAGE_IDS
  - [x] GET_STORAGE_INFO
  - [x] GET_OBJECT_HANDLES
  - [x] GET_OBJECT_INFO
  - [x] GET_OBJECT
  - [x] DELETE_OBJECT
  - [x] SEND_OBJECT_INFO
  - [x] SEND_OBJECT
  - [ ] MOVE_OBJECT (WIP): Allow moving a file within the device without dependency on the host.
  - [ ] COPY_OBJECT (WIP): Allow copying a file within the device without dependency on the host.

- The implemented commands allows the following operations:
  - Storage(s) Enumeration
  - Read/Copy from device
  - Write to device
  - Delete files

MTP access is enabled by adding `mtp-enabled` property to a fstab node:
```c
fstab {
        compatible = "zephyr,fstab";
        lfs1: lfs1 {
                compatible = "zephyr,fstab,littlefs";
                ...
                mtp-enabled;
        };
}
```

For **memory constrained** devices, they can :
- Enable only USB Full speed which will reduce needed buffers size from 
(High-speed: 3x 512B) to (Full-speed: 3x 64B)
- Reduce Object Handles database size by allowing smaller number of objects to be accessed over MTP

#### Screenshot from Windows 10
[image]

Finomnis, 2024-12-27:
Any draft PR we can already look at?

kartben, 2024-12-30:
> Any draft PR we can already look at?

https://github.com/ExtremeGTX/zephyr/tree/feature/usb_mtp apparently :)