zephyrproject-rtos/zephyr #54498: net: openthread: echo server do not work in userspace

pawelosyp-hid, 2023-02-06:
**Describe the bug**
In echo server demo sample there is a openthread overlay. I have sucessfully tested that with nrf52840dk and border router setup on RPi. Based on that I have made my own app for tests.
I have tried to run it in userspace with no positive result. No matter if it is original sample or my own.
Is there something I am missing? Or current OT implementation in Zephyr do not support userspace?

All tests were performed on Zephyr 3.2 and Nordic NRF52840DK

**To Reproduce**
Steps to reproduce the behavior:
Build sample with OT overlay and additional configs:
CONFIG_USERSPACE=y
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=10
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=10
Try to run the sample

**Expected behavior**
The application should run in userspace.

rlubos, 2023-03-01:
@pawelosyp-hid I've identified some issues with userspace support in the sample (must've decayed over time as I'm pretty convinced it used to work). I've submitted fixes related to the sample itself in https://github.com/zephyrproject-rtos/zephyr/pull/55329.

**Please note** however, that I've identified one other issue that prevents the sample from running in userspace, related to logging (https://github.com/zephyrproject-rtos/zephyr/issues/55323). As a temporary workaround, you can simply disable `CONFIG_LOG_RUNTIME_FILTERING=n` in the project file to prevent a logger crash. I've successfully ran the sample with userspace enabled, after combining this workaround with the changes from my PR.

A second note - if you plan to use `echo_client` as well, it has a similar issue with `k_work` API usage as `echo_server` - fixing this however will require some more serious sample refactoring, hence will be submitted separately later.

pawelosyp-hid, 2023-03-07:
@rlubos Thank you for the responce. I have checked the solution and can confirm that it works both on Zephyrs and mine sample.

closed 2023-03-07 by pawelosyp-hid