zephyrproject-rtos/zephyr #54504: RDClient: Connection resume does not work after network error

chmielewskiandreas, 2023-02-06:
**Describe the bug**

Once the RDClient is in suspended state a socket error (in my example taking the modem offline) causes that the socket is broken after the RDClient is resumed.

Steps to reproduce the behaviour. Please take the example below (just replace lwm2m_client example with this version) and execute

1) client start_network
2) client start
3) client pause
4) client stop_network
5) client start_network
6) client resume

You will see that the state machine loops within ENGINE_UPDATE_SENT until a timeout occurs. A short debugging session tracked this down to socket_fault_cb method where lwm2m_socket_close is called. This method closes the socket and because of this lwm2m_close_socket will not be called later on and connection_suspended flag will not be set to true.

Reading the code of suspending and resuming the RDClient I see a big requirement for refactoring. In particular the place/way socket is closed should be refactored e.g. lwm2m_socket_close and lwm2m_close_socket is quite irritating. 

```
diff --git a/samples/nrf9160/lwm2m_client/prj.conf b/samples/nrf9160/lwm2m_client/prj.conf
index 7759741e9..a7b0266df 100644
--- a/samples/nrf9160/lwm2m_client/prj.conf
+++ b/samples/nrf9160/lwm2m_client/prj.conf
@@ -27,7 +27,7 @@ CONFIG_LWM2M_ENGINE_MAX_OBSERVER=15
 CONFIG_LWM2M_ENGINE_MAX_MESSAGES=15
 CONFIG_LWM2M_ENGINE_MAX_PENDING=15
 CONFIG_LWM2M_ENGINE_MAX_REPLIES=15
-CONFIG_LWM2M_ENGINE_DEFAULT_LIFETIME=30
+CONFIG_LWM2M_ENGINE_DEFAULT_LIFETIME=1200
 CONFIG_LWM2M_DNS_SUPPORT=y
 CONFIG_LWM2M_RW_JSON_SUPPORT=n
 CONFIG_LWM2M_SERVER_DEFAULT_PMIN=1
@@ -94,7 +94,7 @@ CONFIG_FOTA_DOWNLOAD=y
 CONFIG_MCUBOOT_IMAGE_VERSION="1.0.0"

 # Set LwM2M Server IP address here
-CONFIG_LWM2M_CLIENT_UTILS_SERVER="coaps://leshan.eclipseprojects.io:5684"
+CONFIG_LWM2M_CLIENT_UTILS_SERVER="coap://leshan.eclipseprojects.io:5683"

 # Application Event Manager
 CONFIG_APP_EVENT_MANAGER=y
@@ -102,3 +102,7 @@ CONFIG_APP_EVENT_MANAGER=y
 # Date-Time library
 CONFIG_DATE_TIME=y
 CONFIG_DATE_TIME_UPDATE_INTERVAL_SECONDS=86400
+
+CONFIG_LWM2M_VERSION_1_1=y
+CONFIG_LWM2M_RW_SENML_CBOR_SUPPORT=y
+CONFIG_ZCBOR_CANONICAL=y
```

```
/*
 * Copyright (c) 2019 Nordic Semiconductor ASA
 *
 * SPDX-License-Identifier: LicenseRef-Nordic-5-Clause
 */

#include <zephyr/kernel.h>
#include <ctype.h>
#include <zephyr/drivers/gpio.h>
#include <stdio.h>
#include <zephyr/net/lwm2m.h>
#include <modem/nrf_modem_lib.h>
#include <zephyr/settings/settings.h>

#include <net/lwm2m_client_utils.h>
#include <app_event_manager.h>
#include <net/lwm2m_client_utils_location.h>
#include <net/lwm2m_client_utils_location_events.h>
#include <date_time.h>
[... 357 lines cut ...]
[00:00:41.773,956] <inf> net_lwm2m_rd_client: ******* new state is ENGINE_UPDATE_REGISTRATION
[00:00:41.774,566] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:42.273,895] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_REGISTRATION
[00:00:42.273,895] <inf> net_lwm2m_rd_client: ***** update registration
[00:00:42.273,925] <err> net_lwm2m_engine: *******>>>>> lwm2m_engine_connection_resume 0
[00:00:42.274,627] <dbg> net_lwm2m_rd_client: sm_send_registration: registration sent [23.97.187.154]
[00:00:42.274,658] <inf> net_lwm2m_rd_client: ******* new state is ENGINE_UPDATE_SENT
[00:00:42.275,268] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:42.773,498] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:43.273,742] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:43.773,956] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:44.274,200] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:44.774,414] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:45.274,658] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:45.774,871] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:45.858,154] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:46.274,414] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:46.774,658] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:47.274,871] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:47.775,115] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:48.174,102] <err> net_lwm2m_message_handling: Failed to send packet, err 121
[00:00:48.275,329] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:48.775,573] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
[00:00:49.275,787] <inf> net_lwm2m_rd_client: ******* current state is ENGINE_UPDATE_SENT
```

SeppoTakalo, 2023-02-21:
The log message look like this is either quite old, or modified client.

I cannot anymore replicate the exact issue, but I found couple that could be causing this behaviour. Please see #55030 
If you still have the set up, can you try out if this fixes the exact issue that you are facing.

chmielewskiandreas, 2023-02-22:
Thanks @SeppoTakalo. This solved the issue.

PR #55030 (2023-02-22): Various fixes for LwM2M socket handling
## Clear out the timestamp when we start registration
      
There was a corner case that last registration timestamp was used
to detect if we can try UPDATE instead of full registration.

When timestamp was not cleared and DNS resolving failed, it might
cause engine to skip the resolving and continue retrying an UPDATE
message until timeout.

Fixes #54504

## Drop observations when falling back to registration
      
When we have establised an DTLS handshake and try LwM2M Update,
if the server rejects it, we fall back to sending full registration
but wh [... cut]
commit abb3163d (2023-02-22): net: lwm2m: Clear out the timestamp when we start registration

There was a corner case that last registration timestamp was used
to detect if we can try UPDATE instead of full registration.

When timestamp was not cleared and DNS resolving failed, it might
cause engine to skip the resolving and continue retrying an UPDATE
message until timeout.

Fixes #54504
closed 2023-02-22 by chmielewskiandreas