zephyrproject-rtos/zephyr #54506: net: ieee802154_6lo: wrong fragmentation of packets with specific payload sizes

niflostancu, 2023-02-06:
**TLDR**
I discovered some buggy behavior of the IEEE 802.15.4 6LoWPAN fragmentation for very specific L2 payload lengths (`~110` bytes).
The bug was also present in least in Zephyr v3.1, still unresolved in main.

**Long Description**
My application's client is reading several temperature sensors from a OneWire bus (which can have any number of sensors, so there's a wide range of payload sizes, but usually >= 200 bytes), which gets packetized and sent over a IEEE 802.15.4 radio as UDP, using link-local IPv6 addresses (with IPHC compression enabled).
So I noticed an issue on very specific (rare) configurations (with low sensor count, ~80 bytes of UDP data) that resulted in no data being received by the server (which I attributed this to radio signal losses at that time and ignored).
When it happened again, I figured out it happens regardless of the wireless distance and it was, indeed, an issue with the protocol (either at L7 - i.e. my fault - or L3/L2... my workaround at the time was: just add another sensor => it works due to different payload size => profit! :D).
Finally, I decided to investigate this issue and figured out it was due to a Zephyr L2/L3 (6LoWPAN) fragmentation bug (debug output contains `DROP: checksum mismatch`, see capture below).

BTW, I'm using the TI CC1352P Sub-GHz radio (though the upper 6Lo layer is responsible for the fragmentation of the frame payload).

**To Reproduce**

Here's my experiment (using 2 x TI CC1352P LaunchXLs with SubGHz radio at 868MHz):

- I wrote a `AF_CAPTURE`-based sniffer which prints a hexdump of the entire L2 frame (log provided below);
- I used a sample C program for testing out which payload sizes trigger the bug:
```c
const uint8_t sample_packet[] = {
	0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08,
	0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18,
	0x21, 0x22, 0x23, 0x24, 0x25, 0x26, 0x27, 0x28,
	0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x37, 0x38,
	0x41, 0x42, 0x43, 0x44, 0x45, 0x46, 0x47, 0x48,
	0x51, 0x52, 0x53, 0x54, 0x55, 0x56, 0x57, 0x58,
	0x61, 0x62, 0x63, 0x64, 0x65, 0x66, 0x67, 0x68,
	0x71, 0x72, 0x73, 0x74, 0x75, 0x76, 0x77, 0x78,
	0x81, 0x82, 0x83, 0x84, 0x85, 0x86, 0x87, 0x88,
	0x91, 0x92, 0x93, 0x94, 0x95, 0x96, 0x97, 0x98,
	0xA1, 0xA2, 0xA3, 0xA4, 0xA5, 0xA6, 0xA7, 0xA8,
	0xB1, 0xB2, 0xB3, 0xB4, 0xB5, 0xB6, 0xB7, 0xB8,
	// 0xC1, 0xC2, 0xC3, 0xC4, 0xC5, 0xC6, 0xC7, 0xC8,
	// 0xD1, 0xD2, 0xD3, 0xD4, 0xD5, 0xD6, 0xD7, 0xD8,
	// 0xE1, 0xE2, 0xE3, 0xE4, 0xE5, 0xE6, 0xE7, 0xE8,
	// 0xF1, 0xF2, 0xF3, 0xF4, 0xF5, 0xF6, 0xF7, 0xF8,
};
/* ... */
sock = socket(AF_INET6, SOCK_DGRAM, IPPROTO_UDP);
/* addr is link-local IPv6 (FE80::) the of the server, UDP port 331 */
connect(sock, (struct sockaddr *)&addr, sizeof(addr));
send(sock, sample_packet, sizeof(sample_packet), 0);
```
- Check server (sniffer) output for success (packet received by L7) / failure (`DROP: checksum mismatch`).

Here's the 4 experimental cases + results:

1. UDP payload len: 80 bytes => OKAY
2. UDP payload len: 81 bytes => BUGGY (wrong fragmentation)
3. UDP payload len: 95 bytes => BUGGY (wrong fragmentation)
4. UDP payload len: 96 bytes => OKAY

As you can see, UDP payload lengths from 81 bytes 'till 95 are wrongly fragmented.

**Expected behavior**

IEEE 802.15.4 6LoWPAN fragmentation should work on all payload sizes.

**Logs and console output**

You can See the [full log output from the receiver / sniffer here (gist)](https://gist.github.com/niflostancu/5c1d0ee9d9534cb366f41553fa599347).

As we can see, the packet is wrongly fragmented in two frames:

```log
first frame:
	41 dc 0e cd ab 00 12 4b  00 1c a6 14 72 00 12 4b |A......K ....r..K
	00 1c a1 ac e5 7e 33 f0  c2 6b 01 4b ed de 01 02 |.....~3. .k.K....
	03 04 05 06 07 08 11 12  13 14 15 16 17 18 21 22 |........ ......!"
	23 24 25 26 27 28 31 32  33 34 35 36 37 38 41 42 |#$%&'(12 345678AB
	43 44 45 46 47 48 51 52  53 54 55 56 57 58 61 62 |CDEFGHQR STUVWXab
	63 64 65 66 67 68 71 72  73 74 75 76 77 78 81 82 |cdefghqr stuvwx..
	83 84 85 86 87 88 91 92  93 94 95 96 97 98       |........ ......
second frame:
	41 dc 0f cd ab 00 12 4b  00 1c a6 14 72 00 12 4b |A......K ....r..K
	00 1c a1 ac e5 a1                                |......
```

_**Analysis:**_ first frame contains my synthetic payload values ranging from `0x01` to `0x98` (note: least significant nibble is `0x1 - 0x8`), while the second frame (wrongly) encapsulates the rest (`0xA1` onwards).
The RFC 4944 dispatch type is `0x7E` for the first frame, and missing altogether from the second one.
For correctly fragmented packets (scenario 4), the dispatch is `0xC0` and `0xE0` (second fragment).

I didn't manage to find the root cause of the bug (I am not familiar with Zephyr's 6lo implementation, but I'll take a look when I have some spare time). The unit tests should also be updated to catch this corner case.

**Environment:**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK 0.15.1
 - Commit SHA: ab764aada99a420ef7fe75ec6ea8984391fae2a5

**Note**: I also did a brief check with an earlier Zephyr release (v3.1) => it was buggy, too! So probably not a regression.

Thanks!

rlubos, 2023-02-16:
@niflostancu Please give https://github.com/zephyrproject-rtos/zephyr/pull/54902 a try.

I must admit this was a tricky one. It turned out that it's not a bug with 15.4 fragmentation or IPHC algorithm per se, but rather how the data was placed within a net_buf after header compression.

I've provided detailed description in the PR, but long story short, if the orignal IPv6 packet required fragmentation, but it was no longer the case after header compression, the 15.4 packet ended up scattered across two net_bufs, which is not expected by drivers. This only applied for a short range of packet sizes, as when the packet still required fragmentation after header compression, the fragmentation logic would kick in and ensure proper packet format.

niflostancu, 2023-02-16:
Hi,
Just tested it (using my TI CC1352 LaunchXLs) and and it's fixed now!

Here's the packet now:
```
first fragment:
			41 dc 1a cd ab 00 12 4b  00 1c a6 14 72 00 12 4b |A......K ....r..K
			00 1c a1 ac e5 c0 90 00  07 7e 33 f0 ea d6 01 4b |........ .~3....K
			00 e9 01 02 03 04 05 06  07 08 11 12 13 14 15 16 |........ ........
			17 18 21 22 23 24 25 26  27 28 31 32 33 34 35 36 |..!"#$%& '(123456
			37 38 41 42 43 44 45 46  47 48 51 52 53 54 55 56 |78ABCDEF GHQRSTUV
			57 58                                            |WX
second fragment:
			41 dc 1d cd ab 00 12 4b  00 1c a6 14 72 00 12 4b |A......K ....r..K
			00 1c a1 ac e5 e0 90 00  08 0c 61 62 63 64 65 66 |........ ..abcdef
			67 68 71 72 73 74 75 76  77 78 81 82 83 84 85 86 |ghqrstuv wx......
			87 88 91 92 93 94 95 96  97 98 a1 a2 a3 a4 a5 a6 |........ ........
			a7 a8 b1 b2 b3 b4 b5 b6  b7 b8                   |........ ..

```
The frame gets correctly fragmented now.

Should I close the issue, or wait for the PR to get merged?

Thanks!

rlubos, 2023-02-16:
> The frame gets correctly fragmented now.

Great!

> Should I close the issue, or wait for the PR to get merged?

No need to, GH will close the issue automaticaly once PR gets merged.

closed 2023-02-20 by nashif