zephyrproject-rtos/zephyr #54507: CONFIG_PM=y results to hard fault system for STM32

LaVoute974, 2023-02-06:
Hello, 

I have a big issue with power management and STM32. I'm working with a STM32L083VZT6. 
I first started some tests about power management in May 2022 and everything was fine. 
But now, even the power management blinky doesn't work... I also have an another project which use callback on GPIO and UART and when I add `CONFIG_PM=y`, it stops working and I have a hard fault system.

I'm using the HSI as the clock : 
`&clk_hsi {`
	`status = "okay";`
`};`

I have declared my stop state for the CPU : 
`power-states {`
		`stop0: state0 {`
			`compatible = "zephyr,power-state";`
			`power-state-name = "suspend-to-idle";`
			`min-residency-us = <2000>;`
			`exit-latency-us = <750>;`
		`};`
	`};`
And I choose stop0 for cpu0 : 
`cpus {`
		`cpu@0 {`
			`cpu-power-states = <&stop0>;`
		`};`
	`};`

erwango, 2023-02-06:
Updated the title as the issue is not as systematic as initially described
(Issue may not be related to this exact SoC, but main point is that this is not all STM32 with PM)

erwango, 2023-02-07:
@LaVoute974 Can you try https://github.com/zephyrproject-rtos/zephyr/pull/54430 ?

LaVoute974, 2023-02-07:
Try, doesn't solve the issue...

LaVoute974, 2023-02-08:
We solved our issue, we were using HAL Librarys, I think it's not compatible with low power or so... I don't really know why. Thanks anyway.

closed 2023-02-08 by LaVoute974