zephyrproject-rtos/zephyr #54522: Can we embrace GNU Build IDs?

attie-argentum, 2023-02-07:
`gcc` (and likely other toolchains) will produce a "_Build ID_", and place it in the `.note.gnu.build-id` section of ELF files. In the firmware world (i.e: not-ELF), we'll typically lose this, but care could be taken to keep it present and accessible.

I've implemented functional out-of-tree access to this, but I was wondering if there's any particular reason why it's explicitly disabled...

https://github.com/zephyrproject-rtos/zephyr/blob/461a13c085de73aabdb5777247a96675472a73ad/cmake/linker/ld/target_base.cmake#L20

I traced this a long way back to a0b54665, and there's no particular commentary on why it was disabled. (the previous default would have been the `sha1` style, which would be ideal).

**Describe the solution you'd like**
I propose re-enabling this feature for supported toolchains, and making the [`Elf_External_Note` structure](https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=include/elf/external.h;h=8f48684e4162fef852162acabe4295427e6432d0;hb=HEAD#l179) available to the application - or at least the build ID as binary / ascii-hex.

As mentioned above, we'll need to ensure that the symbol is at least offered for inclusion in any `*.bin` / `*.hex` files - it can be safely discarded if it's unused.

The value could also be exposed via a configuration value, to allow MD5 or a custom build ID too.

**Additional context**
The `sha1` style consists of a hash over all interesting parts of the binary, giving confidence about what exactly is in an image / device... Being able to report this to the console and then match the running firmware up with a build artifact would be incredibly useful during debug, and using such a strong built-in identifier is preferable by far over manually including another.

pdgendt, 2023-02-07:
@attie-argentum FYI #51532

nordicjm, 2023-02-07:
The reason is because this is a GNUism - things we want to prevent, avoid and remove. See https://github.com/zephyrproject-rtos/zephyr/issues/52226 for further details.

If there is a globally supported way of doing this for _all_ compilers, not only GCC compilers, then it can be considered a good candidate for inclusion.

attie-argentum, 2023-02-07:
@pdgendt - thanks for linking, apologies for missing that one!

@nordicjm - Fair enough, I can appreciate that... Perhaps I'll try to put some thought into an equivalent but generic option if I get a chance.

closed 2023-02-07 by attie-argentum