zephyrproject-rtos/zephyr #54528: log_switch_format testcase failing on intel_adsp_ace15_mtpm

aasthagr, 2023-02-07:
**Describe the bug**
This testcase fails using xtensa toolchain on intel_adsp_ace15_mtpm platform.

**To Reproduce**
Steps to reproduce the behavior:
west build -p -b intel_adsp_ace15_mtpm tests/subsys/logging/log_switch_format/

**Expected behavior**
Testcase needs to pass

**Impact**
Failing testcase

**Logs and console output**
```
In file included from zephyr/lib/libc/newlib/libc-hooks.c:12:
zephyr/include/zephyr/posix/sys/stat.h:40:10: fatal error: 'sys/_timespec.h' file not found
#include <sys/_timespec.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain: RI-2021.7
 - Commit SHA: 461a13c085de73aabdb5777247a96675472a73ad

laurenmurphyx64, 2023-02-07:
@aasthagr Same error observed for logging.log_syst.mipi_syst, so I added it to this bug instead of making you file another one

closed 2023-02-07 by carlescufi

aasthagr, 2023-02-07:
This bug isn't resolved by the linked PR.

PR #54523 (2023-02-07): picolibc: picolibc still not fully supported with many toolchains variants
picolibc: picolibc support still not complete with many toolchains

Clang support still work in progress in zephyr. So for now enable only on gcc based toolchains.

We should enable this back for more toolchains once we have full clang support and are able to verify.

Fixes #54528