zephyrproject-rtos/zephyr #54533: tests/drivers/can/timing fails on nucleo_f746zg

FRASTM, 2023-02-07:
**Describe the bug**
in Tests results https://github.com/zephyrproject-rtos/test_results/pull/1571, the testcase
tests/drivers/can/timing/ is failing on nucleo_f746zg.

Although only Nucleo STM32F746ZG is tested on the test suite, this is an issue that impact all STM32 boards.
Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f746zg tests/drivers/can/timing/
2. west flash3. make
3. See error

**Expected behavior**
test PASSED

**Impact**
CI failure 

**Logs and console output**
```
START - test_timing
testing bitrate 20000, sample point 87.5% (valid):  ret sp_err_min =b4 prop_seg = 0, phase_seg1 = 12, phase_seg2 = 2, prescaler = 180 OK, sample %
testing bitrate 50000, sample point 87.5% (valid):  ret sp_err_min =87 prop_seg = 0, phase_seg1 = 6, phase_seg2 = 1, prescaler = 135 OK, sample p%
testing bitrate 125000, sample point 87.5% (valid):  ret sp_err_min =1b prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 27 OK, sample %
testing bitrate 250000, sample point 87.5% (valid):  ret sp_err_min =1b prop_seg = 0, phase_seg1 = 6, phase_seg2 = 1, prescaler = 27 OK, sample p%
testing bitrate 500000, sample point 87.5% (valid):  ret sp_err_min =6 prop_seg = 0, phase_seg1 = 14, phase_seg2 = 3, prescaler = 6 OK, sample po%
testing bitrate 800000, sample point 80.0% (valid):  ret sp_err_min =0 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:197: test_timing_values: (sp_err >= 0 is false)                    
unknown error -134                                                                                                                                
 FAIL - test_timing in 0.090 seconds           
```                                                                                                   

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain :  Zephyr SDK 0.15.0
 - Commit SHA : 

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2023-02-07:
With 55d0ffa874e503097e47a6c6cb86fe7cda0e9b0d,  the tests/drivers/can/timing/ is configuring the CONFIG_CAN_FD_MODE=y  in its prj.conf
The nucleo_f746zg can node is defined as a compatible = "st,stm32-can" 
--> should the  testcase.yaml  for this tests/drivers/can/timing/  select **fdcan** only with    ` depends_on: fdcan`  or so ?

henrikbrixandersen, 2023-02-07:
@FRASTM This does not look like it is related to 55d0ffa874e503097e47a6c6cb86fe7cda0e9b0d. The test output, you provide, fails at setting a bitrate of 800kbit/s. A classical CAN controller should support that, depending on its CAN clock. Please provide the full test output, as it includes the CAN clock rate for further debugging.

The test should succeed no matter if `CONFIG_CAN_FD_MODE` is enabled or not, as the test checks the CAN capabilities before running any CAN-FD data phase timing tests.

FRASTM, 2023-02-07:
That could be a question of sysclk of the nucleo_f746zg board. 
When sysclk was 72MHz, the test PASSED with sysclock = 216MHz, (can bus clock = 54MHz) the test FAILED : 

```
*** Booting Zephyr OS build v3.3.0-rc1-89-gfcd60cf789b1 ***
Running TESTSUITE can_timing
===================================================================
testing on device can@40006400 @ 54000000 Hz
START - test_set_timing_data_max
 SKIP - test_set_timing_data_max in 0.001 seconds
===================================================================
START - test_set_timing_data_min
 SKIP - test_set_timing_data_min in 0.001 seconds
===================================================================
START - test_set_timing_max
 PASS - test_set_timing_max in 0.001 seconds
===================================================================
START - test_set_timing_min
 PASS - test_set_timing_min in 0.001 seconds
===================================================================
START - test_timing
testing bitrate 20000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 1%
testing bitrate 50000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 6%
testing bitrate 125000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = %
testing bitrate 250000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = %
testing bitrate 500000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = %
testing bitrate 800000, sample point 80.0% (valid): 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:)
unknown error -134
 FAIL - test_timing in 0.080 seconds
===================================================================
START - test_timing_data
 SKIP - test_timing_data in 0.001 seconds
===================================================================
TESTSUITE can_timing failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  66.67% [can_timing]: pass = 2, fail = 1, skip = 3, total = 6 duras
 - SKIP - [can_timing.test_set_timing_data_max] duration = 0.001 seconds
 - SKIP - [can_timing.test_set_timing_data_min] duration = 0.001 seconds
 - PASS - [can_timing.test_set_timing_max] duration = 0.001 seconds
 - PASS - [can_timing.test_set_timing_min] duration = 0.001 seconds
 - FAIL - [can_timing.test_timing] duration = 0.080 seconds
 - SKIP - [can_timing.test_timing_data] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

FRASTM, 2023-02-07:
When reducing the can bus clock to the lowest (apb1-prescaler = <16>;) the error becomes:
```
START - test_timing                                                             
testing bitrate 20000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 1%
testing bitrate 50000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 1%
testing bitrate 125000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = %
testing bitrate 250000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = %
testing bitrate 500000, sample point 87.5% (valid):                             
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:)
sample point error 98 too large                                                 
 FAIL - test_timing in 0.070 seconds           
```

FRASTM, 2023-02-07:
Note that commit  9151af224a6c06d39b46780ab3f8010b988fa249  changed the clock config for the nucleo_f746zg board and reverting this commit makes the test PASSED:
```
Running TESTSUITE can_timing
===================================================================
testing on device can@40006400 @ 36000000 Hz
START - test_set_timing_data_max
 SKIP - test_set_timing_data_max in 0.001 seconds
===================================================================
START - test_set_timing_data_min
 SKIP - test_set_timing_data_min in 0.001 seconds
===================================================================
START - test_set_timing_max
 PASS - test_set_timing_max in 0.001 seconds
===================================================================
START - test_set_timing_min
 PASS - test_set_timing_min in 0.001 seconds
===================================================================
START - test_timing
testing bitrate 20000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 6, phase_seg2 = 1%
testing bitrate 50000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 = %
testing bitrate 125000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 =%
testing bitrate 250000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 =%
testing bitrate 500000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 6, phase_seg2 = %
testing bitrate 800000, sample point 80.0% (valid): prop_seg = 0, phase_seg1 = 11, phase_seg2 =%
testing bitrate 1000000, sample point 75.0% (valid): prop_seg = 0, phase_seg1 = 8, phase_seg2 =%
testing bitrate 125000, sample point 90.0% (valid): prop_seg = 0, phase_seg1 = 15, phase_seg2 =%
testing bitrate 125000, sample point 80.0% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 =%
testing bitrate 125000, sample point 100.0% (invalid): OK                                       
testing bitrate 8000001, sample point 75.0% (invalid): OK                                       
 PASS - test_timing in 0.122 seconds                                                            
===================================================================                             
START - test_timing_data                                                                        
 SKIP - test_timing_data in 0.001 seconds                                                       
===================================================================                             
TESTSUITE can_timing succeeded                                                                  
                                                                                                
------ TESTSUITE SUMMARY START ------                                                           
                                                                                                
SUITE PASS - 100.00% [can_timing]: pass = 3, fail = 0, skip = 3, total = 6 duration = 0.127 secs
 - SKIP - [can_timing.test_set_timing_data_max] duration = 0.001 seconds                        
 - SKIP - [can_timing.test_set_timing_data_min] duration = 0.001 seconds                        
 - PASS - [can_timing.test_set_timing_max] duration = 0.001 seconds                             
 - PASS - [can_timing.test_set_timing_min] duration = 0.001 seconds                             
 - PASS - [can_timing.test_timing] duration = 0.122 seconds                                     
 - SKIP - [can_timing.test_timing_data] duration = 0.001 seconds                                
                                                                                                
------ TESTSUITE SUMMARY END ------                                                             
                                                                                                
===================================================================                             
PROJECT EXECUTION SUCCESSFUL
```

henrikbrixandersen, 2023-02-07:
> When sysclk was 72MHz, the test PASSED with sysclock = 216MHz, (can bus clock = 54MHz) the test FAILED :

Can you increase the CAN clock divider with a factor of three as well?

FRASTM, 2023-02-07:
No, only /1, /2, /4, /4, /16 are possible values for dividing  the sysclock and  get the PCLK1  (CAN bus APB1  clock) .
And the PCLK1 (CAN bus APB1 clock)  is limited to 54MHz Max.

FRASTM, 2023-02-08:
During the test_timing sequence of tests/drivers/can/timing/
The can_calc_timing_int does not find a combination for the bitrate 800000 at this freq of CAN clock = 54MHz
and then z_impl_can_calc_timing gives sample point at 0xFFFF.
This corresponds to the -ENOTSUP error code in the test_timing_values of tests/drivers/can/timing/main.c

FRASTM, 2023-02-08:
Might the tests/drivers/can/timing/main.c  simply accept that some combinations are not valid instead of reporting an error.

 In the zephyr documentation
> int can_calc_timing(const struct device *dev, struct can_timing *res, uint32_t bitrate, uint16_t sample_pnt)
> Calculate timing parameters from bitrate and sample point.
> 
> Calculate the timing parameters from a given bitrate in bits/s and the sampling point in permill (1/1000)
> of the entire bit time. The bitrate must always match perfectly. If no result can be reached for the given parameters, -EINVAL is returned.

FRASTM, 2023-02-08:
BTW I'd propose to  change the return code of the z_impl_can_calc_timing to fit with [CAN API  spec](https://docs.zephyrproject.org/latest/hardware/peripherals/canbus/controller.html#can-controller-api-reference):
> -EIO - if [can_get_core_clock()] is not available.

```
diff --git a/drivers/can/can_common.c b/drivers/can/can_common.c
index 4eafb0f1cc..0b81bcdd9e 100644
--- a/drivers/can/can_common.c
+++ b/drivers/can/can_common.c
@@ -180,7 +180,7 @@ int z_impl_can_calc_timing(const struct device *dev, struct can_timing *res,
 
 	ret = can_get_core_clock(dev, &core_clock);
 	if (ret != 0) {
-		return ret;
+		return -EIO;
 	}
 
 	return can_calc_timing_int(core_clock, res, min, max, bitrate, sample_pnt);
```

closed 2023-02-10 by stephanosio