zephyrproject-rtos/zephyr #54534: PSoC6/Cat1 add binary blob for Cortex-M0+ core

npal-cy, 2023-02-07:
## Origin

I would like to add binary blob (CM0P_SLEEP) for Cortex-M0+, which are using in PSoC 6 /Cat1 (cy8cproto_062_4343w).

The background for using pre-built image is documented in RFC (https://github.com/zephyrproject-rtos/zephyr/issues/42883)
> In the PSoC 6-based integration, the family's Cortex-M0+ will be controlled by a pre-built image, rather than being open to the user. This aligns with our current PSoC 6 strategies. 

## Type

- [ ] Precompiled library
- [v] Firmware image

## Module

The Zephyr module that this blob(s) will be referenced from:
boards::arm::cy8cproto_062_4343w for now.

## Purpose

Prebuilt application images are executed on the Cortex M0+ core of the CAT1 dual-core MCU. 

Note: Each application image has a variant based on the hardware die (e.g. psoc6_01, psoc6_02, psoc6_03, ...) 

**CAT1A Images**
[COMPONENT_CM0P_SLEEP](https://github.com/Infineon/cat1cm0p/blob/master/COMPONENT_CM0P_SLEEP/README.md)

This image starts CM4 core  and puts CM0+ core into a deep sleep mode.

[COMPONENT_CM0P_CRYPTO](https://github.com/Infineon/cat1cm0p/blob/master/COMPONENT_CM0P_CRYPTO/README.md)

This image starts crypto server on CM0+ core, starts CM4 core and puts CM0+ core into a deep sleep mode.

[COMPONENT_CM0P_BLESS](https://github.com/Infineon/cat1cm0p/blob/master/COMPONENT_CM0P_BLESS/README.md)

This image starts BLE controller on CM0+ core, starts CM4 core and puts CM0+ core into a deep sleep mode.

[COMPONENT_CM0P_SECURE](https://github.com/Infineon/cat1cm0p/blob/master/COMPONENT_CM0P_SECURE/README.md)

This image starts CM4 core at address corresponding to Secure Boot policy, sets required security settings, initializes and executes code of Protected Register Access driver, puts CM0+ core into a deep sleep mode.

**In scope of this ticket, we add support only for CM0P_SLEEP image.**

## Pull Request

1. https://github.com/zephyrproject-rtos/zephyr/pull/44211 (commit https://github.com/zephyrproject-rtos/zephyr/pull/44211/commits/eb665f5c71a93c28dd78cfc6ca1420a294d936b1)

2. https://github.com/zephyrproject-rtos/hal_infineon/pull/6/ - Update hal_infineon,  added blob support for cm0+ images.

## License

```
Copyright 2020-2022, Cypress Semiconductor Corporation (an Infineon company)
SPDX-License-Identifier: Apache-2.0

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

    http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.
```

cc: @ifyall

sjg20, 2023-02-08:
(adding here since the discussion is supposed to be here)
IMO Zephyr should be the repository of its code. It is one thing to have a module for a library, but this is core SoC support. Worse, it is a binary blob!

This should be implemented in C within Zephyr. The DT files can be generated but then should be checked into Zephyr.

Zephyr already supports Cortex-M0. I heard mention of a problem with the size...how much bigger is it with Zephyr than without?

ifyall, 2023-02-08:
The pre-compiled Cortex-M0+ images accomplish two goals:
- Providing the default image for the CM0+ for PSoC 6 variants where the CM0+ is not open as an application processor.
- Providing an image for the CM0+ when the CM0+ is open as an application processor but the user doesn't want to use it. This might be because they are evaluating the solution and don't need the complexity of writing code for both cores or their final system doesn't require developing their own code for the Cortex-M0+. In scope of the current integration, the Cortex-M0+ will not be open. We will be revisiting this in the future as we complete more work and bring additional devices to Zephyr.

In either of these cases, the code for the CM0+ must be as small as possible. In the current solution, 8 kb of flash have been allocated to the CM0+ sleep image. If we compare that size to the size of the simple Zephyr blinky sample on PSoC 6 Cortex-M4, that comes to over 33 kb of flash.

Yes, Zephyr supports CM0+, but in this case Zephyr is either significant overkill for the goal (e.g. a bare-metal main() that sets one register and then calls a deep sleep routine in an infinite loop) and/or incompatible with the license of the device (e.g. a CM0+ that is not open for application code).

nordicjm, 2023-02-09:
> In either of these cases, the code for the CM0+ must be as small as possible. In the current solution, 8 kb of flash have been allocated to the CM0+ sleep image. If we compare that size to the size of the simple Zephyr blinky sample on PSoC 6 Cortex-M4, that comes to over 33 kb of flash.

33KB? Sounds like something very wrong with your HAL or libraries, seeing 19KB here for that on an M4 and 16KB for that on an M0.

This can be fixed with sysbuild anyway, espressif have a similar requirement for a bootloader being required - see discussion here https://discord.com/channels/720317445772017664/733037524498382881/1072154327977840660 and that can be done using sysbuild. In fact, sysbuild can be used to build, include or flash an image that is not generated in zephyr, so you could add a sysbuild target here which builds the firmware for the M0+ core from scratch without using the zephyr libraries (though same compiler and host tools I would assume) and then have it flashed when someone runs their application.

ifyall, 2023-02-09:
@nordicjm,

Thanks for the pointers to the sysbuild work. We will take a look at that when we bring in the full multi-core support for the devices.

Regarding the flash size, it turns out I gave numbers for the button sample, not blinky. However, your point stands. We will audit our implementation to understand if this can be optimized.

Thanks,

Ian

sjg20, 2023-02-13:
IMO this is an odd use of a binary blob...

a) the source is available but sometimes need to be secret sometimes
b) or maybe it will not be secret but it is too big when using Zephyr (or will it?)
c) we are unable to build something with two binaries (or are we?)

If this goes ahead, I hope it can be quickly removed once one of these is resolved

carlescufi, 2023-02-15:
Approved by TSC, closing.

closed 2023-02-15 by carlescufi