zephyrproject-rtos/zephyr #54554: arch.arm.swap.tz fails on v2m_musca_b1_ns

stephanosio, 2023-02-07:
**Describe the bug**

`tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz` test fails on `v2m_musca_b1_ns`.

**To Reproduce**

```
scripts/twister -v -N -M -p v2m_musca_b1_ns -s tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz
```

**Expected behavior**
Test passes.

**Impact**
Unclear

**Logs and console output**

```
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/zephyr/zephyr/tests/arch/arm/arm_thread_swap_tz
-- CMake version: 3.20.5
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /__w/zephyr/zephyr/.cache
-- Zephyr version: 3.3.0-rc1 (/__w/zephyr/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: v2m_musca_b1_ns
-- Found host-tools: zephyr 0.15.2 (/opt/toolchains/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/opt/toolchains/zephyr-sdk-0.15.2)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/arm/v2m_musca_b1/v2m_musca_b1_ns.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/v2m_musca_b1_ns/tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /__w/zephyr/zephyr/twister-out/v2m_musca_b1_ns/tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/v2m_musca_b1_ns/tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz/zephyr/dts.cmake

warning: FPU (defined at soc/arm/nxp_imx/mcimx6x_m4/Kconfig.defconfig.mcimx6x_m4:11,
soc/arm/nxp_s32/s32ze/Kconfig.defconfig.series:15, arch/Kconfig:798) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: ((SOC_MCIMX6X_M4 && SOC_SERIES_IMX_6X_M4)
|| SOC_SERIES_S32ZE_R52 || CPU_HAS_FPU) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FPU and/or look up FPU in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: FPU_SHARING (defined at arch/Kconfig:818) was assigned the value 'y' but got the value 'n'.
Check these unsatisfied dependencies: FPU (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FPU_SHARING and/or look up FPU_SHARING in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

Parsing /__w/zephyr/zephyr/tests/arch/arm/arm_thread_swap_tz/Kconfig
Loaded configuration '/__w/zephyr/zephyr/boards/arm/v2m_musca_b1/v2m_musca_b1_ns_defconfig'
Merged configuration '/__w/zephyr/zephyr/tests/arch/arm/arm_thread_swap_tz/prj.conf'
Configuration saved to '/__w/zephyr/zephyr/twister-out/v2m_musca_b1_ns/tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz/zephyr/.config'
Kconfig header saved to '/__w/zephyr/zephyr/twister-out/v2m_musca_b1_ns/tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /__w/zephyr/zephyr/twister-out/v2m_musca_b1_ns/tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz
[1/166] Generating include/generated/version.h
-- Zephyr version: 3.3.0-rc1 (/__w/zephyr/zephyr), build: v3.3.0-rc1-97-g432ff20a72e1
[2/166] Preparing syscall dependency handling

[3/166] Generating misc/generated/syscalls_subdirs.trigger
[4/166] Generating ../../tfm/CMakeCache.txt
FAILED: tfm/CMakeCache.txt 
cd /__w/zephyr/zephyr/twister-out/v2m_musca_b1_ns/tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz/tfm && /usr/local/bin/cmake -GNinja -DTFM_TOOLCHAIN_FILE=/__w/zephyr/modules/tee/tf-m/trusted-firmware-m/toolchain_GNUARM.cmake -DCROSS_COMPILE=/opt/toolchains/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi -DCMAKE_BUILD_TYPE=MinSizeRel -DTFM_PLATFORM=arm/musca_b1/sse_200 -DCONFIG_TFM_BUILD_LOG_QUIET=ON -DTFM_LIB_MODEL=OFF -DCONFIG_TFM_SPM_BACKEND="IPC" -DBL2=TRUE -DMCUBOOT_IMAGE_VERSION_S=0.0.0+0 -DMCUBOOT_IMAGE_VERSION_NS=0.0.0+0 -DNS=FALSE -DTFM_ISOLATION_LEVEL=1 -DMCUBOOT_IMAGE_NUMBER=2 -DTFM_EXCEPTION_INFO_DUMP=OFF -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_SPM_LOG_LEVEL=TFM_SPM_LOG_LEVEL_INFO -DTFM_PARTITION_NS_AGENT_MAILBOX=OFF -DTFM_PARTITION_PROTECTED_STORAGE=ON -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE=ON -DTFM_PARTITION_CRYPTO=ON -DTFM_PARTITION_PLATFORM=ON -DTFM_PARTITION_INITIAL_ATTESTATION=ON -DTFM_PARTITION_FIRMWARE_UPDATE=OFF -DCRYPTO_RNG_MODULE_DISABLED=FALSE -DCRYPTO_KEY_MODULE_DISABLED=FALSE -DCRYPTO_AEAD_MODULE_DISABLED=FALSE -DCRYPTO_MAC_MODULE_DISABLED=FALSE -DCRYPTO_HASH_MODULE_DISABLED=FALSE -DCRYPTO_CIPHER_MODULE_DISABLED=FALSE -DCRYPTO_ASYM_SIGN_MODULE_DISABLED=FALSE -DCRYPTO_ASYM_ENCRYPT_MODULE_DISABLED=FALSE -DCRYPTO_KEY_DERIVATION_MODULE_DISABLED=FALSE -DMCUBOOT_PATH=/__w/zephyr/bootloader/mcuboot -DCONFIG_TFM_FP=soft -DMBEDCRYPTO_PATH=/__w/zephyr/modules/crypto/mbedtls -DTFM_TEST_REPO_PATH=/__w/zephyr/modules/tee/tf-m/trusted-firmware-m/../tf-m-tests -DPSA_ARCH_TESTS_PATH=/__w/zephyr/modules/tee/tf-m/trusted-firmware-m/../psa-arch-tests /__w/zephyr/modules/tee/tf-m/trusted-firmware-m
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Error at config/tfm_platform.cmake:29 (Message):
  Platform arm/musca_b1/sse_200 is not found in TF-M
Call Stack (most recent call first):
  CMakeLists.txt:28 (include)

-- Configuring incomplete, errors occurred!
[5/166] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.2
 - Commit SHA: 432ff20a72e1bfbc95b40c589b55075d4d82f22e

**Additional context**

`arch.arm.swap.tz_off` also fails.

microbuilder, 2023-06-20:
> @stephanosio This was closed due to being stale. Should this be closed or is this still an active issue?

Should be open, sorry I missed the stale bot.

jhedberg, 2023-08-29:
@microbuilder any update on this one?

microbuilder, 2023-09-07:
@jhedberg I just tried this with the latest code on `main` and I get the following:

```bash
$ scripts/twister -v -N -M -p v2m_musca_b1_ns -s tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-3331-g2f05af891de2
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/testplan.json
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 73.04 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 608 platforms (0.16%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/twister.json
INFO    - Writing xunit report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

Do you have time to do a quick test to verify that this is no longer an issue for you?

jhedberg, 2023-09-07:
@microbuilder I get the same successful result, so I'll just close this issue. Thanks!

closed 2023-09-07 by jhedberg