zephyrproject-rtos/zephyr #54555: Connection error with Android 13

xeanhort, 2023-02-07:
**Describe the bug**
Hello, I am writing because we have BLE connection problems in some Android 13 devices. I want to emphasize that these problems are only occurring since the release of android version 13 and that only occur in some models of cell phones, in most models everything works correctly.

As you can see in the attached log, these mobiles only manage to connect once in two times (some of them fail to connect the first time, the second time works, the third doesn't work, etc.; and other devices connect the 1st, 3rd, ... and fail to connect the 2nd, 4th, ...). When they don't connect, they show the error bt_att: Invalid len 2 for code 0x0e continously until a disconnection is made.

We are working with nordic nrf52840 and zephyr 2.7.3 but it is not working either with 2.7.99 or 2.5.0 and we can not move to the last version for now.

I also added a sniffer trace when this occur.

Thank you

**Expected behavior**
The device connects through BLE.

**Impact**
Uncertainty about how many devices are affected.

**Logs and console output**
[image]
[image]

[connection_error.pcapng (remove the leading `.log`](https://github.com/zephyrproject-rtos/zephyr/files/10676306/connection_error.pcapng.log)

**Environment (please complete the following information):**
 - Zephyr 2.7.3 (also 2.7.99 and 2.5.0)

carlescufi, 2023-02-07:
@xeanhort could you please try to reproduce with the latest `main` branch?

carlescufi, 2023-02-07:
Also @xeanhort, what platform/board are you running Zephyr on?

xeanhort, 2023-02-07:
@carlescufi we sre using a custom board, with the nrf52840.

This board worked for a long time (~june) with several phones, but we are finding that lately some Android 13 are failing with this strange behaviour. In fact, we first detected the issue with a OnePlus 8T that was working for several months and, right after its update to Android 13, it started to fail (also with older FW versions it keet failing).

We have the test with the main branch scheduled, but it might take some time to migrate everything.

jori-nordic, 2023-02-08:
@xeanhort can you try disabling `CONFIG_BT_GATT_READ_MULTIPLE`? The failing opcode is `BT_ATT_OP_READ_MULT_REQ` which is a new feature (so it makes sense it starts failling only on recent android stack versions).

xeanhort, 2023-02-08:
@jori-nordic We disabled this flag but it keeps failing. Nevertheless, now we are having different behavior.

- The project configuration:
[image]
- Now we are receiving different messages when the connection fails: 
[image]
- Phone 1 (Realme GT Neo2):
    - This phone keeps the same behavior: when it connects it shows the new log messages (Unhandled ATT), and the odd connection it works (like before).
    - Attached sniffer file with the new connection process: 
[connection_error_config_multiple_disabled.pcapng (remove the leading `.log`)](https://github.com/zephyrproject-rtos/zephyr/files/10685078/connection_error_config_multiple_disabled.pcapng.log)
- Phone 2 (OnePlus 8T):
    - This phone was working half of the times, but now it always fails (it never connects). The log messages are the same (Unhandled ATT).

Thank you

nordicjm, 2023-02-15:
@Thalley any thoughts on this?

Thalley, 2023-02-16:
Hmm, the unhandled request is still `BT_ATT_OP_READ_MULT_REQ`. Weird that it continues to attempt to read using that, since we clearly respond with `BT_ATT_ERR_NOT_SUPPORTED`.

For the first issue, it is Android being incorrect, as it uses `BT_ATT_OP_READ_MULT_REQ` with a single handle (thus len = 2). The core spec clearly states that at least 2 handles shall be used: 
[image]

This is clearly a bug in Android, and we should close it here

closed 2023-02-16 by jori-nordic