zephyrproject-rtos/zephyr #54567: Assertion in z_add_timeout() fails in drivers.uart.uart_mix_poll_async_api test

anangl, 2023-02-07:
**Describe the bug**
Execution of the `tests/drivers/uart/uart_mix_fifo_poll/drivers.uart.uart_mix_poll_async_api` test on the `nrf52840dk_nrf52840` board usually ends up with an assertion failure reported for the `z_add_timeout` function (`kernel/timeout.c:99`).

While debugging, I figured out that this is caused by a race condition in the execution of the timer expiration handler. When a timer is restarted from a high priority interrupt, it may happen that the timer is re-added to the timeout list right after it is removed from that list prior to execution of its expiration handler but before that execution actually occurs. `z_add_timeout()`, called from `z_timer_expiration_handler()` for periodic timers, turns out then to be adding a timeout that is already added to the timeout list and the assertion fails.

This problem is also reported in #53613 but since is not the only one there, I decided to create a separate issue for it.

**To Reproduce**
Steps to reproduce the behavior:
1. Execute `west twister -p nrf52840dk_nrf52840 -s tests/drivers/uart/uart_mix_fifo_poll/drivers.uart.uart_mix_poll_async_api --device-testing --device-serial /dev/ttyACM0 --fixture gpio_loopback`
2. See `twister-out/nrf52840dk_nrf52840/tests/drivers/uart/uart_mix_fifo_poll/drivers.uart.uart_mix_poll_async_api/handler.log`

**Logs and console output**
```
*** Booting Zephyr OS build v3.3.0-rc2-59-g36421f2efc8d ***
Running TESTSUITE uart_mix_fifo_poll
===================================================================
START - test_mixed_uart_access
ASSERTION FAIL [!sys_dnode_is_linked(&to->node)] @ WEST_TOPDIR/zephyr/kernel/timeout.c:99
E: r0/a1:  0x00000004  r1/a2:  0x00000063  r2/a3:  0x00000000
E: r3/a4:  0x00000021 r12/ip:  0x00000000 r14/lr:  0x0000aed3
E:  xpsr:  0x41000021
E: Faulting instruction address (r15/pc): 0x0000e22a
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x200000d8 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.1
 - Commit 36421f2efc8d18ba520287804f0cc9df5cf570d8

andyross, 2023-02-07:
> When a timer is restarted from a high priority interrupt, it may happen that the timer is re-added to the timeout list right after it is removed from that list prior to execution of its expiration handler but before that execution actually occurs.

I'm not sure I understand this?  Can you walk me through the sequence you're seeing?  Presumably you have a higher priority interrupt than the timer that has interrupted execution of a timeout handler?  What is it trying to do?

andyross, 2023-02-07:
Oh... it's in k_timer, not the timeout subsystem?  I'll comment in the PR.

andyross, 2023-02-07:
Unless I'm misunderstanding, I think this can be closed?  It's absolutely true (as you discovered with k_timer) that clients of the timeout subsystem need to be prepared for the case where the callback is called in a context where the object has already been requeued.  And maybe we should put a warning about that somewhere.   But I think that's mostly an "as-designed" thing, because we **really** don't want to have to hold a lock across the timeout callbacks for latency reasons.

Most clients can handle this as you did with k_timer, or via other means.  For thread timeouts (the other big user), we're OK as the behavior is idempotent (though it does show up as behavior: threads might wake up due to the race or not depending on how quickly you cancelled the timeout, but that's only because you were racing the wakeup in the first place!).

PR #54568 (2023-02-08): kernel: timer: Eliminate a race condition in expiration handling
When a timer is restarted from a high priority interrupt, it may happen that the timer is re-added to the timeout list right after it is removed from that list prior to execution of its expiration handler but before that execution actually occurs. This leads to an assertion failure reported for `z_add_timeout()` because then that function, called from `z_timer_expiration_handler()` for periodic timers, turns out to be adding a timeout that is already added to the timeout list.
This commit detects such situation in `z_timer_expira [... cut]
closed 2023-02-08 by carlescufi