zephyrproject-rtos/zephyr #54572: QEMU networking breakage (Updating nrf-sdk 2.1->2.2 , implies zephyr 3.1 -> 3.2)

clamattia, 2023-02-07:
**Describe the bug**
 - target: `qemu_x86`
 - dockerized environment based on ubuntu 22.04, zephyr-sdk: 0.15.0
 - This is a regression and worked well with nrf-sdk 2.1

Updating nrf-sdk (and therefore zephyr) breaks the `qemu_x86` build related to networking. Commenting out `CONFIG_NETWORKING=y` makes the simulation ok again. An example to reproduce attached below. The error reads:
```
Booting from ROM..
[00:00:00.000,000] <err> os: Page fault at address (nil) (error code 0x2)
[00:00:00.000,000] <err> os: Linear address not present in page tables
[00:00:00.000,000] <err> os: Access violation: supervisor thread not allowed to write
[00:00:00.000,000] <err> os: PTE: not present
[00:00:00.000,000] <err> os: EAX: 0x00000040, EBX: 0x001116c0, ECX: 0x00000206, EDX: 0x00000000
[00:00:00.000,000] <err> os: ESI: 0x001196b0, EDI: 0x00000000, EBP: 0x001185f4, ESP: 0x001185ec
[00:00:00.000,000] <err> os: EFLAGS: 0x00000207 CS: 0x0008 CR3: 0x0011f000
[00:00:00.000,000] <err> os: call trace:
[00:00:00.000,000] <err> os: EIP: 0x0010c232
[00:00:00.000,000] <err> os:      0x00106e8a (0x1196b0)
[00:00:00.000,000] <err> os:      0x0010504a (0x118628)
[00:00:00.000,000] <err> os:      0x0010c942 (0x0)
[00:00:00.000,000] <err> os:      0x0010c98e (0x118644)
[00:00:00.000,000] <err> os:      0x00101197 (0x0)
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x112320 (unknown)
[00:00:00.010,000] <err> os: Halting system
```

west.yml:
```
manifest:
  remotes:
    - name: ncs
      url-base: https://github.com/nrfconnect
  projects:
    - name: nrf
      repo-path: sdk-nrf
      remote: ncs
      revision: v2.1.0 # updating to v2.2.0 here breaks the qemu
      import:
        name-allowlist:
          - zephyr
          - nrfxlib
          - hal_nordic
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project with above `west.yml` and:
2. `CMakeLists.txt` :
```
cmake_minimum_required(VERSION 3.13.1)

list(APPEND BOARD_ROOT ${CMAKE_CURRENT_SOURCE_DIR})

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})

project(breakage)

target_sources(app PRIVATE main.c)
```
3. `Kconfig`
```
source "$ZEPHYR_BASE/Kconfig.zephyr"
```
4. `main.c`
```
#include <zephyr/logging/log.h>
#include <zephyr/kernel.h>

LOG_MODULE_REGISTER(logging_main);

void main(void)
{
    for(;;){
        LOG_INF("Running ...");
        k_msleep(300);
    }
}
```
5. `prj.conf`
```
# Console and log output
CONFIG_CONSOLE_SUBSYS=y
CONFIG_LOG=y

## Networking QEMU
CONFIG_NET_L2_ETHERNET=y
#CONFIG_ETH_DRIVER=y not available in 2.1
CONFIG_NET_QEMU_ETHERNET=y
CONFIG_ETH_E1000=y
CONFIG_PCIE=y

## Networking
# uncommenting this for nrf-sdk version 2.2 (as opposed to 2.1) will cause page fault when running qemu
CONFIG_NETWORKING=y
```
6. build: `west build -b qemu_x86`
7. run ok:
```
sudo WESTWORK/tools/net-tools/net-setup.sh up
west build -t run
sudo WESTWORK/tools/net-tools/net-setup.sh down
```
8. change `v2.2.0` in `west.yml`, rebuild, rerun, will  fails with error as described above
9. comment `CONFIG_NETWORKING=y` in `prj.conf`, rebuild, rerun, will run ok again

**Expected behavior**
App should start and main loop should start printing (independent of version):
```
Booting from ROM..
*** Booting Zephyr OS build v3.1.99-ncs1  ***
[00:00:00.010,000] <inf> logging_main: Running ...
[00:00:00.320,000] <inf> logging_main: Running ...
[00:00:00.630,000] <inf> logging_main: Running ...
```

**Impact**
We need a solution/workaround to maintain the qemu infrastructure, when updating to newer version.

rlubos, 2023-02-10:
It seems to be the same problem already reported in https://github.com/zephyrproject-rtos/zephyr/issues/51829, fixed with https://github.com/zephyrproject-rtos/zephyr/pull/51845. Please use newer Zephyr revision.

@clamattia Since this is Zephyr project space, please use Zephyr revisions when reporting problems, it'll make our lives simpler. `sdk-nrf` issues should be reported to Nordic DevZone.

clamattia, 2023-02-13:
@rlubos Thank you for analyzing and for the quick suggestion. Cherry-picking this indeed solved the hard-fault, when starting qemu.

Thanks, I will use the zephyr revision in future issue reports or report to DevZone.

closed 2023-02-13 by clamattia