zephyrproject-rtos/zephyr #54576: Error during IPv4 defragmenting

ssharks, 2023-02-07:
**Describe the bug**
During testing with the Maxwell Pro, with quite a few IPv4 defragmenting tests errors occurred on the console indicating the error in the IPv4 fragmenting stacks. The list of failing IPv4 fragmenting is given by:

https://github.com/zephyrproject-rtos/test_results/issues?q=is%3Aissue+is%3Aopen++Fragment+IPv4

The testsuite used the echo server sample with the config

The bug is reproduced by feeding fragmented packets to the echo server with the config file of:
https://github.com/hakehuang/zephyr/blob/tcp_ip_testing_maxwell/samples/net/sockets/echo_server/overlay-maxwell.conf

**Expected behavior**
IPv4 fragmenting should work and running it should not throw errors

**Impact**
At least a significant portion of the IPv4 fragmenting tests are failing.

**Logs and console output**
Some logging provide by @hakehuang :
<err> net_ipv4: Reassembled IPV4 verify failed, dropping id 215
<err> net_ipv4: no slots available for 0xd7
<err> net_ipv4: no slots available for 0xd7
…
Or
<err> net_ipv4: no slots available for 0xc

nordicjm, 2023-02-08:
So I see IPv4 fragmentation Kconfigs are set in that configuration, but `NET_IPV4_FRAGMENT_MAX_PKT` is not updated from the default of 2, so whilst 16 incoming packets can be fragmented, there can only be 2 fragments per packet. What is actually being used to test here - what is the size of the packets? What is the MTU of the transport? What are the fragment sizes? Because from the linked tests there, nearly all mention 4 fragments, which isn't going to work in the default configuration

carlescufi, 2023-02-08:
> So I see IPv4 fragmentation Kconfigs are set in that configuration, but `NET_IPV4_FRAGMENT_MAX_PKT` is not updated from the default of 2, so whilst 16 incoming packets can be fragmented, there can [...]

@hakehuang could you please fill in the missing information here from the Maxwell reports?

hakehuang, 2023-02-09:
> @hakehuang could you please fill in the missing information here from the Maxwell reports?

@carlescufi 
this is the failure reason：

tcp.v4 got no echo response, which is not an expected result.

```
Datagrams are fragmented into MTU sized chunks, starting at the front of the datagram. The fragments overlap with fragment N+1 overlapping the end of fragment N by 16 bytes (two Network Fragmentation Blocks). E.g. Here is a sequence of NFBs in the fragments each holding 6 NFBs:

pkt 0: ABCDEF
pkt 1:     EFGHIJ
pkt 2:         IJKLMN
...

The MTU is 512. A final short fragment may occur at the end. Fragments are transmitted in order of increasing segment offset.
```

nordicjm, 2023-02-09:
With the configuration linked above, anything with more than 2 fragments will be discarded.

ssharks, 2023-02-09:
I took a look at the code and as I understand it that NET_IPV4_FRAGMENT_MAX_COUNT give the maximum number of packets of which fragments can be kept, which each maximally NET_IPV4_FRAGMENT_MAX_PKT fragments,
The required memory space is the product of both for that reason.

For the Maxwell Pro tests can be rather small (4) or something like that, but the number of required NET_IPV4_FRAGMENT_MAX_PKT could be large, like > 16.
Or am I misunderstanding things wrongly?

nordicjm, 2023-02-10:
That sounds about right. I assume it would be running 1 test at a time, with 1 connection (or maybe a few) but if it's trying to test things like reassembly then it will be using many fragments of a small size.

ssharks, 2023-02-10:
After rethinking, as far as I understand the code of IPv4 and IPv6 fragmenting, they only recombine at the end when all fragments are present, there is no recombination of already adjacent fragments in the meantime?
Is it designed to handle overlapping segments?

I see a lot similarities with the out of order reception logic of TCP.
https://github.com/zephyrproject-rtos/zephyr/blob/a5e04ccd4ab976b99f45d106e3de873fc8dd882a/subsys/net/ip/tcp.c#L770
and 
https://github.com/zephyrproject-rtos/zephyr/blob/a5e04ccd4ab976b99f45d106e3de873fc8dd882a/subsys/net/ip/tcp.c#L1860

It took me considerable time to get that to work OK by modifying in the original structure. At the moment is only able to handle a single section of out of order data. I'm not 100% happy with the code structure and I see a lot of similarities with the IP defragmenting. In an ideal case, it would be nice to put the ordering logic into a common piece of code separated from the protocol specifics and use it in TCP, IPv4 and IPv6. This would also help in the testing.

ssharks, 2023-02-16:
@nordicjm: In the following branch: https://github.com/ssharks/zephyr/tree/net/ip4v_frag_test

I created a few tests that test with customized fragmented packets. Reversed order fragments are processed correctly, similar as 4 fragments, although the stack does not accept overlapping fragments, which are contributing for the most failing tests. Feel free to modify the test or provide feedback.

nordicjm, 2023-02-17:
Correct, it does not work with overlapping fragments. To which I would argue there's not really much point in supporting anyway, the IPv4 RFC is from 1981 - that's 42 years ago, technology has moved on vastly since then, we don't have dial up links or slow serial links between routers these days, I think you would be far stretched to find any sort of networking device released in the past 15 years that has a MTU of less than 1500 bytes. So from my point of view, unless there is a real need for a complex system capable of dealing with overlapping fragments, is support for it really needed? It makes the code use more flash space and work in a more complex way for what is probably not needed, also given that devices these days are even using 64KB jumbo frames.

carlescufi, 2023-03-28:
> Correct, it does not work with overlapping fragments. To which I would argue there's not really much point in supporting anyway, the IPv4 RFC is from 1981 - that's 42 years ago, technology has moved [...]

@ssharks @rlubos could you please assess whether @nordicjm's suggestion to simply not do anything about this since he does not deem it worthwhile is OK with you?

rlubos, 2023-03-28:
> @ssharks @rlubos could you please assess whether @nordicjm's suggestion to simply not do anything about this since he does not deem it worthwhile is OK with you?

Fine by me. IPv6 explicitly forbids handling of overlapping fragments (due to security concerns), so unless there's a strong need to have this feature, I don't see a good reason to implement it merely for the sake of satisfying Maxwell pro.

closed 2023-03-29 by ssharks