zephyrproject-rtos/zephyr #54609: driver: led: kconfig symbols mix up

erwango, 2023-02-08:
**Describe the bug**
`driver/led` is an unusual subsystem, it covers 2 generic (ie hw agnostic) drivers: `led_gpio` and `led_pwm`.
Since they are generic, these 2 drivers can be enabled at same time in the same build.

A top level Kconfig symbol controls the whole subsystem: `CONFIG_LED`.
Each driver has its own symbol: `CONFIG_LED_GPIO` and `CONFIG_LED_PWM`.

Initially, enabling one driver required enabling both top and driver kconfig symbols, for instance:
```
CONFIG_LED=y
CONFIG_LED_PWM=y
```

Since https://github.com/zephyrproject-rtos/zephyr/pull/48827, `CONFIG_LED_GPIO` and `CONFIG_LED_PWM` values are set based on presence of an enabled node with matching compatible.
Then `CONFIG_LED_PWM` and `CONFIG_LED_GPIO` are not enabled in samples anymore.
Drivers are enabled only based on boards device tree descriptions.
Problem is that, most often, the same hardware resource is described by nodes with compatibles `"gpio-leds"` and `"pwm-leds"`: the pin connected to the board LED.

As a consequence, when CONFIG_LED is set, both drivers `led_gpio` and `led_pwm` are compiled in and try to control the same pin.
On STM32 at least this lead to functional issue, as pin is:
- first configured using pinctrl thought pwm device (used by `pwm-leds` driver)
- then configured using gpio driver (used `led_gpio` driver)

This is the issue that https://github.com/zephyrproject-rtos/zephyr/pull/54019 tried to fix, by disabling nodes depending on the driver used by the sample.
But this is not sufficient, potentially all boards are impacted. And this isn't in line with recommendations https://docs.zephyrproject.org/latest/build/dts/dt-vs-kconfig.html:
> Use Kconfig to configure software support to build into the final image. Examples include whether to add networking support, which drivers are needed by the application, etc.

In current case, device tree is dictating which software component is using the hardware.

**Expected behavior**
Back to the initial situation:
`CONFIG_LED_PWM` and `CONFIG_LED_GPIO` should control which driver is used.

**Impact**
If this isn't fixed, for each board which want to support both `"gpio-leds"` and `"pwm-leds" (which is likely, why would it be otherwise ?), following should be done:
- device tree should describe both types of nodes and leave one of them disabled (say `"pwm-leds" node)
- in applications using `led_pwm` (`samples/drivers/led_pwm` for instance):
  - disable `"gpio-leds"` node
  - enable `"pwm-leds" node

(Basically this requires applying https://github.com/zephyrproject-rtos/zephyr/pull/54019 to all boards)

simonguinot, 2023-02-10:
Hi @erwango,

I don't think it is a bug. CONFIG_LED_PWM and CONFIG_LED_GPIO still control which driver is selected at the end. The difference since #48827 is that the LED drivers are now automatically selected if a compatible and enabled node is found in DT. But you can unselect the driver you don't want from your board configuration file (or from a sample).

As an example let's take the nucleo_u575zi_q board in https://github.com/zephyrproject-rtos/zephyr/pull/54019.

"pwm-leds" and "gpio-leds" can be both enabled in the board DT. And in the led_pwm sample the led_gpio driver can be disabled by adding the following configuration into _samples/drivers/led_pwm/boards/nucleo_u575zi_q.conf_:
```
CONFIG_LED_GPIO=n
```

And you can drop the overlay file. That's it ! I don't understand why @FRASTM is ignoring this suggestion. This is closer to what you are trying to achieve than disabling nodes in DT.

I believe that your use case is quite uncommon. Most of the boards don't have conflicting LED devices enabled in DT. And auto-selecting the LED drivers from DT removed 19 lines (See [0e1a7e0241135bae576698eab5b4bc7b73cc824b](https://github.com/zephyrproject-rtos/zephyr/commit/0e1a7e0241135bae576698eab5b4bc7b73cc824b)) So IMHO it is a good thing even if we have to unselect a LED driver for few boards.

In addition I don't think that DT is dictating the driver selection because Kconfig is still in control at the end.

Now that being said, I am not sure I am right. So let's ask @galak and @Mani-Sadhasivam their opinion.

galak, 2023-04-06:
I agree with @Mani-Sadhasivam that this should be handled by having the devicetree only enabling one of the hw configs and utilize overlays to get the proper HW config based on what the application wants.

closed 2023-08-20 by github-actions[bot]