zephyrproject-rtos/zephyr #54610: samples: kernel: metairq_dispatch: sample.kernel.metairq_dispatch hangs

nordic-piks, 2023-02-08:
**Describe the bug**
The samples/kernel/metairq_dispatch/sample.kernel.metairq_dispatch hangs sometimes

Observed for:
- nrf5340dk_nrf5340_cpunet

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340_cpunet connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/kernel/metairq_dispatch -p nrf5340dk_nrf5340_cpunet --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-02-08T01:29:10.664Z] [0mERROR   - *** Booting Zephyr OS build v3.3.0-rc2-44-g461a13c085de ***[0m
[2023-02-08T01:29:10.664Z] [0mI: Starting Thread0 at priority -2[0m
[2023-02-08T01:29:10.664Z] [0mI: Starting Thread1 at priority -1[0m
[2023-02-08T01:29:10.664Z] [0mII: ERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mMERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0mERROR: Queue full, event dropped![0m
[2023-02-08T01:29:10.664Z] [0m0 T0 mirq 2 disp 3 proc 2 real 5[0m
[2023-02-08T01:29:10.664Z] [0mI: M6 T0 mirq 100 disp 2873 proc 0 real 0[0m
[2023-02-08T01:29:10.664Z] [0mI: M8 T0 mirq 1880 disp 2894 proc 3 real 3[0m
[2023-02-08T01:29:10.664Z] [0mI: M9 T1 mirq 1880 disp 3019 proc 2 real 2[0m
[2023-02-08T01:29:10.664Z] [0mI: M10 T1 mirq 1880 disp 3142 proc 3 real 3[0m
[2023-02-08T01:29:10.664Z] [0mI: M30 T1 mirq 793 disp 1389 proc 3 real 3[0m
[2023-02-08T01:29:10.664Z] [0mI: M32 T1 mirq 792 disp 1513 proc 1 real 1[0m
[2023-02-08T01:29:10.664Z] [0m: Starting Thread2 at priority 0[0m
[2023-02-08T01:29:10.664Z] [0mI: M1 T2 mirq 2 disp 3737 proc 1 real 1[0m
[2023-02-08T01:29:10.664Z] [0mI: M3 T2 mirq 2 disp 3842 proc 0 real 0[0m
[2023-02-08T01:29:10.664Z] [0mI: M4 T2 mirq 102 disp 3954 proc 1 real 1[0m
[2023-02-08T01:29:10.664Z] [0mI: M5 T2 mirq 101 disp 4074 proc 3 real 3[0m
[2023-02-08T01:29:10.664Z] [0mI: M31 T2 mirq 793 disp 2210 proc 3 real 3[0m
[2023-02-08T01:29:10.664Z] [0mI: Starting Thread3 at priority 1[0m
[2023-02-08T01:29:10.664Z] [0mI: M2 T3 mirq 2 disp 4431 proc 0 real 0[0m
[2023-02-08T01:29:10.664Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: v3.3.0-rc2-44-g461a13c085de

tmon-nordic, 2023-03-09:
I cannot reproduce this on PCA10095 2.0.0 2022.10
```
*** Booting Zephyr OS build zephyr-v3.3.0-953-gf19e6d481164 ***
I: Starting Thread0 at priority -2
I: Starting Thread1 at priority -1
I: M0 T1 mirq 2 disp 3 proc 3 real 5
I: M3 T1 mirq 2 disp 176 proc 2 real 2
I: M6 T1 mirq 2 disp 277 proc 2 real 2
I: M11 T1 mirq 5 disp 367 proc 2 real 2
I: M12 T1 mirq 3 disp 482 proc 3 real 3
I: M13 T1 mirq 3 disp 599 proc 2 real 2
I: M14 T1 mirq 5 disp 710 proc 0 real 0
I: M15 T1 mirq 3 disp 825 proc 1 real 1
I: M20 T1 mirq 3 disp 929 proc 2 real 2
I: M23 T1 mirq 2 disp 1034 proc 0 real 0
I: M37 T1 mirq 2 disp 1099 proc 1 real 1
I: M1 T0 mirq 2 disp 1353 proc 2 real 2
I: M2 T0 mirq 4 disp 1465 proc 3 real 3
I: M7 T0 mirq 3 disp 1560 proc 2 real 2
I: M22 T0 mirq 2 disp 1630 proc 2 real 2
I: M27 T0 mirq 2 disp 1728 proc 3 real 3
I: M29 T0 mirq 2 disp 1839 proc 1 real 1
I: M30 T0 mirq 4 disp 1953 proc 0 real 0
I: M32 T0 mirq 2 disp 2067 proc 3 real 3
I: M33 T0 mirq 2 disp 2183 proc 3 real 3
I: M35 T0 mirq 2 disp 2296 proc 3 real 3
I: M36 T0 mirq 4 disp 2413 proc 3 real 3
I: M40 T0 mirq 1 disp 2518 proc 1 real 1
I: Starting Thread2 at priority 0
I: M9 T2 mirq 2 disp 2852 proc 1 real 1
I: M16 T2 mirq 3 disp 2947 proc 0 real 0
I: M18 T2 mirq 3 disp 3060 proc 2 real 2
I: M21 T2 mirq 4 disp 3169 proc 0 real 0
I: M25 T2 mirq 2 disp 3275 proc 1 real 1
I: M26 T2 mirq 2 disp 3390 proc 0 real 0
I: M28 T2 mirq 2 disp 3500 proc 1 real 1
I: M38 T2 mirq 4 disp 3580 proc 1 real 1
I: M39 T2 mirq 2 disp 3697 proc 3 real 3
I: Starting Thread3 at priority 1
I: M4 T3 mirq 4 disp 4049 proc 3 real 3
I: M5 T3 mirq 2 disp 4165 proc 0 real 0
I: M8 T3 mirq 2 disp 4266 proc 3 real 3
I: M10 T3 mirq 2 disp 4374 proc 2 real 2
I: M17 T3 mirq 3 disp 4471 proc 2 real 2
I: M19 T3 mirq 3 disp 4585 proc 3 real 3
I: M24 T3 mirq 4 disp 4688 proc 0 real 0
I: M31 T3 mirq 2 disp 4780 proc 3 real 3
I: M34 T3 mirq 4 disp 4887 proc 3 real 3
I:         ---------- Latency (cyc) ----------
I:             Best    Worst     Mean    Stdev
I: MetaIRQ        2        5        3        1
I: Thread0     1353     2413     1862      333
I: Thread1        3     1099      591      345
I: Thread2     2852     3697     3274      274
I: Thread3     4049     4887     4474      269
I: MetaIRQ Test Complete
```

tmon-nordic, 2023-03-09:
While I could not reproduce it yesterday, I can sometimes reproduce it today. It is worthwhile to mention that the test only flashes the network core but leaves whatever application was flashed (and running) on the application core - therefore what software you end up running on the target depends on what you flashed previously to the application core.

My suspicious is that the network core is not fully independent from the application core, i.e. that the application core can cause delays on network core and thus ruin this test.

tmon-nordic, 2023-03-10:
The failure strongly depends on `k_cycle_get_32()` output. Most `k_cycle_get_32()` calls contribute to small deviations but the initial seed value obtained in https://github.com/zephyrproject-rtos/zephyr/blob/68955ebdb6d8a9c6598810446fecc316209089e1/samples/kernel/metairq_dispatch/src/msgdev.c#L31-L47
strongly determines the outcome. Forcing state to `555810816` gives `Queue full, event dropped!` on almost every run. There are probably much worse seeds but `555810816` is first faulty seed value that I observed on nRF5340DK.

tmon-nordic, 2023-03-10:
Crude patch to make the issue 100% reproducible on nrf52840dk_nrf52840 is to force RTC reset and wait for the counter to reach 0x2121.

```diff
diff --git a/samples/kernel/metairq_dispatch/src/main.c b/samples/kernel/metairq_dispatch/src/main.c
index a767cb8d76..e58000dbe4 100644
--- a/samples/kernel/metairq_dispatch/src/main.c
+++ b/samples/kernel/metairq_dispatch/src/main.c
@@ -223,8 +223,18 @@ static void thread_fn(void *p1, void *p2, void *p3)
        }
 }
 
+#include <hal/nrf_rtc.h>
+
 void main(void)
 {
+       nrf_rtc_task_trigger(NRF_RTC1, NRF_RTC_TASK_TRIGGER_OVERFLOW);
+       while (NRF_RTC1->COUNTER > 0x0010) {
+               /* Make sure the counter resets */
+       }
+       while (NRF_RTC1->COUNTER != 0x2121) {
+               /* Wait for one of the problematic values */
+       }
+
        for (long i = 0; i < NUM_THREADS; i++) {
                /* Each thread gets a different priority.  Half should
                 * be at (negative) cooperative priorities.  Lower
```

tmon-nordic, 2023-03-14:
> Crude patch to make the issue 100% reproducible on nrf52840dk_nrf52840 is to force RTC reset and wait for the counter to reach 0x2121.

This patch is not only crude but completely bogus. It causes really bad issues. Instead of directly triggering overflow, `z_nrf_rtc_timer_trigger_overflow()` should be called (ensure that `CONFIG_NRF_RTC_TIMER_TRIGGER_OVERFLOW` is enabled) - but then the issue does not reproduce on nrf52840dk_nrf52840.

tmon-nordic, 2023-03-16:
I am observing very strange feedback loop. I am not sure how much it is affected due to stepping on breakpoint in debugger (i.e. if it is possible to observe such behavior without breakpoint - which does not stop RTC timer).

The feedback loop is in `sys_clock_announce()`
https://github.com/zephyrproject-rtos/zephyr/blob/b4538ffcdbfe6b0384601861c9469b93740f881a/kernel/timeout.c#L213-L226

the entry on list calls `dev_timer_expired()` indirectly via `t->fn(t);`. At the end of `dev_timer_expired()` new entry is scheduled in `timeout_reset()`:
https://github.com/zephyrproject-rtos/zephyr/blob/b4538ffcdbfe6b0384601861c9469b93740f881a/samples/kernel/metairq_dispatch/src/msgdev.c#L71-L81

Once the control returns back to `sys_clock_announce()` loop, it picks up the just added timeout and executes it (i.e. calls `dev_timer_expired()`) and thus the cycle repeats. The feedback loop ends when `m.seq < MAX_EVENTS` is no longer true.

@andyross @nordic-krch Does this sound like some issue with `sys_clock_announce()` and/or the sample (`dev_timer_expired()`) or is it rather indicating something wrong with `nrf_rtc_timer.c` implementation?

#55729 prevents loss of messages even if the feedback loop hits (at most `MAX_EVENTS` messages are produced before the feedback loop breaks), but it does not address the feedback loop phenomena itself.

tmon-nordic, 2023-03-17:
> I guess it's plausibly due to your debug rig, but something is locking up interrupt deliver for an unacceptable amount of time.

The debug problem is that whenever you are stopped at breakpoint on nRF52840, the RTC timer (name is confusing, it is just 24-bit up-counter) keeps going. After resuming from any breakpoint, there is rather significant amount of pending ticks to announce (and these ticks do get announced - which triggers the feedback loop).

> The root cause here is that the timer driver was **eleven ticks behind** the time it was supposed to have fired.

The eleven ticks was just one of the debug attempts I did... In reality, it is much worse! Hopefully now I see the full picture, so I am actually fine with simply letting the workaround in (as not much can really be done in tickless kernel case).

The first time a queue gets full depends heavily on RNG seed which is obtained from RTC COUNTER. This value is mostly the same on every reset (because RTC COUNTER gets reset when system resets - which happens to be non-problematic RNG seed), but sometimes (when this issue occurs) the value can get off the "expected constant" (largest deviation from the "expected constant" I have seen was 2593 counts).

When you get the problematic RNG seed, then if `printk()` is printing to 115200 baud UART, the `printk()` itself takes between 96 and 97 RTC counts. Note that the `printk()` is called in ISR, so there cannot be any tick interrupts. The lack of tick interrupts does not really matter, because the system is running `TICKLESS_KERNEL`. When `dev_timer_expired()` calls `timeout_reset()`, which in turn calls `z_add_timeout()`, the `curr_tick` is already at least 96 ticks behind! Considering that `metairq_dispatch` sample adds timeouts in range `[0; 7]` ticks - it is inevitable that at least 13 next `dev_timer_expired()` will happen back-to-back in `sys_clock_announce()` loop. And when the loop finishes, there will be even more unannounced ticks - so the first time `k_msgq_put()` in `dev_timer_expired()` fails (due to lack of space in queue), snowballs the remaining events to also fail in the feedback loop.

tmon-nordic, 2023-03-17:
@andyross Could you please comment on following silly example from kernel perspective:

```
struct k_timer timer, timer2;
uint32_t cycles_before, cycles_after;

void expiry_fn2(struct k_timer *timer)
{
	cycles_after = k_cycle_get_32();
	printk("%d - %d = %d\n", cycles_after, cycles_before, cycles_after - cycles_before);
}

void expiry_fn(struct k_timer *timer)
{
	cycles_after = k_cycle_get_32();
	printk("%d - %d = %d\n", cycles_after, cycles_before, cycles_after - cycles_before);
	k_busy_wait(5 * 1000 * 1000);
	cycles_before = k_cycle_get_32();
	k_timer_start(&timer2, K_MSEC(10000), K_NO_WAIT);
}

void main(void)
{
	k_timer_init(&timer, expiry_fn, NULL);
	k_timer_init(&timer2, expiry_fn2, NULL);
	cycles_before = k_cycle_get_32();
	k_timer_start(&timer, K_MSEC(10000), K_NO_WAIT);
}
```

The expectation would be that the two lines printed by this silly example would have relatively small difference. However, on nRF52840DK, this gives:
```
*** Booting Zephyr OS build zephyr-v3.3.0-1145-g063c03948088 ***
339821 - 12140 = 327681
667500 - 502566 = 164934
```

which means that the `k_timer_start()` inside fires **before its timeout expires**. It all boils down to what is the base - if you consider the `expiry_fn()` entry as the base, then `ktimer2` expiry time is expected. However, if you consider the time when `k_timer_start()` was called as the base, then it is wrong.

The above silly example seriously exaggerates what is happening in metairq sample.

andyross, 2023-03-17:
Just to be clear: that example is spinning for five (!) seconds inside the timer ISR.

It's also comparing the "time of timeout expiration" (which is a tick value computed inside the timer ISR) to a cycle count (generally the underlying hardware counter) from within the timeout, which runs afoul of the "virtual ticks" rule with timeouts: they appear to be running at the tick time they were scheduled at, even if they're late.  This is so that simple/naive algorithms that rely on rescheduling timeouts from their handlers get a steady stride and don't lose time due to late interrupts or other latency.

Finally, it appears that you expect that the time you spend in k_busy_wait() in the ISR is going to be reflected in the number of ticks being announced in real time, from within the ISR.  That's not the way it works, the tick delta is computed once, in the timer driver, and then announced to completion.  Only then does the system return from the ISR and potentially schedule another interrupt that would announce the timeout you tried to schedule synchronously.

Basically: the fundamental assumption of the timeout subsystem is that timeouts are arriving with low latency.  And when they don't, it tries to do the right thing anyway by "faking" the current time.  But nothing can really recover latency after a late interrupt; it's gone.

tmon-nordic, 2023-03-20:
> And when they don't, it tries to do the right thing anyway by "faking" the current time. But nothing can really recover latency after a late interrupt; it's gone.

Yes, nothing can recover latency. But Zephyr tries to recover the latency - by shortening the relative timeouts scheduled in timeout handlers executed from `sys_clock_announce()`. Question is should such shortening take place?

Duration shortening that happens when `sys_clock_announce()` is running late effectively violates
https://github.com/zephyrproject-rtos/zephyr/blob/ec6122cd98beafe05a0910bac2df09f1688ef4d9/doc/kernel/services/timing/timers.rst#L55-L60

In the silly example, `K_MSEC(10000)` was requested, but the timer effectively fired `K_MSEC(5000)` after `k_timer_start` was called on `timer2`.

Or is it that the guarantee is only about the synthesized tick counter value and not about `K_MSEC(x)` representing x real-world milliseconds (within the source clock accuracy)?

andyross, 2023-04-04:
> But isn't this a wrong approach of those algorithms? Shouldn't they use a periodic timer instead?

The periodic behavior of k_timer is **precisely** the use case in question!  k_timer is based on the timeout mechanism we're discussing, and it doesn't "lose time" due to delayed interrupts because of this feature.  I buy that this isn't without controvery: this isn't the first time we've had a fight over how "current time" is exposed to timeout callbacks and the fact that in extreme conditions like this it can show a mismatch vs. the free-running cycle counter.  But on balance, I think we've made the right choice here.

PR #55729 (2023-04-05): samples: metairq_dispatch: Ensure large enough queues
Make queues used in metairq dispatch sample capable of storing all generated messages. This prevents seemingly random hangs that tend to happen with some random seeds.

Fixes: #54610
commit cbd8f8b4 (2023-04-05): samples: metairq_dispatch: Ensure large enough queues

Make queues used in metairq dispatch sample capable of storing all
generated messages. This prevents seemingly random hangs that tend to
happen with some random seeds.

Fixes: #54610