zephyrproject-rtos/zephyr #54622: Zephyr Input Subsystem proposal

fabiobaltieri, 2023-02-08:
### TL:DR;

Linux-like events for Zephyr, proof of concept here: https://github.com/zephyrproject-rtos/zephyr/pull/54620

## Introduction

Implement a subsystem in upstream Zephyr that can handle input events from various types of input devices and distribute them to other threads in the application.

### Problem description

Zephyr is lacking a generic input subsystem. Right now the closest thing is the [kscan](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/kscan.h) api, which is only meant to support [keyboard matrixes](https://github.com/zephyrproject-rtos/zephyr/blob/60a20471b561a0a3a74b377991fe1976c2ea83d7/include/zephyr/drivers/kscan.h#L35-L46) and has been somewhat abused to implement [touchscreens](https://github.com/zephyrproject-rtos/zephyr/blob/60a20471b561a0a3a74b377991fe1976c2ea83d7/drivers/kscan/kscan_ft5336.c#L103) and other [input](https://github.com/zephyrproject-rtos/zephyr/blob/60a20471b561a0a3a74b377991fe1976c2ea83d7/drivers/kscan/kscan_cap1203.c#L99) devices.

Meanwhile devices that didn't quite fit that model somehow found their way into other subsystems, for example the "nrfx_qdec" driver, commonly used for scroller encoders ended up as a sensor rotation device, meaning that the "clicks" gets an [angle conversion](https://github.com/zephyrproject-rtos/zephyr/blob/60a20471b561a0a3a74b377991fe1976c2ea83d7/drivers/sensor/qdec_nrfx/qdec_nrfx.c#L104-L109), just to convert the angle [back to clicks](https://github.com/nrfconnect/sdk-nrf/blob/f3d31cc926d75ef5933259c2e998d34794a225bc/applications/nrf_desktop/src/hw_interface/wheel.c#L104-L106) in the application.

Lastly, input devices that would not fit any subsystem started to declare [their own API](https://github.com/zephyrproject-rtos/zephyr/blob/60a20471b561a0a3a74b377991fe1976c2ea83d7/include/zephyr/drivers/gpio_keys.h#L32-L36) as if they were a subsystem on their own, with more proposals for [device specific APIs](https://github.com/zephyrproject-rtos/zephyr/pull/51493/files) or very [specific functionalities](https://github.com/zephyrproject-rtos/zephyr/pull/48342/files#r1058390891).

### Requirements

- Support a many-to-many model, that is multiple input devices can send events to many different listeners, use iterable sections for registering the listeners for modularity
- Be flexible enough to support any input device (buttons, keyboards, mice, touchscreens etc...) without bloating the API (no specialized functions)
- Work as a replacement for the current kscan API, provide a backward compatibility layer, be usable in the current drivers, with current applications
- Provide a way to extend the basic functionalities (long press, double click...) in a generic way

### Proposed change

Implement a Zephyr input subsystem inspired by the Linux input events model.

Linux supports input devices using an input event subsystem. Each device [registers itself](https://lxr.missinglinkelectronics.com/linux/drivers/input/joystick/psxpad-spi.c#L362) with the system (causing the "/dev/inputX" node to be created among other things), sets the supported [capabilities](https://lxr.missinglinkelectronics.com/linux/drivers/input/joystick/psxpad-spi.c#L313), and then can start sending [events](https://lxr.missinglinkelectronics.com/linux/drivers/input/joystick/psxpad-spi.c#L229) whenever available.

An event describes a single entity within the input device, may that be a button, a relative or absolute movement over a single axis or even a [scancode](https://lxr.missinglinkelectronics.com/linux/drivers/input/keyboard/cros_ec_keyb.c#L195)). 

Each burst of events is terminated by a [synchronization marker](https://lxr.missinglinkelectronics.com/linux/drivers/input/joystick/psxpad-spi.c#L279), which is used to indicate that the last set of events is in a stable state and can be used. This allows representing for example an X-Y coordinate and multi-tracking for a touchscreen device.

## Detailed RFC

The proposed Zephyr implementation is conceptually similar to the Linux one, but somewhat simplified with the idea that a Zephyr application would be more resource constrained, but also be exposed to a lower number of devices and listeners.

There's no special device registration and no capabilities, but the device identifier is passed down with the event, which would looks like this:

```c
struct input_event {
        const struct device *dev;
        uint16_t type; /* key, relative, absolute... */
        uint16_t code; /* button ID, axis name (x/y/z)... */
        int32_t value; /* 0/1 for keys, coordinate for axis... */
};
```

Events are either handled by a single message queue and a dedicated "input" thread, or no queue at all, in which case listeners are called synchronously. This allows simple use cases to just get all input events in the system, or complex use cases to register a synchronous notifier and then pass that to their own event system.

Listeners registers a callback in an iterable section and can optionally request to get notified for events of only a specific device:

```c
static void input_cb(struct input_event *evt, bool sync)
{
        /* get the events for all devices */
}
INPUT_LISTENER_CB_DEFINE(NULL, input_cb);

static void buttons_cb(struct input_event *evt, bool sync)
{
        /* just get events for the buttons device */
}
INPUT_LISTENER_CB_DEFINE(DEVICE_DT_GET(DT_NODELABEL(buttons)), buttons_cb);
```

Device drivers can send events in the queue at any time, the format and identifiers are similar to the Linux ones

```c
if (pressed) {
[... 89 lines cut ...]

Followup would be the compatibility adapter, migrating some drivers (gpio-keys, the various EC keyboard scan, ft5336...) out of their current subsystem into a new "drivers/input" directory on the new API and adding some filters as an example.

Ultimately it should be possible to deprecate and drop the kscan API.

Further improvements may include:

- Shell commands for event dumping, debugging, usage stats
- An API to enable/disable the input device (is DEVICE_PM enough for that?)
- Document a suggested way of implementing vendor specific event (so that upstream does not break it), may be enough to define a "INPUT_EV_VND" code

### Concerns and Unresolved Questions

- How much devicetree should we use in this? Does it make sense to have commonly used "filters" (longpress, triple click...) represented by a device?
- Should we try to track Linux set of type/code or come up with our own?
- Does it make sense to have embedded sync messages? It saves a message but makes some situation a lot harder, like cycling through keys and syncing on the last one that changed.
- What driver should we convert? Are sensor input devices or sensors (qdec for example).

## Alternatives

Keep doing per-device APIs and abuse the current ones. There's enough of those already that PRs are starting to be pushed back with that as a reason.

Use an application specific input system out-of-tree. That's what is currently done already, ZMK for example has an application wide event system which includes [keycode state change](https://github.com/zmkfirmware/zmk/blob/main/app/include/zmk/events/keycode_state_changed.h) events but does a lot more and has more features. There's nothing wrong with that and it should be possible for the project to upstream their [keyboard](https://github.com/zmkfirmware/zmk/blob/main/app/drivers/kscan/kscan_gpio_matrix.c) and [encoder](https://github.com/zmkfirmware/zmk/blob/main/app/drivers/sensor/ec11/ec11.c) drivers to use the input APIs and then plumb that back in synchronous mode to their own event subsystem with minimal overhead.

An implementation detail worth highlighting is the modes of operation (synchronous or statically allocated queue). Both [ZMK](https://github.com/zmkfirmware/zmk/blob/c065d451cb82e2dceda4efdb2991aeeeef1cdbbf/app/include/zmk/event_manager.h#L51) and [NCS](https://github.com/nrfconnect/sdk-nrf/blob/d0bade499df9fd37d6a87f60d33f052f6f1acf10/subsys/app_event_manager/app_event_manager.c#L119) seem to be using dynamic memory in their event implementation. It may be interesting at a later stage to provide a third implementation option that uses dynamically allocated memory for the events, if there's demand for that in some particular use case.

fabiobaltieri, 2023-02-08:
cc @petejohanson @gmarull @carlescufi @mbolivar-nordic  @nordicjm @galak @dcpleung @tbursztyka (few people that I saw interacting with previous proposals)

mbolivar-nordic, 2023-02-08:
It would be nice to see some discussion on the suitability (or lack thereof) of zbus as an event dispatch mechanism for this. cc @rodrigopex

rodrigopex, 2023-02-08:
At first sight, it is possible to zbus to implement the event distribution for the Input Subsystem. I looked at the `input.h` and `input.c`, and I could see that part zbus would be helpful. 

This change is probably simple (I can help with that), and the listener would be a zbus listener instead of a different listener for inputs. Another topic is whether all the events would be distributed at the same channel or separately based on the configuration files (dts, for example).

The `struct input_event` can be seen as a message transmitting all the input events or some of that in a channel (or different channels).

```c
struct input_event_msg { 	 
const struct device *dev; 	 
uint16_t type; 	 
uint16_t code; 	 
int32_t value; 	 
};
```
I need to go deep to propose something more accurate. However, my first impression is the input subsystem is reimplementing part of the zbus.

Zbus does not use internal message queues to store events. The `input_event(...)` function can keep pushing to the event queue for being processed by the input thread when INPUT_THREAD is enabled and could publish directly to the channel when it is not. The input thread could consume events from the queue and publish them to the channel.

rodrigopex, 2023-02-08:
> An implementation detail worth highlighting is the modes of operation (synchronous or statically allocated queue). Both [...]

Zbus allocates the "events"(messages, in fact) statically by default. However, you can also allocate that dynamically. So I guess zbus solve your needs.

rodrigopex, 2023-02-08:
> Linux supports input devices using an input event subsystem. Each device [registers itself](https://lxr.missinglinkelectronics.com/linux/drivers/input/joystick/psxpad-spi.c#L362) with the system [...]

You could create a channel for each `/dev/inputX` with something like `zdev_input1` `zdev_input2`, and so forth. Then, the input subsystem will publish events to the respective channel.

mbolivar-nordic, 2023-02-08:
> * How much devicetree should we use in this?

IMO as much as you want. Given the complexity of your examples, it seems like the only alternative is a caller-allocated configuration structure that gets passed to the subsystem/driver layers at initialization time, right? How practical is that as an alternative?

> Does it make sense to have commonly used "filters" (longpress, triple click...) represented by a device?

I just want to point out that a devicetree node does not have to correspond to a device here, of course, so this is a separate question from whether it should be present in the DT.

fabiobaltieri, 2023-02-09:
> IMO as much as you want. Given the complexity of your examples, it seems like the only alternative is a caller-allocated configuration structure that gets passed to the subsystem/driver layers at initialization time, right? How practical is that as an alternative?

Cool, yeah looking at stuff like `io-channels` it seems like having more config in devicetree may be the way to go. Mainly concerned with having more `struct device` allocated for software features but it looks like they don't eat that much flash space these days (24 bytes on my test build? guess getting the PM stuff out helped a lot)

mkschreder, 2023-02-12:
I think having things like "longpress" appear as a device tree node is overkill. This can be handled by a generic button driver by simply setting a long press delay property (zero would mean no longpress).

Regarding the way to connect this to the rest of the system, I really like the idea of an input device driver publishing events to a topic and then other drivers subscribing to that topic. Such a mechanism would eliminate a great deal of duplication associated with registering callbacks in other drivers as well.

fabiobaltieri, 2023-02-13:
> I think having things like "longpress" appear as a device tree node is overkill. This can be handled by a generic button driver by simply setting a long press delay property (zero would mean no longpress).

I think there's value in keeping input driver barebones (generating the minimum amount of events) and having further processing separately, then you could apply the same code to any driver that generates a "button" (imagine touchscreen presses, maybe drivers for capacitive buttons, keyboard scanners etc...). If anything it should be in the application code, but something like this I could see being in the main code base, if anything to have it as a reference.

albertofloyd, 2023-02-16:
@fabiobaltieri  is the ultimate goal to deprecate the KSCAN API and convert all the KSCAN drivers in favor of new API? 
Or just the ones currently misusing it? 

Will the backward compatibility for current applications using KSCAN API be temporary or once this in its final version that option won't be available.

> can be used to provide a driver that provides backward compatibility with the current "kscan" API:
> Ultimately it should be possible to deprecate and drop the kscan API.

fabiobaltieri, 2023-02-16:
> @fabiobaltieri is the ultimate goal to deprecate the KSCAN API and convert all the KSCAN drivers in favor of new API? Or just the ones currently misusing it?

The new API should cover keyboard scan devices just fine, so yeah I was thinking about deprecating kscan and converting the drivers and in-tree users (only lvgl as far as I'm aware of).

> Will the backward compatibility for current applications using KSCAN API be temporary or once this in its final version that option won't be available.

I don't see any issue keeping it in the code base as long as necessary.

albertofloyd, 2023-02-16:
@fabiobaltieri I think is better to keep it since it's stable enough. And focus in cleanup the abuse 
Let me go through  https://github.com/zephyrproject-rtos/zephyr/pull/54908 https://github.com/zephyrproject-rtos/zephyr/pull/54620 to see how a potential coexistence without the removal but also without the driver duplication can be work out.

nordicjm, 2023-03-24:
@fabiobaltieri I guess this can be closed now?

fabiobaltieri, 2023-03-24:
> @fabiobaltieri I guess this can be closed now?

Yeah, my plan was to close it after merging https://github.com/zephyrproject-rtos/zephyr/pull/55556 https://github.com/zephyrproject-rtos/zephyr/pull/55787 and https://github.com/zephyrproject-rtos/zephyr/pull/56128 since I touched on those features as part of the core proposal, but the reviews are taking ages so we may as well close this now.

closed 2023-03-24 by fabiobaltieri