zephyrproject-rtos/zephyr #54630: memcpy crashes with NEW_LIBC

jchabod, 2023-02-08:
**Describe the bug**
memcpy crashes due to unaligned access under certain circumstances (Fully reproductible)

My `BOARD` is a ARM7  more specificly a `stm32h747i_disco_m7`.
I am currently based a few commits above ZEPHYR `v3.3.0-rc2`.

The issue only appears when `CONFIG_NEWLIB_LIBC=y`.
It is reproductible by copying a 3-bytes buffer from even addresses.

**To Reproduce**
Compile this simple program on any STM32 board. (Personnaly I just edited "hello_world" sample)
```
#include <zephyr/kernel.h>
#include <string.h>

void main(void)
{
	printk("Hello World! %s\n", CONFIG_BOARD);

	char buff[256];

	for (size_t i = 0 ; i< 256; i++)
	    buff[i] = i;
	{
	    char* src = &buff[0x10];
	    char* dst = &buff[0x82];
	    const size_t len = 3;
	    printk("MemCpy(0x%p, 0x%p, %d)\n", src,dst,len);
	    memcpy(src, dst, len);
        printk("OK\n");
	}
}
```

And in prj.conf:
```
CONFIG_NEWLIB_LIBC=y
```

**Expected behavior**
"OK" line is not displayed, an exception is raised in memcpy at this point :
[image]

The First odd byte is copied correctly thus bringing src (R0) and dst (R1) to odd addresses. The `ldrh` raises an exception due to unaligned address

**Impact**
Blocking (no workaround because it actually crashes in ring buffers from uart in my project)
Moreover, I think there is another issue (in ARM7 fault management), because when activated, the "BUS FAULT"  report on console does not display the correct fault address. It took me avery long time using the STLINK to figure out that the actual fault address exists in the AMR register but that it is not the right one which is displayed (I guess I have to createanother issue for that one?)

**Logs and console output**
```
*** Booting Zephyr OS build v3.3.0-rc2-10-g51d526408a5d ***
Hello World! stm32h747i_disco_m7
MemCpy(0x0x240011b8, 0x0x2400122a, 3)

```

**Environment (please complete the following information):**
 - OS: Windows
 - `ZEPHYR_TOOLCHAIN_VARIANT=zephyr`
 - `ZEPHYR_SDK_INSTALL_DIR=D:\Z\zephyr-sdk-0.15.2`
 - Zephyr v3.3.0-rc2
 - `BOARD=stm32h747i_disco_m7`

nordicjm, 2023-02-09:
This seems like a bug in the STLINK wrongly enabling the unaligned memory trap, the M7 (like the M4) should be able to handle unaligned memory reads and writes without issue. Assigning to @erwango

jchabod, 2023-02-09:
I was using ST-LINK(ST-LINK GDB Server)
 which displays:
```
STMicroelectronics ST-LINK GDB server. Version 6.0.0
Copyright (c) 2021, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 1
        Listen Port Number         : 61234
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled
        InitWhile                  : Enabled
```

Now trying with ST-LNK (OpenOCD) leads to the same effect

jchabod, 2023-02-13:
Note:
It seems that https://github.com/zephyrproject-rtos/zephyr/issues/54670 does **not** solve the debugger problem.

jchabod, 2023-02-13:
> HI @jchabod I try to reproduce the issue but i can't for the moment ( all work well on my side) [Doc1.docx](https://github.com/zephyrproject-rtos/zephyr/files/10699189/Doc1.docx)
> 
> Could you tell me the version of cube IDE you use please ?

V1.8.0

closed 2023-04-29 by github-actions[bot]