zephyrproject-rtos/zephyr #54649: eth_stm32: allow MAC address to be set from application before interface goes up

arbrauns, 2023-02-09:
**Is your enhancement proposal related to a problem? Please describe.**
The application I'm writing stores the ethernet MAC address in EEPROM, which means it can't be specified using the `local-mac-address` device tree property. The best way I've to do this at all so far is this:

```
net_if_down(iface);

struct ethernet_req_params params = /* load from EEPROM */;
net_mgmt(NET_REQUEST_ETHERNET_SET_MAC_ADDRESS, iface, &params, sizeof(params));

net_if_up(iface);
```

However, this still causes the interface to use its default MAC address (derived from device ID) during startup, and a couple frames go out with this address.

**Describe the solution you'd like**
There should be a way to set the ethernet MAC address before the interface goes up for the first time, i.e. by delaying interface startup until the application says so.

**Describe alternatives you've considered**

\-

**Additional context**

\-

rlubos, 2023-02-09:
Network stack has a built-in mechanism to prevent network interface from being brough up at boot, by setting `NET_IF_NO_AUTO_START` flag on the interface. This however needs to be supported by particular L2/driver in order to work, as the flag needs to be set during initialization.

Some L2s define a Kconfig to skip auto-start on boot, see for example:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ieee802154/Kconfig#L16
It can then be used during the initialization to set the flag on the interface:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ieee802154/ieee802154.c#L459

I think that similiar could be implemented for Ethernet L2. Feel free to send a PR, this should be a pretty straightforward change.

maass-hamburg, 2024-09-10:
I would recommend you to use a `SYS_INIT()`, with a priority higher than the ethernet driver and lower than `CONFIG_NET_CONFIG_INIT_PRIO`, if  you still want to use the auto init. With that so you can do the part you described above before the network subsystem brings the interfaces up.