zephyrproject-rtos/zephyr #54662: Add Board: Lilygo SIM800L AXP192

mkettn, 2023-02-09:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
I'ld like to add an board to Zephyr RTOS. It is called LilyGO T-Call-SIM800. it is an esp32-wrover-b with an SimCom SIM800L GSM Modem and PMU AXP192. cf. https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/doc/SIM800L_AXP192.MD

**Describe why you are asking for this support?**
Neither the board nor the modem in particular are already included in the Zephyr Repo. However the CPU is already supported. For the modem I'd need to take a look at the sim7080 gsm module - hence I don't have to start from scratch there. Also I want to get familar with devicetrees for Zephyr and though of it as a exercise.

Unfortunately I wont be able to maintain Zephyr support once I've got the gsm module running.

**Additional context**
[image]
SIM800 AT command reference:
https://cdn-shop.adafruit.com/datasheets/sim800_series_at_command_manual_v1.01.pdf

mkettn, 2023-02-09:
I was able to create a board specification by copying most of the stuff from esp32_wrover_kit. I added an alias to led0 for GPIO13 as I found it in the schematics. When I run the blinky example on the board the green LED gets toggled. As expected.

mkettn, 2023-02-09:
In the [Kconfig for gsm](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/Kconfig.gsm#L27) I've read that Simcom Modems are already supported:
`	  Use this if you have SIMCOM based modem like SIM800 etc.
`
Hence next I'll try the gsm module example

mkettn, 2023-02-10:
Now I ran into a problem, where I am not sure on how to solve it. ( see bottom of this message for my question)

For reference I used the [Arduino AT Debug Example](https://github.com/mkettn/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_AT_Debug/) I can send AT Commands to the SIM800L with this and get a reply.

First, the SIM800L is connected to uart1 using the GPIO_27 (tx) and GPIO_26 (rx). I extended the (formerly original [esp_wrover_kit](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/xtensa/esp_wrover_kit/esp_wrover_kit-pinctrl.dtsi)) -pinctrl.dtsi file with this entry after uart0_default:
```
	uart1_default: uart1_default {
		group1 {
			pinmux = <UART0_TX_GPIO27>;
		};
		group2 {
			pinmux = <UART0_RX_GPIO26>;
			bias-pull-up;
		};
	};
```
... and added a uart1 node to the dts file:

```
&uart1 {
	status = "okay";
	current-speed = <115200>;
	pinctrl-0 = <&uart1_default>;
	pinctrl-names = "default";
	sim800l {
		status = "okay";
		compatible = "zephyr,gsm-ppp";
	};
};
```

... this will at least link the gsm-modem sample.

However, to start the SIM800L device I also have to [fiddle with other pins](https://github.com/mkettn/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_AT_Debug/Arduino_AT_Debug.ino#L19):
```c
void setupModem()
{
#ifdef MODEM_RST
    // Keep reset high
    pinMode(MODEM_RST, OUTPUT);
    digitalWrite(MODEM_RST, HIGH);
#endif

    pinMode(MODEM_PWRKEY, OUTPUT);
    pinMode(MODEM_POWER_ON, OUTPUT);

    // Turn on the Modem power first
    digitalWrite(MODEM_POWER_ON, HIGH);

    // Pull down PWRKEY for more than 1 second according to manual requirements
    digitalWrite(MODEM_PWRKEY, HIGH);
    delay(100);
    digitalWrite(MODEM_PWRKEY, LOW);
    delay(1000);
    digitalWrite(MODEM_PWRKEY, HIGH);

    // Initialize the indicator as an output
    pinMode(LED_GPIO, OUTPUT);
    digitalWrite(LED_GPIO, LED_OFF);
}
```
Id like to tell these pins to the gsm_ppp driver so that the modem turns on when the `gsm_ppp_start(gsm_dev)` gets called (or the uC gets powered on). I know from [other modem drivers](https://docs.zephyrproject.org/latest/build/dts/api/bindings/modem/swir,hl7800.html) that one can add these pins to the node in the device tree. However it doesn't seem that I can tell this to the gsm_ppp driver. it just expects it to be already turned on lol. Setting `CONFIG_MODEM_GSM_SIMCOM=y` in [prj.conf](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/gsm_modem/prj.conf) does not help either - imho the help text is misleading too, since I have a sim800l and it is not working after I turned that on. And I think this does only do some mumbo jumbo during muxing (see [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/gsm_ppp.c#L902)), when the device is already operational.

- Do I have to write my own modem driver to turn on the uart device during startup or do I just miss something?
- In case I have to, how do I do this to get just PDP working? (the modem also supports TCP/IP offloading, but I wont get into that for now)
- If so, how can I add and use node specific properties? e.g. inside the &uart1 node:
```
	sim800l {
		status = "okay";
		compatible = "simcom,sim800";
// I am not sure if I have to set them open drain. just c&p this from another example
		mdm-reset-gpios = <&gpio0 5 (GPIO_OPEN_DRAIN | GPIO_ACTIVE_LOW)>;
		mdm-pwr-on-gpios = <&gpio0 23 (GPIO_OPEN_DRAIN | GPIO_ACTIVE_LOW)>;
	};
```

Help would be appreciated, many thanks!

ihora, 2023-03-25:
Dear sir,

I am also facing problems with SIMCOM 800L. Actually I am trying to run the modem on nrf52840_dongle.
I am not skilled in modems much so maybe I do not understand your problem correctly, but I was also simillar Arduino examples.
I personally think, to operate the SIMCOM 800L properly, one needs to write a Zephyr driver simmilar to the driver sim7080.

But if you want to achieve this:
pinMode(MODEM_RST, OUTPUT);
    digitalWrite(MODEM_RST, HIGH);
I guess it is just a modem setup which could be done before starting ppp, isn't is possible just to manually write voltage value to pin with method?:
 ret = gpio_pin_configure_dt();

Actually, I would like to kindly ask you, how did you manage to send AT commands to the modem via Zephyr since I unable to acomplish this simple task.

My goal for the start is really simple, I want to send sms from modem. Since I did not find the driver for SIMCOM 800L, I checked this file as example:
 https://github.com/zephyrproject-rtos/zephyr/blob/a1f55c63b37dd642733510364186184a3bfa5861/drivers/modem/simcom-sim7080.c

Since there is an AT command in SIMCOM documentation file, my intention was to create and properly send specific AT command (the one responsible for sending SMS) to SIMCOM module.
But how to do that in Zephyr code? There is a certain cmd_handler in:
https://github.com/zephyrproject-rtos/zephyr/blob/a1f55c63b37dd642733510364186184a3bfa5861/drivers/modem/simcom-sim7080.c
which is responsible for sending AT commands to sim7080 so I guess reconstructing the code responsible for sending AT commands would probably be sufficient to operate the module with AT commands only.
But this is way too much difficult for me and I think this looks like writing the driver for SIMCOM 800L from scratch anyway.

mkettn, 2023-03-27:
Concerning the GPIOs it is very eas, once you understood how one can define device properties. You can copy this file https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/modem/simcom%2Csim7080.yaml and add the gpios mdm-power-gpios and mdm-pwrkey-gpios properties (similar to the symbols from the setupModem() routine, see my [previous post](https://github.com/zephyrproject-rtos/zephyr/issues/54662#issuecomment-1425833596)).
Using the device tree macros we can then access these properties and get either the modem node or the parent node (which is the uart device):

```c
#define MODEM_DEVICE DT_COMPAT_GET_ANY_STATUS_OKAY(simcom_sim800)
#define UART_DEVICE DT_PARENT(MODEM_DEVICE)
static const struct device *const gsm_dev = DEVICE_DT_GET(UART_TO_DEVICE);
struct gpio_dt_spec mdm_power =
	GPIO_DT_SPEC_GET(MODEM_DEVICE, mdm_power_gpios);
struct gpio_dt_spec mdm_pwrkey =
	GPIO_DT_SPEC_GET(MODEM_DEVICE, mdm_pwrkey_gpios);
```

right now I did not develop a modem driver but rather access the modem from within a small application which just turns on the modem and sends some AT commands to the device thru the UART API. I do this to get familar with the modem.

ihora, 2023-03-27:
Regarding this:
... But does this mean, I can create a tcp/ip connection to a server using the Zephyr modem driver and Zephyrs tcp/ip stack?

If I understand this correctly. If I would like to connect to the internet via SIMCOM 800L via GPRS utilizing this modem under Zephyr, I would use the example https://docs.zephyrproject.org/latest/samples/net/gsm_modem/README.html#gsm-modem-sample. But this example uses generic modem driver which I understand almost any modem could use this driver. But the internet connection is also using another layer called PPP. I guess this is also very popular in Linux kernel as any standard modem can use PPP to send tcp/ip packets over PPP encapsulation no matter if it is USB or UART modem, but tcp/ip packets cannot be just so simply transferred over UART as they must be encapsulated with PPP.
But I guess PPP sits on top of sending AT command, maybe AT commands initialize GPRS connections and then packets are transferred over PPP, but I guess it is possible to send SMS just by giving AT commands to modem without PPP as there are no tcp/ip packets.
I guess connecting to the internet would be much easier for me as there already is an example gsm_modem, but there is no example sending simple AT commands for example to send SMS.

Regarding the hardware connection and GPIO voltage levels, I presumed I will only copy the sequence which is mentioned for example in your tutorial:
https://github.com/mkettn/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_AT_Debug/Arduino_AT_Debug.ino
But according to your speech it sounds far more complicated.
I assumed the Zephyr gsm_modem example does everything by default, but I never tested it since my goal was different.

mkettn, 2023-03-30:
> But according to your speech it sounds far more complicated.

I just wrote down my thoughts about the modem driver. As I said before I am new to this too. Also I tend to overthink things before breaking them down into simple solutions ;)

ihora, 2023-04-01:
:)) That's kind of a engineering curse, I used to have the same way of thinking.
Anyway, I tried to connect the modem to my nrf52,compile the Zephyr SDK modem example and run it. 
It compiles very fine, it also runs and seems to do something, the left diode shines red and the right one is blinking red. Above the diode is written "NET" so I guess the modem is not connected to cellular network.
No I have to dive into debugging of the code.
I also found a simple AT command example on github which should send a simple AT command.
Just in case, here is the example if you cannot find it:
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/gsm_modem
I would give it a try in Linux but but these days it is hard to get Raspberry anywhere around so there is no motivation for me to suit it to Raspberry, if only one day there will be a board purely open-source anyone can manufacture or easily available, maybe there is, I just do not know about it yet but seems the older I am the more naive this attitude seems to me.

ihora, 2023-04-02:
Just in case, here is the Zephyr tiny example to send single AT command:
https://github.com/gfnord/stm32_lte
I got stuck on activating UART. 
ay_uart_driver_open() equals 1 
 This: https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/gsm_modem also does not work for me

ihora, 2023-04-02:
No matter what I do, I cannot initialize UART0:
    modem_dev = device_get_binding(MODEM_UART);

prj. conf:
CONFIG_GPIO=y
#CONFIG_RTC=y
#CONFIG_RTC_STM32=y
#CONFIG_UART_STM32=y
#CONFIG_UART_1=y
##
CONFIG_SERIAL=y
CONFIG_MODEM=y
CONFIG_UART_NRFX=y
CONFIG_UART_INTERRUPT_DRIVEN=y
#CONFIG_SYS_LOG=y
CONFIG_NRFX_RTC0=y

mkettn, 2023-04-03:
Hello ihora,

> No matter what I do, I cannot initialize UART0:

I am not familar with nrf52840dongle_nrf52840, but as far as I know how things in zephyr work UART0 is hard wired to the serial console you can hook up to from your host machine, not the modem. You'ld need a second uart connection to communicate with the modem. From the [product spec](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52840%2Fkeyfeatures_html5.html) of your dongle it reads: "2x UART (CTS/RTS) with EasyDMA" so a second uart connection is possible, however no uart1 is configured in your device tree file, see https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840-pinctrl.dtsi#L7 maybe ask a nordic developer on how to set up a second UART to communicate to your modem - however, this is not related to the board in this thread: LilyGO SIM800L AXP192 on which a espressif esp32 (different µC!) communicates with the SIM800L modem.

>I also deactivated everything related to UART in :
>./zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840.dts

if you turn off uart, it will not work ;) Also I'd recommend that you modify the device tree from withing your application (app.overlay or boards/<board>.overlay), instead of modifying the dts of the board in the zepyhr repo. see here: https://docs.zephyrproject.org/latest/develop/application/index.html#overview

in the app.overlay file you would add something like:
```
&uart1 {
	status = "okay";
	current-speed = <115200>;
	pinctrl-0 = <&uart1_default>;
	pinctrl-names = "default";
	sim800l {
		status = "okay";
		compatible = "zephyr,simcom-sim800l";
	};
};
```
you can then find the sim800l device using device tree macros:
```c
#define MODEM_DEVICE DT_COMPAT_GET_ANY_STATUS_OKAY(simcom_sim800)
```
... and the uart device:
```c
#define UART_DEVICE DT_PARENT(MODEM_DEVICE)
static const struct device *const gsm_dev = DEVICE_DT_GET(UART_TO_DEVICE);
```
see my previous post: https://github.com/zephyrproject-rtos/zephyr/issues/54662#issuecomment-1484933576

Unfortunately at first the whole zephyr device driver stuff looks very complex, hence you have a steep learning curve until you get stuff done. But it is necessary since zephyr wants to support alot different microcontrollers, periphery and boards without changing the application code all the time.

ihora, 2023-05-01:
Hello,

I finally managed to send SMS from SIM800L paired with NRF52840 DONGLE running on vanilla Zephyr (not NRF sdk). I did not do any GPIO voltage level initial sequence. The only thing I have to make sure is that the modem starts and connects to the network before the dongle starts the program otherwise it works. I do not take into account any sleep functionality that is beyond my current skill.

closed 2023-07-16 by github-actions[bot]