zephyrproject-rtos/zephyr #54670: memcpy crashes with NEWLIBC

jchabod, 2023-02-09:
**Describe the bug**
I use LVGL on top of `https://github.com/dariis/zephyr/tree/stm32h7-mipi-dsi` (a fork on top of v3.3.0-rc2 to support DISPLAY on STM32H747-DK)

Please note that this is very close to [54630](https://github.com/zephyrproject-rtos/zephyr/issues/54630) but not under the same conditions.
I have the same issue: memcpy crashes due to unaligned access under certain circumstances. However, contrary to #54630 it happens even without the STLINK.

For the moment I could not create a simple reproducer, since it only happens when my whole program is present. I'll try to do that.

The bad news is that due to #54630 I am not able to ue debugger to provde detailled informations. All I can say is that:
- The issue only appears when `CONFIG_NEWLIB_LIBC=y`.
- The issue happens in memcpy.

I found a workaround by using `-Wl,--wrap=memcpy` and implementing the "stupid" version on memcpy (partially copied from Zephyr implementation in "minimal")
```c

/* Workaround for unaligned memory access */
void * __wrap_memcpy(void *ZRESTRICT d, const void *ZRESTRICT s, size_t n)
{
    unsigned char *d_byte = (unsigned char *)d;
    const unsigned char *s_byte = (const unsigned char *)s;

    /* do byte-sized copying until finished */

    while (n > 0) {
        *(d_byte++) = *(s_byte++);
        n--;
    }
    return d;
}
```

** Notes **
I have instrumented thread stacks and I can quite surely say that this is not due to a thread stack overflow. (However, there remain the "MAIN" stack size that I did not check. Is there any way to do that simply with `CONFIG_xxxx` ?

**Impact**
(Low, I found a workaround)

**Logs and console output**
```
[00:00:05.285,000] <err> os: ***** USAGE FAULT *****
[00:00:05.285,000] <err> os:   Unaligned memory access
[00:00:05.285,000] <err> os: r0/a1:  0xd010dd49  r1/a2:  0x2400b1d5  r2/a3:  0x80000000
[00:00:05.285,000] <err> os: r3/a4:  0x0000f6f6 r12/ip:  0xd010dd18 r14/lr:  0x08019c9d
[00:00:05.285,000] <err> os:  xpsr:  0xa1030000
[00:00:05.285,000] <err> os: Faulting instruction address (r15/pc): 0x080018dc
[00:00:05.285,000] <err> os: >>> ZEPHYR FATAL ERROR 31: Unknown error on CPU 0
[00:00:05.285,000] <err> os: Current thread: 0x24007aa8 (DEMO)
[00:00:05.331,000] <err> os: Halting system
```
 Map file:
```
D:/Z/zephyr-sdk-0.15.2/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp\libc.a(lib_a-memcpy.o)
                0x00000000080017fc                memcpy
 .text          0x0000000008001930       0x50 
```

I cannot run wih debugger, but I can disassemble at faulty address (`0x080018dc`), which happens to be quiet close to #54630 because this corresponds to the store of half-word with odd address (dst= `r0`= `0xd010dd49`).
```asm
          memcpy:
080017fc:   mov     r12, r0
080017fe:   orr.w   r3, r1, r0
08001802:   ands.w  r3, r3, #3
08001806:   bne.n   0x80018e4 <memcpy+232>
08001808:   subs    r2, #64 ; 0x40
0800180a:   bcc.n   0x8001890 <memcpy+148>
0800180c:   ldr.w   r3, [r1], #4
08001810:   str.w   r3, [r0], #4
08001814:   ldr.w   r3, [r1], #4
08001818:   str.w   r3, [r0], #4
0800181c:   ldr.w   r3, [r1], #4
08001820:   str.w   r3, [r0], #4
08001824:   ldr.w   r3, [r1], #4
08001828:   str.w   r3, [r0], #4
0800182c:   ldr.w   r3, [r1], #4
08001830:   str.w   r3, [r0], #4
08001834:   ldr.w   r3, [r1], #4
08001838:   str.w   r3, [r0], #4
0800183c:   ldr.w   r3, [r1], #4
08001840:   str.w   r3, [r0], #4
08001844:   ldr.w   r3, [r1], #4
08001848:   str.w   r3, [r0], #4
0800184c:   ldr.w   r3, [r1], #4
08001850:   str.w   r3, [r0], #4
08001854:   ldr.w   r3, [r1], #4
08001858:   str.w   r3, [r0], #4
0800185c:   ldr.w   r3, [r1], #4
08001860:   str.w   r3, [r0], #4
08001864:   ldr.w   r3, [r1], #4
08001868:   str.w   r3, [r0], #4
0800186c:   ldr.w   r3, [r1], #4
08001870:   str.w   r3, [r0], #4
08001874:   ldr.w   r3, [r1], #4
08001878:   str.w   r3, [r0], #4
0800187c:   ldr.w   r3, [r1], #4
08001880:   str.w   r3, [r0], #4
08001884:   ldr.w   r3, [r1], #4
08001888:   str.w   r3, [r0], #4
0800188c:   subs    r2, #64 ; 0x40
0800188e:   bcs.n   0x800180c <memcpy+16>
08001890:   adds    r2, #48 ; 0x30
08001892:   bcc.n   0x80018b8 <memcpy+188>
08001894:   ldr.w   r3, [r1], #4
08001898:   str.w   r3, [r0], #4
0800189c:   ldr.w   r3, [r1], #4
080018a0:   str.w   r3, [r0], #4
080018a4:   ldr.w   r3, [r1], #4
080018a8:   str.w   r3, [r0], #4
080018ac:   ldr.w   r3, [r1], #4
080018b0:   str.w   r3, [r0], #4
080018b4:   subs    r2, #16
080018b6:   bcs.n   0x8001894 <memcpy+152>
080018b8:   adds    r2, #12
080018ba:   bcc.n   0x80018c8 <memcpy+204>
080018bc:   ldr.w   r3, [r1], #4
080018c0:   str.w   r3, [r0], #4
080018c4:   subs    r2, #4
080018c6:   bcs.n   0x80018bc <memcpy+192>
080018c8:   adds    r2, #4
080018ca:   beq.n   0x80018de <memcpy+226>
080018cc:   lsls    r2, r2, #31
080018ce:   itt     ne
080018d0:   ldrbne.w        r3, [r1], #1
080018d4:   strbne.w        r3, [r0], #1
080018d8:   bcc.n   0x80018de <memcpy+226>
080018da:   ldrh    r3, [r1, #0]
080018dc:   strh    r3, [r0, #0]
080018de:   mov     r0, r12
080018e0:   bx      lr
080018e2:   nop     
080018e4:   cmp     r2, #8
080018e6:   bcc.n   0x8001910 <memcpy+276>
080018e8:   lsls    r3, r1, #30
080018ea:   beq.n   0x8001808 <memcpy+12>
080018ec:   ands.w  r3, r0, #3
080018f0:   beq.n   0x8001808 <memcpy+12>
080018f2:   rsb     r3, r3, #4
080018f6:   subs    r2, r2, r3
080018f8:   lsls    r3, r3, #31
080018fa:   itt     ne
080018fc:   ldrbne.w        r3, [r1], #1
08001900:   strbne.w        r3, [r0], #1
08001904:   bcc.n   0x8001808 <memcpy+12>
08001906:   ldrh.w  r3, [r1], #2
0800190a:   strh.w  r3, [r0], #2
0800190e:   b.n     0x8001808 <memcpy+12>
08001910:   subs    r2, #4
08001912:   bcc.n   0x80018c8 <memcpy+204>
08001914:   subs    r2, #1
08001916:   ldrb.w  r3, [r1], #1
0800191a:   strb.w  r3, [r0], #1
0800191e:   bcs.n   0x8001914 <memcpy+280>
08001920:   ldrb    r3, [r1, #0]
08001922:   strb    r3, [r0, #0]
08001924:   ldrb    r3, [r1, #1]
08001926:   strb    r3, [r0, #1]
08001928:   ldrb    r3, [r1, #2]
0800192a:   strb    r3, [r0, #2]
0800192c:   mov     r0, r12
0800192e:   bx      lr
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - `ZEPHYR_TOOLCHAIN_VARIANT=zephyr`
 - `ZEPHYR_SDK_INSTALL_DIR=D:\Z\zephyr-sdk-0.15.2`
 - Zephyr v3.3.0-rc2
 - `BOARD=stm32h747i_disco_m7`

stephanosio, 2023-02-09:
> memcpy crashes due to unaligned access under certain circumstances

Unaligned half word store should not generate an exception under normal circumstances for Cortex-M7.

It looks like the memory region you are writing to via `memcpy` is configured as "Device" memory type at the hardware level and that is causing the unaligned access fault -- see https://developer.arm.com/documentation/ka002886/latest

erwango, 2023-02-09:
Thanks @stephanosio for the pointer.

@jchabod can you try the following, which should change MPU attribute for SDRAM2:

In board.dts:
```
	sdram2: sdram@d0000000 {
		compatible = "zephyr,memory-region", "mmio-sram";
		device_type = "memory";
		reg = <0xd0000000 DT_SIZE_M(32)>;
		zephyr,memory-region = "SDRAM2";
+		zephyr,memory-region-mpu = "RAM_NOCACHE";
	};
```

jchabod, 2023-02-13:
I made this modificaiton in `stm32h747i_disco_m7.dts` .

Thanks, this seems OK.
However, I hoped that would also solve #54630, but it does not...

jchabod, 2023-02-21:
Please, can you confirm that the "RAM_NOCACHE" parameter won't have horrible impact on performances ? In particular I moved  MbedTLS heap in SDRAM2 due to low RAM issue in my configuration.

erwango, 2023-02-22:
> Please, can you confirm that the "RAM_NOCACHE" parameter won't have horrible impact on performances ? In particular I moved MbedTLS heap in SDRAM2 due to low RAM issue in my configuration.

This depends on your application and usage. The available memory regions attributes are described in `dts/bindings/base/zephyr,memory-region.yaml` binding:
```
  zephyr,memory-region-mpu:
    type: string
    enum:
      - "RAM"
      - "RAM_NOCACHE"
      - "FLASH"
      - "PPB"
      - "IO"
```
By default, "IO" was applied and that was the root cause of your issue.
Up to you to find the best alternative for your own need.

closed 2023-03-03 by carlescufi

jchabod, 2023-04-24:
I created a patch in MBEDTLS, and proposed it in ZEPHYR dedicated thread under discord, but had no feedback

https://discord.com/channels/720317445772017664/884102502021210152/1075821863701336064

Does this answer your question?

bonetor, 2023-04-24:
It answers it, thanks! 

I`m doing it in the same way but I get the "Unaligned memory access" error during the SSL handshake . Strange...

bonetor, 2023-04-25:
Yes I did, but it did not solve the issue. 

I am currently using Zephyr v3.1.0. Maybe I must update to Zephyr v3.3.0.

PR #55079 (2023-03-03): boards: stm32f7/stm32h7: Configure external SDRAM region with RAM att…
…ribute

On M7 based STM32 boards, external SDRAM may be present and mapped in the area 0xC0000000-0xDFFFFFFF.
According to ARMv7-M Architecture Reference Manual chapter B3.1 (table B3-1), this area is specified as Device Memory Type, with the consequence that all accesses should be naturally aligned, otherwise a hard fault will be triggered whatever UNALIGN_TRP status in the CCR reg.

To avoid this issue, when available, configure external SDRAM memory regions as a MPU with RAM attribute.

Fixes #54670
commit c9b57cdf (2023-03-03): boards: stm32f7/stm32h7: Configure external SDRAM region with RAM attribute

On M7 based STM32 boards, external SDRAM may be present and mapped in the
area 0xC0000000-0xDFFFFFFF.
According to ARMv7-M Architecture Reference Manual chapter B3.1
(table B3-1), this area is specified as Device Memory Type, with the
consequence that all accesses should be naturally aligned, otherwise
a hard fault will be triggered whatever UNALIGN_TRP status in the CCR reg.

To avoid this issue, when available, configure external SDRAM memory
regions as a MPU with RAM attribute.

Fixes #54670