zephyrproject-rtos/zephyr #54705: CDC USB shell receives garbage when application starts

nordicjm, 2023-02-10:
**Describe the bug**
When the shell is enabled, with serial backend, using USB CDC directly without console, at startup the application seems to receive what can only be described as complete garbage:
```
^[[m
^[[m: command not found
uart:~$ 
uart:~$ 
uart:~$ 
*** Booting Zephyr OS build v3.3.0-rc2-111-g226062396ec4 ***
[00:00:00.003,021] <inf> littlefs: LittleFS version 2.5, disk version 2.0
[00:00:00.003,204] <inf> littlefs: FS at flash-controller@4001e000:0xdc000 is 4 0x1000-byte blocks with 512 cycle
[00:00:00.003,234] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.003,540] <inf> littlefs: /lfs1 mounted
[00:00:00.006,622] <inf> smp_bt_sample: Advertising successfully started
[00:00:00.006,683] <err> smp_sample: Failed to enable USB
uart:~$ ^[[1;32mu
```

This seemingly seems to be data that the USB CDC/shell has put into the transmit buffer as it looks to be a colour escape code.

**To Reproduce**
Build samples/mgmt/mcumgr/smp_svr for nrf52840dongle_nrf52840 using USB CDC

**Expected behavior**
No garbage to be present

**Impact**
Annoyance. Possibly indicative of a larger memory issue.

**Environment (please complete the following information):**
 - Commit SHA or Version used: a5e04ccd4ab976b99f45d106e3de873fc8dd882a

jfischer-no, 2023-02-10:
It is already described and fixed here https://github.com/zephyrproject-rtos/zephyr/issues/36948
Is CONFIG_SHELL_BACKEND_SERIAL_CHECK_DTR set?
Please also note https://github.com/zephyrproject-rtos/zephyr/issues/47120

nordicjm, 2023-02-10:
`CONFIG_SHELL_BACKEND_SERIAL_CHECK_DTR` is set (it can't be disabled apparently). This is not echoed by the terminal, and it actually feeds itself back to the USB CDC port's input (verified with wireshark that when the module boots, it sends the characters then sends the command not valid response indicating that it gave input to itself without a terminal being attached)

nordicjm, 2023-02-10:
See trace here [usb.zip](https://github.com/zephyrproject-rtos/zephyr/files/10707361/usb.zip), terminal is connected (which sends nothing to the module, just receives), raw output is:
```
\1b[m

\1b[1;32muart:~$ \1b[m^[[m
\1b[1;31m^[[m: command not found
\1b[m\1b[1;32muart:~$ \1b[m
\1b[1;32muart:~$ \1b[m
\1b[1;32muart:~$ \1b[m
\1b[1;32muart:~$ \1b[m^[\1b[10D\1b[J*** Booting Zephyr OS build v3.3.0-rc2-111-g226062396ec4 ***
\1b[1;32muart:~$ \1b[m^[\1b[10D\1b[J[00:00:00.002,197] \1b[0m<inf> littlefs: LittleFS version 2.5, disk version 2.0\1b[0m
```

jfischer-no, 2023-02-10:
>CONFIG_SHELL_BACKEND_SERIAL_CHECK_DTR is set (it can't be disabled apparently).

>To Reproduce
Build samples/mgmt/mcumgr/smp_svr for nrf52840dongle_nrf52840 using USB CDC

I am not convinced of this. Please provide exact information how to reproduce.

nordicjm, 2023-02-10:
> I am not convinced of this. Please provide exact information how to reproduce.

- Get current main
- cd bootloader/mcuboot/boot/zephyr
- mkdir build
- cd build
- cmake -GNinja -DBOARD=nrf52840dongle_nrf52840 ..
- ninja
- Get https://github.com/NordicSemiconductor/pc-nrfutil/releases
- ./nrfutil-linux pkg generate --application zephyr/zephyr.hex --application-version 1 --hw-version 52 --sd-req 0x00 out.zip
- ./nrfutil-linux dfu serial -pkg out.zip -p /dev/ttyACM0
- Go to root directory where you did west init
- cd zephyr/samples/subsys/mgmt/smp_svr
- mkdir build
- cd build
- cmake -GNinja -DBOARD=nrf52840dongle_nrf52840 -DOVERLAY_CONFIG="overlay-bt.conf" '-DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE="<ROOT_DIR>/bootloader/mcuboot/root-rsa-2048.pem"' ..
- ninja
- Ensure you have the go mcumgr application
- Remove nrf52840 dongle, press and hold SW1 and insert with it still held, keep held for 2 seconds
- mcumgr --conntype serial --connstring dev=/dev/ttyACM0,mtu=2048 image upload zephyr/zephyr.signed.bin
- mcumgr --conntype serial --connstring dev=/dev/ttyACM0,mtu=2048 reset
- Open terminal on /dev/ttyACM0 and watch

jfischer-no, 2023-02-10:
> * cd bootloader/mcuboot/boot/zephyr
... 
> * cmake -GNinja -DBOARD=nrf52840dongle_nrf52840 ..

> cmake -GNinja -DBOARD=nrf52840dongle_nrf52840 -DOVERLAY_CONFIG="overlay-bt.conf" '-DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE="<ROOT_DIR>/bootloader/mcuboot/root-rsa-2048.pem"' ..

None of this images enables CONFIG_SHELL_BACKEND_SERIAL_CHECK_DTR.
Can you please try `samples/subsys/shell/shell_module/` using `cmake -DBOARD=nrf52840dongle_nrf52840 -DDTC_OVERLAY_FILE=usb.overlay -DOVERLAY_CONFIG=overlay-usb.conf ../` ?

nordicjm, 2023-02-10:
Same output (on the second ttyACM device):
```
[1;32muart:~$ ^[[m
^[[m: command not found
uart:~$ 
uart:~$ 
uart:~$ 
uart:~$ ^[
^[: command not found
```
And `CONFIG_SHELL_BACKEND_SERIAL_CHECK_DTR` is enabled for this

nordicjm, 2023-02-10:
Same output/behaviour if that Kconfig is disabled too

nordicjm, 2023-02-10:
I can't reproduce this on an nrf52840dk without mcuboot, only on the dongle, wonder if it's something to do with that but can't think why that would have any impact on it

tmon-nordic, 2023-03-22:
> See trace here [usb.zip](https://github.com/zephyrproject-rtos/zephyr/files/10707361/usb.zip), terminal is connected (which sends nothing to the module, just receives), raw output is:

The data is coming out from host. Packet 213 contains data from device `1b 5b 6d 0d 0a 0d 0a 1b` and  then hosts sends data in packet 215 and following ones. Unfortunately Wireshark CDC ACM dissector doesn't support follow stream feature and hence there is no easy way to see all the data flowing in and out on the interface.

When ttyACM is opened in terminal application the "garbage" quickly ends. However, then the output is simply read with `cat /dev/ttyACM0`, there is endless feedback where the Linux host driver sends some data to device. The stream can be stopped by running `stty -F /dev/ttyACM0 raw` in another terminal.

tmon-nordic, 2023-03-22:
This seems to be related to echoing control characters. Doing `stty -F /dev/ttyACM0 -echo` followed by `cat /dev/ttyACM0` gives expected result.

tmon-nordic, 2023-03-23:
I believe the problem is in Linux kernel actively reading IN endpoint immediately after ttyACM `open()` call before application has a chance to set TTY parameters, i.e. disable ECHO. This seems really like a race condition and the solution would be to make Linux read IN endpoint only after user-space application issues `read()`. Note that the race is between userspace `open()` and `ioctl()` and is completely unrelated to when enumeration happens. If ttyACM device is not opened, kernel does not read IN endpoint. Device has no reliable way to determine when `open()` call happens.

The workaround on device side would be to arm IN endpoint with ZLP once interface is enabled and once the host reads the ZLP add some timeout before real data can be armed. But this is really trying to solve the issue not where it exists and as such is prone to fail.

CDC ACM here is used as a "virtual COM" that does not care about baud rate and simply provides data transport mechanism. If the CDC ACM interface is connected to device that does not transmit on its own (e.g. only replies to commands received) the issue is not there at all. Also the baud rate minimizes the "race window" - in Zephyr case we can consider the baud rate to be infinite as it is not arbitrarily limited.

tmon-nordic, 2023-03-23:
The issue does not happen on Windows because it does not ECHO characters received by default.

It is not easily observable on other boards, especially on targets that transmit with real UART to JLink because:
- default UART baud rate in Linux is 9600 but Zephyr uses 115200
- JLink stays connected across target device resets (i.e. `kernel reboot cold` will not show the issue, because it is the same ttyACM instance and not a new one)

The easiest way to observe the issue on nRF52840DK is to change default baud rate to 9600, e.g. with following patch:
```diff
diff --git a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
index bc87fa7077..f6c4bb38eb 100644
--- a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
+++ b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
@@ -149,7 +149,7 @@
 &uart0 {
        compatible = "nordic,nrf-uarte";
        status = "okay";
-       current-speed = <115200>;
+       current-speed = <9600>;
        pinctrl-0 = <&uart0_default>;
        pinctrl-1 = <&uart0_sleep>;
        pinctrl-names = "default", "sleep";
```

After flashing, the steps to reproduce are:
- disconnect JLink
- stop all terminal processes that could try to auto configure tty once it becomes available (e.g. stop GTKTerm)
- connect JLink, see kernel log for ttyACM number (it is `ttyACM0` if no other devices are present)
- `cat /dev/ttyACM0` (note that this command does not disable echo but does call `open()` on ttyACM)
- press RESET button on the DK (this resets the target, leaves JLink running)
- observe endless feedback loop

To break out of the loop, disable echo using `stty -F /dev/ttyACM0 -echo`. If you call `stty` without stopping `cat` you will observe that eventually the queue will empty. Then you can send some command with e.g. `echo "help" > /dev/ttyACM0`

tmon-nordic, 2023-03-24:
Related patch in linux-usb [Re: [PATCH] cdc-acm: acm_init: Set initial BAUD to B0](https://lore.kernel.org/linux-usb/20200710095445.GS3453@localhost/#r) led me to check `SHELL_BACKEND_SERIAL_CHECK_DTR` and it seems that `SHELL_BACKEND_SERIAL_CHECK_DTR` is not working as expected.

`SHELL_BACKEND_SERIAL_CHECK_DTR` is introducting changes only in `uart_callback`. However during initialization `shell_start()` just calls `z_shell_vt100_color_set()` which after few calls goes down to `z_shell_write()` that does call `shell->iface->api->write` (`write()` function in `shell_uart.c`) which calls `irq_write()` that simply puts the data to tx ringbuf and calls `uart_irq_tx_enable()`. This arms the endpoint really early, but the data is only transmitted after Linux user-space opens ttyACM device (before host does not try to read the IN endpoint). The actual `IN` data packet response is sent **before** host sets DTR.

`SHELL_BACKEND_SERIAL_CHECK_DTR` is obviously not respected in Zephyr and it must be fixed before proceeding further on this issue.

nordicjm, 2023-03-24:
CC @jakub-uC

closed 2023-04-06 by carlescufi

xyzzy42, 2023-06-26:
> I believe the problem is in Linux kernel actively reading IN endpoint immediately after ttyACM `open()` call before application has a chance to set TTY parameters, i.e. disable ECHO. This seems [...]

That is indeed the problem.  There is an inherit race that's been in there since before POSIX.  It's not possible to configure the tty parameters until after the device is opened.  But local processing of the tty input starts at open, e.g. ECHO, ISIG.

Not doing input processing until read would be a change to POSIX semantics.  I don't think that would ever be acceptable.  Consider the situation where Linux has a serial console.  There'd be no echo back on the console until a process reads from it.

The ESP-C3 has a dedicated USB CDC ACM serial/debug device (not an external chip as used on e.g. ESP32 and not a general purpose USB end-point peripheral that uses the zephyr usb end-point stack).  It has this same problem and obviously changes the cdc acm driver aren't going to fix it.

While this has always existed, it's hard to hit on traditional RS-232 ports for a couple reasons.

One is that RS-232 transmitters (not using flow-control) don't buffer output until the receiver is ready to receive it.  This means a character must be transmitted during the small window between when the process opens the tty and when it calls ioctl() to set the desired termios parameters.

On the other hand Zephyr's CDC ACM driver buffers output indefinitely until a receiver is ready to read it.  But only in interrupt mode, in polling mode it doesn't.  This is non-bug #53319

Because the zephyr ACM driver effectively has non-optional DTR/DSR hardware flow always enabled, and because USB always gets to transmit an URB when the port is opened, data will _always_ arrive during the window between open() and ioctl().  So instead of narrow race that rarely happens it's always a problem.

Another reason this was less of problem with non-USB ports, is that termios parameters persist across open() calls.  So the very first open of /dev/ttyS0 might have echo enabled, all opens afterward will not and not have this race.  But the USB CDC tty device is removed and re-created on the host for every zephyr boot.  So it is constantly reset to the racy default of echo enabled each time.

IMHO, the right place to solve this is by fixing the POSIX race.  The no shell output until DTR feature doesn't work for anything outputting serial data that is not the shell.  And it doesn't work for other USB serial devices like the serial/debug peripheral in the ESP32-C3 or an external USB-RS232 chip.

PR #56510 (2023-04-06): shell: uart: Fix wait for DTR before TX
Fail compilation when incorrect configuration is detected, i.e. when SHELL_BACKEND_SERIAL_CHECK_DTR is set but UART_LINE_CTRL is not set.

Use periodic timer to wait for DTR instead of waiting in uart callback to prevent blocking caller workqueue and/or sleeping in ISR.

DTR check was only ever supported with interrupt driven backend so add appropriate depends on to Kconfig.

Fixes: e9f238889b1a ("shell: uart: Add waiting on DTR signal before sending data")
Fixes: #47120
Fixes: #54705
commit 59f33a60 (2023-04-06): shell: uart: Fix wait for DTR before TX

Fail compilation when incorrect configuration is detected, i.e. when
SHELL_BACKEND_SERIAL_CHECK_DTR is set but UART_LINE_CTRL is not set.

Use periodic timer to wait for DTR instead of waiting in uart callback
to prevent blocking caller workqueue and/or sleeping in ISR.

DTR check was only ever supported with interrupt driven backend so add
appropriate depends on to Kconfig.

Fixes: e9f238889b1a ("shell: uart: Add waiting on DTR signal before
sending data")
Fixes: #47120
Fixes: #54705