zephyrproject-rtos/zephyr #54713: LVGL Module File System Memory Leaks

jonathanlee-uk, 2023-02-10:
**Describe the bug**
Every time LVGL reads the file system through lvgl_fs_open(), a structure fs_file_t is allocated by malloc().
But this file structure is not released when LVGL closes the file through lvgl_fs_close().

Similar operation on directory with lvgl_fs_dir_open() & lvgl_fs_dir_close(), we can see fs_dir_t is properly malloced and released.

reference: https://github.com/zephyrproject-rtos/lvgl/blob/zephyr/zephyr/lvgl_fs.c

A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   nRF5340
 - What have you tried to diagnose or workaround this issue?
   just added free(file) to lvgl_fs_close()
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
   n/a

**To Reproduce**
Steps to reproduce the behavior:
just create multiple LVGL image objects, the bug occurs when the platform running out of heap memory as a result of file structures not being released. 

**Expected behavior**
LVGL library will no longer able to interact with file system as it couldn't open any files.
Images would becomes blank or not being displayed at all.

**Impact**
not all images on file system could be shown on screen.

**Logs and console output**
n/a

**Environment (please complete the following information):**
 - Toolchain (e.g Zephyr SDK, ...)
Nordic Connect SDK V2.2.0
- Zephyr
VERSION_MAJOR = 3
VERSION_MINOR = 2
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =

**Additional context**
n/a

kartben, 2023-03-16:
Hi @jonathanlee-uk , and thanks for the bug report! Would you be willing to submit a PR with your fix?

kartben, 2023-04-04:
Ping @jonathanlee-uk  -- any chance you'd be able to work on a pull request, as it looks like this should be a reasonably trivial fix? Thanks!

jonathanlee-uk, 2023-04-05:
Sorry for the delay, I've created a PR. Thanks!

faxe1008, 2023-08-20:
Issue is still present, PR needs to be resubmitted against the main repo: [PR](https://github.com/zephyrproject-rtos/lvgl/pull/36)

closed 2023-08-22 by carlescufi