zephyrproject-rtos/zephyr #54717: --generate-hardware-map produces TypeError: expected string or bytes-like object on Windows

bp-amitchone, 2023-02-10:
Hi, 

Attempting to run `python ../zephyr/scripts/twister --generate-hardware-map map.yaml`  produces the following traceback:

```
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.2.0-14-gdbb434c4dc03
INFO    - Using 'gnuarmemb' toolchain.
INFO    - Scanning connected hardware...
Traceback (most recent call last):
  File "C:\brill-zephyr-workspace\zephyr\scripts\twister", line 405, in <module>
    ret = main()
  File "C:\brill-zephyr-workspace\zephyr\scripts\twister", line 286, in main
    ret = hwm.discover()
  File "C:\brill-zephyr-workspace\zephyr\scripts/pylib/twister\twisterlib\hardwaremap.py", line 149, in discover
    self.scan(persistent=self.options.persistent_hardware_map)
  File "C:\brill-zephyr-workspace\zephyr\scripts/pylib/twister\twisterlib\hardwaremap.py", line 293, in scan
    if re.match(p, d.product):
  File "C:\Python3_10_8\lib\re.py", line 190, in match
    return _compile(pattern, flags).match(string)
TypeError: expected string or bytes-like object
```

Python version: 3.10.8
Zephyr version: zephyr-v3.2.0-14-gdbb434c4dc03
I'm using an STLINK V3 on a nucleo_g474re development board

It looks as if `type(d.product)` is `NoneType`.

nashif, 2023-04-06:
Can you provide the content of map.yaml?

bp-amitchone, 2023-04-19:
@nashif I'll take a look at this again tomorrow and grab the contents of map.yaml for you

bp-amitchone, 2023-04-26:
Here are the contents of map.yaml: 

```
- connected: true
  id: 001800375632500920313236
  platform: unknown
  product: unknown
  runner: unknown
  serial: COM6
```

FYI: My project is currently on Zephyr v3.3 and this issue doesn't occur; I had to downgrade to v3.2.0 to reproduce this error.

closed 2023-07-11 by github-actions[bot]