zephyrproject-rtos/zephyr #54719: STM32 clock frequency claculation error

Logiase, 2023-02-10:
**Describe the bug**
Some STM32 model use a wrong pll frequency calculation which will cause an overflow.

**To Reproduce**

Build sample `samples/subsys/usb/console` on board `blackpill_stm32f411ce` or other stm32 board (except H7 MP1 U5) which support USB.

**Expected behavior**

USB sample work fine.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

USB clock check will not pass if `CONFIG_USB_DC_STM32_CLOCK_CHECK=y`

**Logs and console output**
```
<err> usb_dc_stm32: USB Clock is not 48MHz (5050327)
```

**Environment (please complete the following information):**
 - OS: ArchLinux
 - Zephyr SDK
 - fd1f28efb6d9f1e337d5794c157ada8f65b17c1b

Logiase, 2023-02-10:
See #54716

erwango, 2023-02-22:
Reopen to allow backport in V3.3

FRASTM, 2023-03-10:
backport is now merged: closing

PR #54716 (2023-02-22): driver: clock_control: fix stm32 pll freq calculation
change stm32 pll frequency calculation to resolve uint32_t overflow.

The origin calculation cause an overflow.

For example, use a stm32f411ce
- HSE 25MHz
- PLLM 25
- PLLN 192
- PLLP 2
- PLLQ 4
such a configuration leads to 48MHz for USB.

[image]

```c
    uint32_t pllsrc_freq = 25000000;
    int pllm_div = 25;
    int plln_mul = 192;
    int pllout_div = 4;
    
    uint32_t origin_result = (pllsrc_freq * plln_mul) / (pllm_div * pllout_div); // 5050327
    uint32_t new_result = pllsrc_freq / pllm_div * plln_mul / pllout_div;        // 48 [... cut]