zephyrproject-rtos/zephyr #54746: ESP32 SPI word size is not respected

morsisko, 2023-02-11:
I need to communicate with SPI device. To do so, 32 bits need to be send in single SPI transaction, as this is required by the receiving device. However, when you set SPI word size to 32 in Zephyr, only 8 bits will be send when you use ESP32 SPI driver. 

**To Reproduce**
Initialize your SPI driver like this:
```
SPI_DT_SPEC_INST_GET(_num, \
			SPI_OP_MODE_MASTER | \
			SPI_TRANSFER_MSB | \
			SPI_MODE_CPOL | \
			SPI_WORD_SET(32), \
			0),
```

Send your payload like this:
```
        uint8_t buffer_tx[4] = { 0xAA, 0xBB, 0xCC, 0xDD };
	uint8_t buffer_rx[4];

	const struct spi_buf tx_buf = {
		.buf = buffer_tx,
		.len = sizeof(buffer_tx),
	};

	const struct spi_buf_set tx = {
		.buffers = &tx_buf,
		.count = 1
	};

	const struct spi_buf rx_buf= {
		.buf = buffer_rx,
		.len = sizeof(buffer_rx),
	};

	const struct spi_buf_set rx = {
		.buffers = &rx_buf,
		.count = 1
	};

	spi_transceive_dt(&config->bus, &tx, &rx));
```

**Expected behavior**
0xAA 0xBB 0xCC 0xDD should be send over SPI in single transaction

**Impact**
This would make SPI sensor drivers that use 32 bit word size unable to work

**Logs and console output**
Screen from logic analyzer:
[image]

As you can see, only 8 bits are being sent when 32 bit word size is used.
**Environment:**
 - OS: Linux Ubuntu 22
 - Toolchain: zephyr-sdk-0.15.0_linux-x86_64
 - Commit SHA: bb8b74502091a90bff08de1b66b560a04b7c28e1

**Additional context**
Board that I used: esp32

Interestingly, when you use `SPI_WORD_SET(16)`, indeed 16 bits will be sent in single transaction:
[image]

And when you set `SPI_WORD_SET(8)`... 32 bits will be sent in single transaction! (In my opinion this is what should happen when `SPI_WORD_SET(32)` is set):
[image]

For my tests I can use `SPI_WORD_SET(8)` so this is not a problem for me, however as I said above, this will make all sensor drivers that would use `SPI_WORD_SET(32)` unable to work, as the sensor won't receive all the data.

morsisko, 2023-02-12:
Update. I just did the same experiment with STM32 board (to be more precise: nucleo_f429zi). Unfortunately this driver doesn't support 32 bit word length, but this is what happen when I set the word size to 16:
[image]
It looks like the driver treat `len` property of `spi_buf` as number of words, not number of bytes, however ESP32 driver treat is as number of bytes. Which behavior is correct?

LucasTambor, 2023-03-28:
@morsisko can you please test #56283?  I believe it solves the issue. Thanks!

morsisko, 2023-05-05:
Hello @LucasTambor. Sorry, I forgot about the issue. I checked it right now. At this moment, no matter what spi_len buffer size I choose, all 32 bits are sent in single transaction. If this is the desired behavior it works ok.

By the way, I see some inconsistency between SPI drivers. Please look at this code:

```
const struct spi_buf tx_buf = {
	.buf = buffer_tx,
	.len = sizeof(buffer_tx),
};
```
What is the .len here? Is it the length of the buffer in bytes, or length of the buffer in words?

LucasTambor, 2023-05-07:
Hello @morsisko. No problem :) I believe the correct behaviour of the driver is now expected since #57426 .

closed 2023-05-21 by morsisko