zephyrproject-rtos/zephyr #54754: outdated version of rpi_pico hal configures USB PLL incorrectly

purdeaandrei, 2023-02-13:
**Describe the bug**

I have noticed that the rpi_pico hal configures the FVCO of the USB PLL out of range.
See this line: https://github.com/zephyrproject-rtos/hal_rpi_pico/blob/zephyr/src/rp2_common/hardware_clocks/clocks.c#L157
Says: `pll_init(pll_usb, 1, 480 * PICO_MHZ, 5, 2);`
Also see section 2.18 of the RP2040 datasheet, says:
`Oscillator frequency (FOUTVCO) must be in the range 750MHz → 1600MHz`

This has been fixed in the upstream pico-sdk in this merge request: https://github.com/raspberrypi/pico-sdk/pull/869

Please upgrade pico-sdk, or include this PR. I would prefer upgrading the whole thing.

Note that I have attempted rebasing the zephyr-specific commits on top of a fresher pico-sdk, however I ran into a symbol name collision. I hacked around it by just renaming the symbol here: https://github.com/purdeaandrei/hal_rpi_pico/commit/f2be0629671f0c4e7ff6188e9c28586e44fcb4bc

But not sure what the right thing to do here is, and also not sure how hal fork updates should be handled. (merge/rebase/etc..)

yonsch, 2023-02-13:
@soburi started working on it here https://github.com/zephyrproject-rtos/hal_rpi_pico/pull/3 but it did not make it in time for the 3.3 feature freeze. If what you're describing is enough to consider this a bug worth updating the hal for then it can be done. Maybe @stephanosio or @laurenmurphyx64 can help here

soburi, 2023-03-14:
@purdeaandrei @yonsch 
Now SDK is updated to 1.5.0.

https://github.com/zephyrproject-rtos/zephyr/commit/93f7a6730d12dd5f8eae8cadb41adeb66c1136d9

closed 2023-03-14 by yonsch