zephyrproject-rtos/zephyr #54758: How to resolve warning: initialization of 'uint32_t (*)(const TIM_TypeDef *), from incompatible pointer type 'uint32_t (*)(TIM_TypeDef *)' {aka 'unsigned int (*)(TIM_TypeDef *)'} [-Wincompatible-pointer-types]

Niranjan-eaton, 2023-02-13:
## Origin

How to resolve Warning, while building Multi_Timers in 3.2.0 Zephyr version.
zephyr-v3.2.0/zephyr/drivers/counter/counter_ll_stm32_timer.c:40:9: warning: initialization of 'uint32_t (*)(const TIM_TypeDef *)
{aka 'unsigned int (*)(const TIM_TypeDef *)'} from incompatible pointer type 'uint32_t (*)(TIM_TypeDef *)' {aka 'unsigned int (*)(TIM_TypeDef *)'}
 [-Wincompatible-pointer-types]

## Type
zephyr-v3.2.0
board: nucleo_h743zi

## Module
Multi_Timers example

nucleo_h743zi.overlay
&timers2 {
	st,prescaler = <95>;
	counter {
			compatible = "st,stm32-counter";
			status = "okay";
		};
};

## Purpose
Resolve below warnings,
zephyr/drivers/counter/counter_ll_stm32_timer.c:40:9: warning: initialization of 'uint32_t (*)(const TIM_TypeDef *)'
 {aka 'unsigned int (*)(const TIM_TypeDef *)'} from **incompatible pointer type** 'uint32_t (*)(TIM_TypeDef *)' {aka 'unsigned int (*)(TIM_TypeDef *)'} [-Wincompatible-pointer-types]
   40 |         LL_TIM_OC_GetCompareCH1, LL_TIM_OC_GetCompareCH2,
[image]

zephyr-v3.2.0\zephyr\drivers\counter\counter_ll_stm32_timer.c

			/** Channel to compare get function mapping. */
Line No:38 #if !defined(CONFIG_SOC_SERIES_STM32F4X)
Line No:39 static uint32_t(*const get_timer_compare[TIMER_MAX_CH])( const TIM_TypeDef *) = {
Line No:40 	LL_TIM_OC_GetCompareCH1, LL_TIM_OC_GetCompareCH2, 
Line No:41	LL_TIM_OC_GetCompareCH3, LL_TIM_OC_GetCompareCH4,
			};
			#else
			static uint32_t(*const get_timer_compare[TIMER_MAX_CH])(TIM_TypeDef *) = {
				LL_TIM_OC_GetCompareCH1, LL_TIM_OC_GetCompareCH2,
				LL_TIM_OC_GetCompareCH3, LL_TIM_OC_GetCompareCH4,
			};
			#endif

Note: After removed "const" in static uint32_t(*const get_timer_compare[TIMER_MAX_CH])( const TIM_TypeDef *), warnings are not getting, but not correcting to remove const.

nordicjm, 2023-02-13:
@Niranjan-eaton You have selected the wrong category for this, if this is an issue then it should be opened with "bug" not "TSC" - the technical steering committee of zephyr does not need to read this or vote on it.

FRASTM, 2023-02-13:
@Niranjan-eaton  Can you please check your version of the modules/hal/stm32/stm32cube/stm32h7xx    
I guess the pb you see is fixed with version v1.11.0 of the modules/hal/stm32/stm32cube/stm32h7xx
And it will come with the next zephyr 3.3.0
Maybe in between you could west update and get the [hal_stm32](https://github.com/zephyrproject-rtos/hal_stm32)  master branch c36136b253a46ed8dd63525eeee659393047ce83

Niranjan-eaton, 2023-02-14:
Tried, west update in zephyr-v3.2.0>west update , but not resolved.
[west_update_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/10730947/west_update_log.txt)

Please help, suggest/Idea to remove warnings.