zephyrproject-rtos/zephyr #54760: net_lwm2m_engine: fcntl(F_GETFL) failed (-22) on es-wifi

jvermillard, 2023-02-13:
**Describe the bug**

On board `b_l4s5i_iot01a` I'm trying to use the LWM2M client with the es-wifi driver.

But receive a strange fcntl error.

**To Reproduce**

Try to use the  lwm2m client on this board with the wifi driver. I can provide the code if needed (even if it's just a copy paste of the wifi and the lwm2m example app)

**Expected behavior**

Connection

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-3973-gf16e2a9d80c0  ***
no sec done
server done
Zephyr Wifi/LWM2M app connecting 1.0.0
<inf> HTS221: Cannot enable trigger without drdy-gpios
<dbg> net_lwm2m_engine: lwm2m_engine_init: LWM2M engine socket receive thread started
<dbg> net_lwm2m_obj_security: security_create: Create LWM2M security instance: 0
<dbg> net_lwm2m_obj_server: server_create: Create LWM2M server instance: 0
<dbg> net_lwm2m_obj_device: device_create: Create LWM2M device instance: 0
<dbg> net_lwm2m_obj_firmware: firmware_create: Create LWM2M firmware instance: 0
<inf> net_config: Initializing network
<dbg> net_lwm2m_registry: lwm2m_engine_set: path:0/0/2, value:0x2000a987, len:1
<dbg> net_lwm2m_registry: lwm2m_engine_set: path:0/0/10, value:0x2000a986, len:2
<dbg> net_lwm2m_registry: lwm2m_engine_set: path:1/0/0, value:0x2000a986, len:2
<inf> net_lwm2m_rd_client: Start LWM2M Client: b_l4s5i_iot01a
<dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1, buf:0x200097a7, buflen:1
<dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10, buf:0x200097de, buflen:2
<dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1, buf:0x200097b0, buflen:4
<inf> net_lwm2m_rd_client: RD Client started with endpoint 'b_l4s5i_iot01a' with client lifetime 30
<dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://leshan.eclipseprojects.io
uart:~$ WiFi connected
Connected

Successful connected to SSID:[93]
WiFi done
Setting up LWM2M
<err> net_lwm2m_engine: fcntl(F_GETFL) failed (-22)
<err> net_lwm2m_rd_client: Cannot init LWM2M engine (-22)
<dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1, buf:0x200097a7, buflen:1
<dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10, buf:0x200097de, buflen:2
<dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1, buf:0x200097b0, buflen:4
<inf> net_lwm2m_rd_client: RD Client started with endpoint 'b_l4s5i_iot01a' with client lifetime 30
<dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://leshan.eclipseprojects.io
<err> net_lwm2m_engine: fcntl(F_GETFL) failed (-22)
<err> net_lwm2m_rd_client: Cannot init LWM2M engine (-22)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.15.2
 - Commit SHA or Version used: zephyr-v3.1.0-3973-gf16e2a9d80c0

jvermillard, 2023-02-13:
The code:
```
/*
 * Copyright (c) 2021 Nordic Semiconductor ASA
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>

#include <zephyr/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/drivers/gpio.h>

#include "app_version.h"

#include <zephyr/net/wifi.h>
#include <zephyr/net/net_if.h>
#include <zephyr/net/net_event.h>
#include <zephyr/net/net_mgmt.h>
#include <zephyr/net/wifi_mgmt.h>

#include <zephyr/net/lwm2m.h>

#include <stdio.h>

#define LOG_MODULE_NAME net_lwm2m_client_app
#define LOG_LEVEL LOG_LEVEL_DBG

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(LOG_MODULE_NAME);

//LOG_MODULE_REGISTER(main, CONFIG_APP_LOG_LEVEL);

#define AUTO_CONNECT_SSID "*****"

#define AUTO_CONNECT_SSID_PSK "******"
#define LWM2M_SERVER "coap://leshan.eclipseprojects.io"
//#define LWM2M_SERVER "coap://23.97.187.154"

#define WIFI_MGMT_EVENTS (NET_EVENT_WIFI_SCAN_RESULT |		\
				NET_EVENT_WIFI_SCAN_DONE |		\
				NET_EVENT_WIFI_CONNECT_RESULT |		\
				NET_EVENT_WIFI_DISCONNECT_RESULT)

static struct net_mgmt_event_callback cb;
static struct k_sem net_cb_sem;

static struct lwm2m_ctx client;

#define CLIENT_MANUFACTURER	"Zephyr"
#define CLIENT_MODEL_NUMBER	"OMA-LWM2M Sample Client"
#define CLIENT_SERIAL_NUMBER	"345000123"
#define CLIENT_FIRMWARE_VER	"1.0"
#define CLIENT_DEVICE_TYPE	"OMA-LWM2M Client"
#define CLIENT_HW_VER		"1.0.1"

static uint8_t bat_idx = LWM2M_DEVICE_PWR_SRC_TYPE_BAT_INT;
static int bat_mv = 3800;
static int bat_ma = 125;
static uint8_t usb_idx = LWM2M_DEVICE_PWR_SRC_TYPE_USB;
static int usb_mv = 5000;
static int usb_ma = 900;
static uint8_t bat_level = 95;
static uint8_t bat_status = LWM2M_DEVICE_BATTERY_STATUS_CHARGING;
static int mem_free = 15;
static int mem_total = 25;

static void Wifi_check_connect_result( struct net_if *iface, struct net_mgmt_event_callback *cb)
{
	const struct wifi_status *status = (const struct wifi_status *) cb->info;
[... 241 lines cut ...]
	int ret;

	ret = lwm2m_setup();

	if (ret <0) {
		printk("errror setting up LWM2M\n");
	}

	(void)memset(&client, 0x0, sizeof(client));

	/* client.sec_obj_inst is 0 as a starting point */
	lwm2m_rd_client_start(&client, CONFIG_BOARD, 0, rd_client_event, observe_cb);

	printk("Zephyr Wifi/LWM2M app connecting %s\n", APP_VERSION_STR);
	// setup wifi
	wifi_autoconnect();

	printk("WiFi done\nSetting up LWM2M\n");

	while (1) {

		k_sleep(K_MSEC(1000));
	}
}
```

jvermillard, 2023-02-13:
Looks like related to : https://github.com/zephyrproject-rtos/zephyr/commit/d0dc67a72d1c360c11977c427f74e771dc2eb6d2 ( thanks @kartben )

jvermillard, 2023-02-13:
After moving to 3.3.0-rc3, I still have the issue (but different):

```
*** Booting Zephyr OS build v3.3.0-rc3 ***
[00:00:00.795,000] <dbg> net_lwm2m_engine: lwm2m_engine_init: LWM2M engine socket receive thread started
[00:00:00.795,000] <dbg> net_lwm2m_obj_security: security_create: Create LWM2M security instance: 0
[00:00:00.795,000] <dbg> net_lwm2m_obj_server: server_create: Create LWM2M server instance: 0
[00:00:00.795,000] <dbg> net_lwm2m_obj_device: device_create: Create LWM2M device instance: 0
Zephyr Wifi/LWM2M app connecting 1.0.0
WiFi connected
Connected

Successful connected to SSID:[93]
WiFi done
Setting up LWM2M
[00:00:04.110,000] <dbg> net_lwm2m_registry: lwm2m_engine_set: path:0/0/2, buf:0x20009d1f, len:1
no sec done
[00:00:04.110,000] <dbg> net_lwm2m_registry: lwm2m_engine_set: path:0/0/10, buf:0x20009d1e, len:2
[00:00:04.110,000] <dbg> net_lwm2m_registry: lwm2m_engine_set: path:1/0/0, buf:0x20009d1e, len:2
server done
[00:00:04.110,000] <inf> net_lwm2m_rd_client: Start LWM2M Client: b_l4s5i_iot01a
[00:00:04.796,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1/0, level 3, buf:0x20008b5f, buflen:1
[00:00:04.796,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10/0, level 3, buf:0x20008b8e, buflen:2
[00:00:04.796,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1/0, level 3, buf:0x20008b68, buflen:4
[00:00:04.796,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'b_l4s5i_iot01a' with client lifetime 30
[00:00:04.796,000] <dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://leshan.eclipseprojects.io
[00:00:04.892,000] <err> net_lwm2m_engine: zsock_fcntl(F_GETFL) failed (-22)
[00:00:04.912,000] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-22)
[00:00:05.912,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1/0, level 3, buf:0x20008b5f, buflen:1
[00:00:05.912,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10/0, level 3, buf:0x20008b8e, buflen:2
[00:00:05.912,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1/0, level 3, buf:0x20008b68, buflen:4
[00:00:05.912,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'b_l4s5i_iot01a' with client lifetime 30
[00:00:05.912,000] <dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://leshan.eclipseprojects.io
[00:00:05.991,000] <err> net_lwm2m_engine: zsock_fcntl(F_GETFL) failed (-22)
[00:00:06.011,000] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-22)
[00:00:07.511,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1/0, level 3, buf:0x20008b5f, buflen:1
[00:00:07.511,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10/0, level 3, buf:0x20008b8e, buflen:2
[00:00:07.511,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1/0, level 3, buf:0x20008b68, buflen:4
[00:00:07.511,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'b_l4s5i_iot01a' with client lifetime 30
[00:00:07.511,000] <dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://leshan.eclipseprojects.io
[00:00:07.597,000] <err> net_lwm2m_engine: zsock_fcntl(F_GETFL) failed (-22)
[00:00:07.617,000] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-22)
[00:00:10.117,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1/0, level 3, buf:0x20008b5f, buflen:1
[00:00:10.117,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10/0, level 3, buf:0x20008b8e, buflen:2
[00:00:10.118,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1/0, level 3, buf:0x20008b68, buflen:4
[00:00:10.118,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'b_l4s5i_iot01a' with client lifetime 30
[00:00:10.118,000] <dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://leshan.eclipseprojects.io
[00:00:10.208,000] <err> net_lwm2m_engine: zsock_fcntl(F_GETFL) failed (-22)
[00:00:10.227,000] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-22)
[00:00:14.728,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1/0, level 3, buf:0x20008b5f, buflen:1
[00:00:14.728,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10/0, level 3, buf:0x20008b8e, buflen:2
[00:00:14.728,000] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1/0, level 3, buf:0x20008b68, buflen:4
[00:00:14.728,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'b_l4s5i_iot01a' with client lifetime 30
[00:00:14.728,000] <dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://leshan.eclipseprojects.io
[00:00:14.808,000] <err> net_lwm2m_engine: zsock_fcntl(F_GETFL) failed (-22)
[00:00:14.827,000] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-22)
```

rlubos, 2023-02-13:
With eswifi driver, is socket offloading enabled? If so, that implementation is incomplete, i. e. it does not support `F_GETFL/F_SETFL`, not allowing to set the socket into non-blocking mode:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/wifi/eswifi/eswifi_socket_offload.c#L592

If that's the case, it's not a LwM2M bug and should be handled by the driver maintainer. 

AFAIR that driver also implements the other offloading we have in Zephyr (no clue why both are implmented) so you can try to disable `CONFIG_NET_SOCKETS_OFFLOAD` and see if it helps. In that case, native socket implementation from Zephyr is used.

jvermillard, 2023-02-13:
@rlubos Thanks! this workaround works: `[00:00:05.337,000] <inf> net_lwm2m_rd_client: Registration Done (EP='DRquVCKuvA')` :tada:

rlubos, 2023-02-14:
Closing as it's not relly an LwM2M issue. If it's still a concern to you, please open an enhancement requrest for `eswifi` to add non-blocking socket support.

closed 2023-02-14 by rlubos