zephyrproject-rtos/zephyr #54769: Error when flashing to LPCXpresso55S06 EVK.

BerkantCiftci, 2023-02-13:
I am working on a IoT system based on Zephyr (work related) and I am using LPCxpresso55S06, 
but I am not able to flash with West in my terminal.
I already did a clean install of zephyr and west following the instructions on the Getting started page.

Please also mention any information which could help others to understand
the problem you're facing:
 - I am using West on my terminal
 - I have been developing on the LPCxpresso55S06 and cannot flash. With my other dev. board LPC55s16 no issues.

When I want to flash blinky on my LPC55S06 EVK, i get an error message with status 1. 
Steps to reproduce the behavior:
1. west build -b lpcxpresso55s06 samples/hello_world
2. west flash

The console output:

C:\Users\Berkant\zephyrproject\zephyr>west build -p always -b lpcxpresso55s06 samples\basic\blinky
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/Berkant/zephyrproject/zephyr/samples/basic/blinky
-- CMake version: 3.25.2
-- Found Python3: C:/Users/Berkant/AppData/Local/Programs/Python/Python311/python.exe (found suitable exact version "3.11.1") found components: Interpreter
-- Cache files will be written to: C:/Users/Berkant/zephyrproject/zephyr/.cache
-- Zephyr version: 3.3.0-rc3 (C:/Users/Berkant/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: lpcxpresso55s06
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (C:/Users/Berkant/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (C:/Users/Berkant/zephyr-sdk-0.15.2)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/Berkant/zephyrproject/zephyr/boards/arm/lpcxpresso55s06/lpcxpresso55s06.dts
-- Generated zephyr.dts: C:/Users/Berkant/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/Berkant/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/Berkant/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing C:/Users/Berkant/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/Berkant/zephyrproject/zephyr/boards/arm/lpcxpresso55s06/lpcxpresso55s06_defconfig'
Merged configuration 'C:/Users/Berkant/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/Users/Berkant/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/Berkant/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Berkant/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
Load components for LPC55S06:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
driver_lpc_gpio component is included.
driver_pint component is included.
driver_inputmux component is included.
driver_inputmux_connections component is included.
driver_flexcomm_usart component is included.
driver_flexcomm component is included.
driver_lpc_iocon component is included.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Berkant/zephyrproject/zephyr/build
←[92m-- west build: building application
[1/156] Generating include/generated/version.h
-- Zephyr version: 3.3.0-rc3 (C:/Users/Berkant/zephyrproject/zephyr), build: v3.3.0-rc3-1-g0d9fea87bb3b
[146/156] Linking C executable zephyr\zephyr_pre0.elf

[150/156] Linking C executable zephyr\zephyr_pre1.elf

[156/156] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       15512 B       246 KB      6.16%
             RAM:        4008 B        80 KB      4.89%
           SRAMX:          0 GB        16 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%

C:\Users\Berkant\zephyrproject\zephyr>west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
-- runners.jlink: JLink version: 7.84
-- runners.jlink: Flashing file: C:\Users\Berkant\zephyrproject\zephyr\build\zephyr\zephyr.bin
FATAL ERROR: command exited with status 1: 'C:\Program Files\SEGGER\JLink\JLink.exe' -nogui 1 -if swd -speed auto -device LPC55S06 -CommanderScript 'C:\Users\Berkant\AppData\Local\Temp\tmpl3qmbf1gjlink\runner.jlink' -nogui 1
```
I have a feeling that the name of my board is not fully listed in the script when I run the west flash command. It says LPC55S06 instead of lpcxpreso55s06. Maybe that's the problem.

henrikbrixandersen, 2023-02-14:
Are you using a Segger J-Link to flash the board? Does `pyocd list` show your board? If it does, try `west flash -r pyocd`.

BerkantCiftci, 2023-02-14:
I have flashed the on-board debugger with Jlink. I am gonna try the pyocd list in a bit

BerkantCiftci, 2023-02-14:
So yes my board is listed when I do ``` pyocd list ``` but then I get this:

[image]

henrikbrixandersen, 2023-02-16:
> So yes my board is listed when I do `pyocd list` but then I get this:
> 
> [image]

Okay, so the board is indeed identified as using a Segger J-Link. I would suggest trying to use the Segger J-Link tools directly and debug why they cannot establish to the CPU core.

BerkantCiftci, 2023-02-16:
I will check tomorrow and reply back. Thanks!

BerkantCiftci, 2023-02-17:
I found what the problem was. I noticed while flashing the JLink firmware that I was using an older LPCScrypt version. This was the cause. After installing and flashing the latest version, I soon saw that I could flash and debug with west.

I want to thank you @henrikbrixandersen for pointing out! 😄

closed 2023-02-17 by BerkantCiftci