zephyrproject-rtos/zephyr #54783: stm32: NULL dereference in net_eth_carrier_on

jchabod, 2023-02-13:
**Describe the bug**
I get a memory fault in function `net_eth_carrier_on`. 

I cannot use the debugger due to another issue (#54630), so that I instrumented the `net_eth_carrier_on` and callers which  pointed out that `net_eth_carrier_on` is called with `NULL`, leading to system crash. 

The `rx_thread` in `drivers/ethernet/eth_stm32_hal.c` accesses to `dev_data->iface` before the device is initialized.

I made the following patch locally to have my program running (Of course I don't know if this may have side-effects, but this is a workaround for me...):

```diff
diff --git a/drivers/ethernet/eth_stm32_hal.c b/drivers/ethernet/eth_stm32_hal.c
index 75b48daff5..050ea7d9f4 100644
--- a/drivers/ethernet/eth_stm32_hal.c
+++ b/drivers/ethernet/eth_stm32_hal.c
@@ -880,7 +880,11 @@ static void rx_thread(void *arg1, void *unused1, void *unused2)
        while (1) {
                res = k_sem_take(&dev_data->rx_int_sem,
                        K_MSEC(CONFIG_ETH_STM32_CARRIER_CHECK_RX_IDLE_TIMEOUT_MS));
+        if (NULL == dev_data->iface){
+            continue;
+        }
                if (res == 0) {
+
                        /* semaphore taken, update link status and receive packets */
                        if (dev_data->link_up != true) {
                                dev_data->link_up = true;

```

** More context **
- I am porting an application that was running currectly under Zephyr v3.2.0. The issue appeared when I moved to Zephyr v3.3.0-rc2 + STM32 fork to handle Display on STM32H747 (See https://github.com/dariis/zephyr/tree/stm32h7-mipi-dsi). 
- The issue appears before the program reaches the "MAIN" procedure.

**To Reproduce**
Run Hello world sample with the following `prj.conf` :

```
# nothing here
CONFIG_NEWLIB_LIBC=y
CONFIG_MAIN_STACK_SIZE=6000

CONFIG_NETWORKING=y   				# Required for networking
CONFIG_NET_IPV4=y					# enables IPV4
CONFIG_NET_IPV4_AUTO=y				# enables ARP automatic answers
CONFIG_NET_TCP=y					# enables tcp stack
CONFIG_NET_SOCKETS=y				# enables sockets
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=5

CONFIG_NET_L2_DUMMY=y				# enables LOOPBACK card
CONFIG_NET_L2_ETHERNET=y			# enables ETHERNET card
CONFIG_NET_LOOPBACK=y				# enables LOOPBACK card
CONFIG_NET_ARP=y					# enables ARP
CONFIG_NET_MGMT=y
CONFIG_DNS_RESOLVER=y

CONFIG_NET_DEFAULT_IF_ETHERNET=y	# 
CONFIG_NET_MAX_CONN=16
CONFIG_NET_MAX_CONTEXTS=16
CONFIG_NET_TX_STACK_SIZE=1500

CONFIG_NET_IF_MAX_IPV4_COUNT=2
CONFIG_NET_IF_MCAST_IPV4_ADDR_COUNT=8
CONFIG_NET_BUF_DATA_SIZE=2048

CONFIG_NET_DRIVERS=y				# Required for communications
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_PROMISCUOUS_MODE=y
CONFIG_NET_NATIVE=y					# uses zephyr net stack

CONFIG_LOG=y
# CONFIG_DEBUG_COREDUMP=y
CONFIG_FAULT_DUMP=2

# provides a console with net tools (after application returns)	
CONFIG_SHELL=y
CONFIG_NET_SHELL=y

CONFIG_DISPLAY=y
CONFIG_OTM8009A=y

CONFIG_STDOUT_CONSOLE=y
CONFIG_PRINTK=y
# CONFIG_KSCAN=y
# CONFIG_NET_LOG=y

CONFIG_LVGL=y
CONFIG_LV_MEM_CUSTOM=y
CONFIG_LV_USE_LOG=y
CONFIG_LV_USE_LABEL=y
# CONFIG_LV_USE_BTN=y
# CONFIG_LV_USE_IMG=y
CONFIG_LV_FONT_MONTSERRAT_22=y
CONFIG_LV_FONT_MONTSERRAT_28=y
CONFIG_LV_Z_MEM_POOL_NUMBER_BLOCKS=8
``` 

**Environment (please complete the following information):**
 - board: `stm32h747i_disco_m7`
 - OS: Windows
 - Zephyr v3.3.0-rc2 + STM32 fork to handle Display on STM32H747 (See https://github.com/dariis/zephyr/tree/stm32h7-mipi-dsi). There are only commits for DISPLAY on top of rc3.3.0-rc2 
 - `ZEPHYR_TOOLCHAIN_VARIANT=zephyr`
 - `ZEPHYR_SDK_INSTALL_DIR=D:\Z\zephyr-sdk-0.15.2`

jchabod, 2023-02-13:
Adding a `printk(NULL == dev_data->iface\n)` call in the added code confirms:
```
NULL == dev_data->iface
*** Booting Zephyr OS build v3.3.0-rc2-10-g51d526408a5d ***
Hello World! stm32h747i_disco_m7
```

Desvauxm-st, 2023-02-22:
stm32: NULL dereference in net_eth_carrier_on #54783

Hi @ [jchabod](https://github.com/jchabod)
  I have in charge to reproduce the issue you faced.
  I try to compile the samples/hello_word with all prj.conf you give us

First:
I must remove CONFIG_OTM8009A=y  because : undefined symbol with 
west build --pristine -b stm32h747i_disco_m7 samples/hello_world/

Second :
I must remove 
CONFIG_LVGL=y because many compilation error when 
west build --pristine -b stm32h747i_disco_m7 samples/hello_world/
So I also remove all the CONFIG_LV_xxxx

Third:
I modify the drivers/ethernet/eth_stm32_hal.c like that:

        while (1) {
                res = k_sem_take(&dev_data->rx_int_sem,
                        K_MSEC(CONFIG_ETH_STM32_CARRIER_CHECK_RX_IDLE_TIMEOUT_MS));
                if ( dev_data->iface == NULL){                  
                        while(1){
                        }               }
                if (res == 0) {
                        /* semaphore taken, update link status and receive packets */
                        if (dev_data->link_up != true) {
                                dev_data->link_up = true;
                                net_eth_carrier_on(get_iface(dev_data,
                                                             vlan_tag));
                        }

After that I never enter in your case (dev_data->iface==NULL){
while(1);}

Even with the debugger I never enter in the while(1) after
                if ( dev_data->iface == NULL){                  
                        while(1){
                        }  
                 } 

See the debugger printscreen after in file attachement

 

See the result into the console after in file attachement:
 
[NULL dereference in net_eth_carrier_on.docx](https://github.com/zephyrproject-rtos/zephyr/files/10801660/NULL.dereference.in.net_eth_carrier_on.docx)

 

So I can’t reproduce the issue for the moment

Desvauxm-st, 2023-02-22:
@jchabod 
do you have any idea how i can reproduce the Issue ?

jchabod, 2023-02-22:
Hi,

I think that this issue raises only when all drivers come into action together. Thus, I am not surprised that removing the graphic layer also removes the issue.

Can you update Zephyr to this branch? https://github.com/dariis/zephyr/tree/stm32h7-mipi-dsi 

Please here find the GIT bundle that includes my local modifications on top of this branch (In particular, there is one commit that activates the network card(`c4241c6777`). You'll require SHA1 `ddeecae9a2c` which is on the ST fork for display

[54783.zip](https://github.com/zephyrproject-rtos/zephyr/files/10802288/54783.zip)

jchabod, 2023-02-22:
Ok, It seems easier to directly give you the bundle from rc3.3.0-rc2 up to my working point.

[rc3.3.0_to_54783.zip](https://github.com/zephyrproject-rtos/zephyr/files/10802342/rc3.3.0_to_54783.zip)

Desvauxm-st, 2023-02-22:
Hi 
I update git from this branch :  https://github.com/dariis/zephyr/tree/stm32h7-mipi-dsi:
but after  I have the  same error with CONFIG_LVGL=y

I can't use or I don't known how to use rc3.3.0_to_54783.bundle
i have that issue :

git clone rc3.3.0_to_54783.bundle  my_repo
Cloning into 'my_repo'...
error: Repository lacks these prerequisite commits:
error: 7e20b819541a028b7b298261e053cb000f2a8571
fatal: remote transport reported error

jchabod, 2023-02-22:
SHA1 7e20b81 is the Zephyr tag v3.3.0-rc2.

Sorry, I forgot to explain that with the last bundle you just do not need the "fork" anymore(So just forget it). However, if you fetch zephyr tags, you should get this reference.

erwango, 2023-05-11:
@jchabod Would you mind checking https://github.com/zephyrproject-rtos/zephyr/pull/57779 ?

erwango, 2023-05-17:
Should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/57779

erwango, 2023-05-24:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/57779

closed 2023-05-24 by erwango