zephyrproject-rtos/zephyr #54785: .data and .bss relocation to DTCM & CCM is broken with SDK 0.15+

zagor, 2023-02-13:
Using CONFIG_CODE_DATA_RELOCATION to relocate .data or .bss to a DTCM or CCM area is broken. Relocating to an SRAMn area still works fine.

The problem is that the "start" symbol points to the end of the memory region.

**To Reproduce**
1. Change line 17 in file samples/application_development/code_relocation_nocopy/CMakeLists.txt to relocate to DTCM_DATA instead of SRAM_DATA:
`zephyr_code_relocate(FILES src/ext_code.c LOCATION DTCM_DATA)`
2. Build the sample for a chip with an DTCM area, such as stm32h753:
`west build --pristine -b nucleo_h753zi samples/application_development/code_relocation_nocopy`
3. Inspect the address of the dtcm_data symbols:
```
$ grep dtcm_data build/zephyr/zephyr.map
.dtcm_data      0x0000000020000000        0x4 load address 0x0000000008000298
                0x0000000020000004                __dtcm_data_end = .
                0x0000000020000004                __dtcm_data_start = ADDR (.dtcm_data)
                0x0000000000000004                __dtcm_data_size = SIZEOF (.dtcm_data)
                0x0000000008000298                __dtcm_data_rom_start = LOADADDR (.dtcm_data)
.dtcm_data      0x0000000020000004        0x0 load address 0x0000000008006d84
                0x0000000020000004                __dtcm_data_start = .
 *(SORT_BY_ALIGNMENT(.dtcm_data))
 *(SORT_BY_ALIGNMENT(.dtcm_data.*))
                0x0000000020000004                __dtcm_data_end = .
                0x0000000008000298                __dtcm_data_load_start = LOADADDR (.dtcm_data)
```
Note how **__dtcm_data_start** and **__dtcm_data_end** are the same address even though **__dtcm_data_size** is non-zero!

The same sample built with sdk 0.14.2 produces the following rather more expected result:
```
$ grep dtcm_data build/zephyr/zephyr.map 
.dtcm_data      0x0000000020000000        0x4 load address 0x0000000008000298
                0x0000000020000000                __dtcm_data_start = .
 *(SORT_BY_ALIGNMENT(.dtcm_data))
 *(SORT_BY_ALIGNMENT(.dtcm_data.*))
                0x0000000020000004                __dtcm_data_end = .
                0x0000000020000004                __dtcm_data_end = .
                0x0000000020000000                __dtcm_data_start = ADDR (.dtcm_data)
                0x0000000000000004                __dtcm_data_size = SIZEOF (.dtcm_data)
                0x0000000008000298                __dtcm_data_rom_start = LOADADDR (.dtcm_data)
                0x0000000008000298                __dtcm_data_load_start = LOADADDR (.dtcm_data)
```

We can do the same test relocating to the CCM area on a STM32F429:
1. Change line 17 in file samples/application_development/code_relocation_nocopy/CMakeLists.txt to relocate to CCM_DATA instead of SRAM_DATA:
`zephyr_code_relocate(FILES src/ext_code.c LOCATION CCM_DATA)`
2. Build the sample:
`west build --pristine -b nucleo_f429zi samples/application_development/code_relocation_nocopy`
3. Inspect the address of the __ccm_data symbols:
`grep ccm_data build/zephyr/zephyr.map`:
```
$ grep ccm_data build/zephyr/zephyr.map
.ccm_data       0x0000000010000000        0x4 load address 0x00000000080001ac
                0x0000000010000004                __ccm_data_end = .
                0x0000000010000004                __ccm_data_start = ADDR (.ccm_data)
                0x0000000000000004                __ccm_data_size = SIZEOF (.ccm_data)
                0x00000000080001ac                __ccm_data_rom_start = LOADADDR (.ccm_data)
.ccm_data       0x0000000010000004        0x0 load address 0x0000000008006a6c
                0x0000000010000004                __ccm_data_start = .
 *(SORT_BY_ALIGNMENT(.ccm_data))
 *(SORT_BY_ALIGNMENT(.ccm_data.*))
                0x0000000010000004                __ccm_data_end = .
                0x00000000080001ac                __ccm_data_rom_start = LOADADDR (.ccm_data)
```

This problem does not occur when relocating to a SRAMn area:

1. Change line 17 in file samples/application_development/code_relocation_nocopy/CMakeLists.txt to relocate to SRAM4_DATA instead of SRAM_DATA:
`zephyr_code_relocate(FILES src/ext_code.c LOCATION SRAM4_DATA)`
2. Build the sample for a target with an SRAM4 area:
`west build --pristine -b nucleo_h753zi samples/application_development/code_relocation_nocopy`
3. Inspect the address of the __sram4_data symbols:
```
$ grep sram4_data build/zephyr/zephyr.map
                0x0000000038000004                __sram4_data_end = .
                0x0000000038000000                __sram4_data_start = ADDR (_SRAM4_DATA_SECTION_NAME)
                0x0000000000000004                __sram4_data_size = SIZEOF (_SRAM4_DATA_SECTION_NAME)
                0x0000000008000378                __sram4_data_rom_start = LOADADDR (_SRAM4_DATA_SECTION_NAME)
```

**Impact**
If we relocate a .data section, it does not get initialized. If we relocate a .bss section, it does not get zeroed.
If the memory area is filled to more than 50%, it can generate an early bus error since the memset/memcpy in code_relocation.c will attempt to write to (_data_start + _data_size) which is outside the memory area.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchains tested: Zephyr SDK 0.13.1, 0.14.2, 0.15.0, 0.15.1, 0.15.2, 0.16.0-beta1
 - Commit: 2a15a6350952360571f6d6b6db55b846c8a2094a (i.e main) and 2ddd73feafd3316af2c547c34d6969bea637d5c6 (i.e v3.1.0) for pre-0.15 sdk testing.

**Hypothesis**

My hypothesis is that this has something to do with our inserting double labels for these memory areas in the linker command file, and that the behaviour of this changed between gcc 10 (sdk 0.14) and gcc 12 (sdk 0.15):
```
$ grep dtcm_data build/zephyr/linker.cmd
 .dtcm_data :
        __dtcm_data_end = .;
        __dtcm_data_start = ADDR(.dtcm_data);
        __dtcm_data_size = SIZEOF(.dtcm_data);
        __dtcm_data_rom_start = LOADADDR(.dtcm_data);
 .dtcm_data : SUBALIGN(4)
  __dtcm_data_start = .;
  *(.dtcm_data)
  *(".dtcm_data.*")
  __dtcm_data_end = .;
 __dtcm_data_load_start = LOADADDR(.dtcm_data);
```

stephanosio, 2023-02-13:
cc @carlocaione

carlocaione, 2023-02-13:
Uhm, looking at https://github.com/zephyrproject-rtos/zephyr/commit/b4c5f4b32b8121f1852467e9879869266a8d3126 it seems that the `code_relocation` feature cannot be used on `DTCM` at all because the code for the relocation is already hardcoded at init: https://github.com/zephyrproject-rtos/zephyr/blob/b4c5f4b32b8121f1852467e9879869266a8d3126/kernel/init.c#L199-L202

zagor, 2023-02-13:
Ah, that looks like a good clue.

Well. it certainly _can_ be used, as shown in this bug report. And, by intention or by luck, it works as documented in Zephyr v3.1.0.  It only stopped working after updating to 3.2.

I don't quite follow why we have these sections hardcoded for CCM and DTCM when we don't add them for other sram areas. Especially since it clearly conflicts with code_relocation.

zagor, 2023-02-14:
So if the core problem is section name collision, is the fix to rename the sections generated by code_relocation?

carlocaione, 2023-02-14:
> So if the core problem is section name collision, is the fix to rename the sections generated by code_relocation?

The problem is indeed the name collision between the sections but the renaming is not a fix, because you still would have the conflict between the copying code generated by the `code_relocation` script and the hardcoded snippet in the `init.c.`.

I'm not familiar with DTCM / CCM and why they are treated in a special way, maybe @erwango has a better idea on the why.

But at this point why do you want to keep using the `code_relocation` feature when you can simply put the data you want to be relocated in the `.dtcm_data` section and that will be automatically relocated by the current code?

zagor, 2023-02-14:
> The problem is indeed the name collision between the sections but the renaming is not a fix, because you still would have the conflict between the copying code generated by the code_relocation script and the hardcoded snippet.

Couldn't the relocation code generate a section called `.ccm_bss_relocated` rather than `.ccm_bss`, and put it after `.ccm_bss`?

> But at this point why do you want to keep using the `code_relocation` feature when you can simply put the data you want to be relocated in the `.dtcm_data` section and that will be automatically relocated by the current code?

Because code_relocation offers much more freedom by letting me configure it in CMakeLists.txt rather than in the source code. For example, in one application I have the mbedtls heap relocated to CCM. Doing this in the source would require me to fork and modify mbedtls.

Also, having this control in CMakeLists.txt lets me configure the application memory usage differently for different targets and precisely tune the build for each target's various capabilities.

carlocaione, 2023-02-14:
> Couldn't the relocation code generate a section called .ccm_bss_relocated rather than .ccm_bss, and put it after .ccm_bss?

That should possibly work.

zagor, 2023-05-07:
I'm puzzled. This is a zephyr feature that is shipped broken. I can't be the only person who cares?

carlescufi, 2023-06-20:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/54818

carlescufi, 2023-06-20:
> I'm puzzled. This is a zephyr feature that is shipped broken. I can't be the only person who cares?

You might have been unlucky enough to hit an issue that is not very common, so that's why others did not hit it.
In any case, thank you very much for the report and fix! It is merged now.

closed 2023-06-20 by carlescufi