zephyrproject-rtos/zephyr #54798: net: ipv4: IP packets get dropped in Zephyr when an application is receiving high rate data

cperera-aud, 2023-02-14:
**Describe the bug**
I opened this bug report hoping to get some insight into an issue I'm investigating at the moment. 

Our application receives audio over IPv4 at a high rate which is every IPv4 packet every 333us (16 audio samples which were sampled at 48kHz). The connection to receive the audio is unicast. We have been doing some analysis on the data being received (via glitch tests) and we noticed dropped audio data. This coincides with the  IPv4 drop count increasing as well.  This was observed when running the net stats shell command. 

I've done some investigation into the issue and the IPv4 packet seem to get dropped because a matching connection cannot be found in `net_conn_input` in `subsys/net/ip/connection.c`. When I enable the `NET_DBG("No match found.");` print I see it getting printed.

While trying to understand the code a bit more better, my question is, in some instances a matching connection can be found which leads to receiving the IPv4 packets normally but in some instances a matching connection cannot be found which leads to the IPv4 packet being dropped. Is there any reason why this might be the case? 

Our application uses `zsock_recvfrom` to receive the IPv4 data in a loop. In the case of multicast there are no dropped IPv4 data at all. My understanding is the rules for finding a matching connection for multicast is bit more relaxed compared to unicast? (Sorry if I got it wrong).

carlescufi, 2023-02-14:
Hi @cperera-aud. In order to be able to analyze the issue it will help enormously if you could provide as much info as you can:

- Zephyr version you are using (tag or SHA)
- Local modifications or drivers you might have
- Board you are running on
- Wireshark traces of the packet exchanges if possible
- A simple reproducer app

rlubos, 2023-02-14:
Plus, I miss information on what's actually used on top of IPv4. UDP, TCP? Or do you run packet socket? Generally, please be more specific about conditions for reproducing the issue, I observed no packet drops when running zperf with high throughputs.

cperera-aud, 2023-02-14:
Hi @carlescufi, @rlubos,

I sincerely apologise for the lack of information on the case. Following are some of the details:

* Zephyr version: 3.2
* Board you are running on: Its a custom board based on a STM32H7
* Wireshark traces: Sorry unfortunately there are some parts which are proprietary so I will not be able to include a trace

Currently the issue was seen on our application, unfortunately large parts of it are proprietary. 

The audio over IPv4 is sent over UDP. We don't use packet socket. Thank you for the suggestion regarding zperf, I'll give it a go and match with the data rates typically seen with our application. I'll see if I can reproduce the issue with zperf.

rlubos, 2023-02-16:
Just checked with my `nucleo_h723zg` and zperf sample, and I really see no issue here. The board received packets at 87.76 Mbps rate. I do see quite a few `No match found.` logs, but they're all related to some mcast/bcast packets flying around in my network not the iperf traffic.

cperera-aud, 2023-02-17:
Hi @rlubos,

Thank you for trying out zperf on an STM32H7. I had an opportunity to try zperf with a Nucleo STM32H7 board that I have. I shaped the traffic from iperf running on my to match the audio traffic and I wasn't able to reproduce the dropped packets. At the moment I'm trying to get zperf enabled and get our application to run beside it. And then I'll try to do a zperf/iperf transfer without an audio stream and see what the outcome is like.

carlescufi, 2023-03-01:
@cperera-aud any updates from your testing? Is this a Zephyr issue or an issue with your application?

cperera-aud, 2023-03-03:
Hi @carlescufi, sorry for the late reply from my end. Thank you for following up. I have been doing some testing and see if I can somehow mimic what our application is doing in zperf but unfortunately I'm not able to reproduce the packet drops. 

One thing that makes the problem worse in our application for the packet drops is when low rate (per 1/4 second) data is sent back on the same socket where the data was received. The rationale for this is as you may know, if a unicast receiver stops receiving data (e.g. device crashed), then the ethernet switch connected to the unicast transmitter will start sending the data on all ports of the switch which essentially floods the network. 

I tried doing the same with zperf but it was still able to handle the incoming data without dropping it in the protocol stack. Wireshark showed that the device was indeed sending the packets.

Few other things that our application differs from zperf is non-blocking sockets are used for the receive data. 

A work around that seems to alleviate the problem is if I use a separate socket to send the low rate data. 

At this stage given that I cannot reproduce the issue with zperf, I think it would make sense to close the case. If I can come up with a scenario that can reproduce the issue I'll open a separate case.

closed 2023-03-03 by rlubos