zephyrproject-rtos/zephyr #54804: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property

Niranjan-eaton, 2023-02-14:
**Describe the bug**
Hi im trying to build the Multi_Timers example  nucleo_h743zi board.

Getting below warning:
examples/drivers/Multi_Timers/build/zephyr/zephyr.dts:499.7-528.5: 
Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property

This is my nucleo_h743zi.overlay:

&timers2 {
	st,prescaler = <95>;
	counter {
			compatible = "st,stm32-counter";
			status = "okay";
		};
};

v3/examples/drivers/Multi_Timers/build/zephyr/dts.cmake
D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/build/zephyr/zephyr.dts:499.7-528.5: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property

**To Reproduce**
Steps to reproduce the behavior:
Path : \examples\drivers\Multi_Timers
Command : 
west build -b nucleo_h743zi ./ --pristine

**Expected behavior**
No Warning generator

**Impact**
failing in build server pipe line

**Logs and console output**
[image]

D:\Matrix_repos\LTK-13931_Migration_3_2\edge-rtos-babelfish-v3\examples\drivers\Multi_Timers>west build -b nucleo_h743zi ./ --pristine
-- west build: making build dir D:\Matrix_repos\LTK-13931_Migration_3_2\edge-rtos-babelfish-v3\examples\drivers\Multi_Timers\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers
-- Found Python3: C:/Program Files/Python310/python.exe (found suitable exact version "3.10.0") found components: Interpreter
-- Cache files will be written to: C:/blr_deps/zephyr-v3.2.0/zephyr/.cache
-- Zephyr version: 3.2.0 (C:/blr_deps/zephyr-v3.2.0/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: nucleo_h743zi
-- Found host-tools: zephyr 0.15.2 (C:/Users/40029579/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (C:/Users/40029579/zephyr-sdk-0.15.2)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/blr_deps/zephyr-v3.2.0/zephyr/boards/arm/nucleo_h743zi/nucleo_h743zi.dts
-- Found devicetree overlay: D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/boards/nucleo_h743zi.overlay
-- Generated zephyr.dts: D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/build/zephyr/dts.cmake
D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/build/zephyr/zephyr.dts:499.7-528.5: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property
Parsing D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/Kconfig
Loaded configuration 'C:/blr_deps/zephyr-v3.2.0/zephyr/boards/arm/nucleo_h743zi/nucleo_h743zi_defconfig'
Merged configuration 'D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/prj.conf'
Configuration saved to 'D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/build/zephyr/.config'
Kconfig header saved to 'D:/Matrix_repos/LTK-13931_Migration_3_2/edge-rtos-babelfish-v3/examples/drivers/Multi_Timers/build/zephyr/include/generated/autoconf.h'

**Environment (please complete the following information):**
 - OS zephyr-v3.2.0
 - Zephyr SDK toolchain 0.15.2
 - Commit SHA or Version used

**Additional context**
.overlay file
&timers2 {
	st,prescaler = <95>;
	counter {
			compatible = "st,stm32-counter";
			status = "okay";
		};
};

henrikbrixandersen, 2023-02-14:
Thank you for reporting this. It is, however, a duplicate of #42030. Closing.

laurenmurphyx64, 2023-02-14:
Duplicate of #42030

closed 2023-02-14 by laurenmurphyx64