zephyrproject-rtos/zephyr #54805: when invoke dma_stm32_disable_stream failed in interrupt callback, it will endless loop

Cherish-Gww, 2023-02-14:
In stm32f446rc uart async mode.

code:
https://github.com/zephyrproject-rtos/zephyr/blob/31c11a5dc02a121c659567eeadced7b51a22f3e7/drivers/dma/dma_stm32.c#L264

call stack:
uart_stm32_isr  ---->  uart_stm32_dma_rx_flush ---->  async_evt_rx_rdy ----> async_user_callback  ----> uart_rx_disable

sometimes  stm32_dma_disable_stream failed, the sleep called in interrupt handle, and it will stay in k_sleep.

henrikbrixandersen, 2023-02-14:
Please use the bug template (https://raw.githubusercontent.com/zephyrproject-rtos/zephyr/main/.github/ISSUE_TEMPLATE/bug_report.md) when reporting bugs. You need to fill out the required information there.

FRASTM, 2023-02-20:
Actually, I did not face this situation. However, there are 2 changes with the stm32F4x that could be tested.

1. in the ../modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/stm32f4xx_ll_dma.h, the LL_DMA_IsEnabledStream() could better be written like on other stm32 families   
` return ((READ_BIT(((DMA_Stream_TypeDef *)(dma_base_addr + LL_DMA_STR_OFFSET_TAB[Stream]))->CR, DMA_SxCR_EN) == (DMA_SxCR_EN)) ? 1UL : 0UL);`

2. in the drivers/dma/dma_stm32_v1.c to wait for disabled status directly in the stm32_dma_disable_stream function : 
int stm32_dma_disable_stream(DMA_TypeDef *dma, uint32_t id)
```
{
	LL_DMA_DisableStream(dma, dma_stm32_id_to_stream(id));

	while (stm32_dma_is_enabled_stream(dma, id)) {
	}

	return 0;
}
```

Cherish-Gww, 2023-03-01:
@FRASTM 
we have tested:
item1: no use
item2: run well
item1+2: run well