zephyrproject-rtos/zephyr #54844: RAK5010 board has wrong LIS3DH INT pin configured

joelguittet, 2023-02-14:
**Describe the bug**
The definition of the LIS3DH INT pin on the board RAK5010 is currently wrong. It is configured on P0.15 but instead according to the schematic, the pin should be P0.16

**To Reproduce**
I have not the board myself, this is just reading the code.

**Expected behavior**
Corrdct pin to be used.

**Impact**
Wrong ISR because the pin currently configured is used for something else.

**Logs and console output**
NA

**Environment (please complete the following information):**
- zephyr v3.3.0-rc3

**Additional context**
Schematic of the board: https://downloads.rakwireless.com/Cellular/RAK5010/Hardware-Specification/RAK5010-VB_Schematic.pdf
I have not the board myself so I prefer somebody able to test this making the PR.

nordicjm, 2023-02-15:
@gpaquet85 are you able to comment on this?

joelguittet, 2023-02-15:
To be more accurate, the pin configured as the LIS3DH INT is currently OPT3001 INT (the wrong sensor !)

joelguittet, 2023-02-15:
PR submitted but as indicated above I'm not able to test it because I have not this exact board.

commit 7758ad65 (2023-02-23): board: rak5010_nrf52840: fix lis3dh irq gpio

The INT pin of the LIS3DH sensor is connected to P0.16 on this board.
Fixes #54844.
closed 2023-02-23 by carlescufi