zephyrproject-rtos/zephyr #54861: up_squared: CHRE sample output mangling fails regex verification

LukaszMrugala, 2023-02-15:
**Describe the bug**
The CHRE test requires orderly console output to verify the proper operation of the programme via RegEx. The bug mangles output of six output lines in total, making RegEx not recognise the required lines and timing out the test.

Note this excerpt from the longer logs below:
```
DEBUG   - DEVICE: *** Booting Zephyr OS build v3.3.0-rc3-16-g5786b63e6ca9 ***
DEBUG   - DEVICE: FaiIl:e dC HtRoE  sienti tt,h rveeards inoanm:e  utndefinedo
DEBUG   - DEVICE: I":CHRE "E:v e-n8t8L
DEBUG   - DEVICE: oHoepl lsot aCrHtR
DEBUG   - DEVICE: E!
DEBUG   - DEVICE: D: Instance ID 1 assigned to app ID 0x0000000000000001
DEBUG   - DEVICE: EchoApp::nanoappStart()
DEBUG   - DEVICE: Starting EchoApp... SUCCESS
DEBUG   - DEVICE: Nanoapp count=1
DEBUG   - DEVICE: Finding instance ID... SUCCESS
DEBUG   - DEVICE: Nanoapp count=1
DEBUG   - DEVICE: Instance ID: 1
DEBUG   - DEVICE: Sending event 1...
DEBUG   - DEVICE: EchoApp::nanoappHandleEvent(sender_instance_id=0, event_type=1, event_data@(nil))
DEBUG   - DEVICE: Event (1) complete!
DEBUG   - DEVICE: E: Refusing to unload system nanoapp
DEBUG   - DEVICE: Ending EchoApp... FAIL
DEBUG   - DEVICE: GEocohdobAypep!:
DEBUG   - DEVICE: :nanoappEnd()
DEBUG   - DEVICE: I: Exiting EventLoop
DEBUG   - DEVICE: D: CHRE deinit
DEBUG   - Timed out while monitoring serial output on up_squared
```

After trying to decipher the mangled lines, I've achieved a partial success, where:
```
DEBUG   - DEVICE: FaiIl:e dC HtRoE  sienti tt,h rveeards inoanm:e  utndefinedo
DEBUG   - DEVICE: I":CHRE "E:v e-n8t8L
DEBUG   - DEVICE: oHoepl lsot aCrHtR
DEBUG   - DEVICE: E!
```
Should take on the form (note that `,` is missing here and that the enquoted contents have the lowest confidence):
```
DEBUG   - DEVICE: I: CHRE init
DEBUG   - DEVICE: Failed to set thread name to "undefined version": -88
DEBUG   - DEVICE: I: CHRE Event Loop start
DEBUG   - DEVICE: Hello CHRE!
```
-88 return value means `ENOSYS` error in the `k_set_thread_name` function, that is: trying to set the thread name without setting the option for thread naming. This issue should be orthogonal to the problem here, as additional output lines should not interfere with the RegEx test check.

And this part:
```
DEBUG   - DEVICE: GEocohdobAypep!:
DEBUG   - DEVICE: :nanoappEnd()
```
Should instead take on the form:
```
DEBUG   - DEVICE: Goodbye!
DEBUG   - DEVICE: EchoApp::nanoappEnd()
```

Please also mention any information which could help others to understand
the problem you're facing:
 - Target platform: Two UP Squared boards

**To Reproduce**
Steps to reproduce the behavior:
1. Connect at least one UP Squared board to your controller and map it 
2. Run the CHRE test 
`<path/to>/zephyrproject/zephyr$ ./scripts/twister --device-testing --hardware-map <path/to>/<up2.map> -T samples/modules/chre/ -vv`
3. The test should fail.

**Expected behavior**
The test should pass.

[... 156 lines cut ...]
INFO    - or with west:
INFO    - west build -p -b up_squared -T samples/modules/chre/sample.modules.chre
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

**Environment (please complete the following information):**
 - OS: Ubuntu for the controller
 - Toolchain: Zephyr SDK 15.0
 - Current Zephyr version used

**Additional context**
Multiline RegEx used in the test in question:
```
      regex:
        - "Hello CHRE!"
        - "I: EventLoop start.*"
        - "D: Instance ID 1 assigned to app ID 0x0000000000000001.*"
        - "EchoApp::nanoappStart\\(\\)"
        - "EchoApp::nanoappHandleEvent\\(sender_instance_id=0, event_type=1,
          event_data@(\\(nil\\))|0\\)"
        - "Event \\(1\\) complete!"
        - "EchoApp::nanoappEnd\\(\\)"
        - "I: Exiting EventLoop.*"
```

jhedberg, 2023-11-23:
This is caused by `LOG_MODE_MINIMAL` being used by the sample. The following patch makes it go away:
```patch
--- a/samples/modules/chre/prj.conf
+++ b/samples/modules/chre/prj.conf
@@ -1,6 +1,5 @@
 CONFIG_CHRE=y
 CONFIG_LOG=y
-CONFIG_LOG_MODE_MINIMAL=y
 CONFIG_CHRE_LOG_LEVEL_DBG=y

 CONFIG_CPP=y
```

@fabiobaltieri it seems that you introduced this (with a a seemingly good justification) in commit 80e3d6869128d25244a6cbddca30bad441c84de3

Would using `LOG_MODE_IMMEDIATE` combined with `LOG_IMMEDIATE_CLEAN_OUTPUT` be a possible solution too? `LOG_MODE_MINIMAL` can cause garbled output on SMP platforms, see [the discussion in #55071.](https://github.com/zephyrproject-rtos/zephyr/issues/55071#issuecomment-1823866778)

fabiobaltieri, 2023-11-23:
> Would using `LOG_MODE_IMMEDIATE` combined with `LOG_IMMEDIATE_CLEAN_OUTPUT` be a possible solution too? `LOG_MODE_MINIMAL` can cause garbled output on SMP platforms, see [the discussion in #55071.](https://github.com/zephyrproject-rtos/zephyr/issues/55071#issuecomment-1823866778)

Yeah could be an option, but looking at the sample I'm also thinking that maybe we could shut logging off altogether since the sample code uses `printk` and just check for the printk lines in the output.

jhedberg, 2023-11-24:
> Yeah could be an option, but looking at the sample I'm also thinking that maybe we could shut logging off altogether since the sample code uses `printk` and just check for the printk lines in the output.

That could still cause garbled output, though, couldn't it? I mean, `LOG_MODE_MINIMAL` uses printk itself, and it's the lack of locking in this kind of setup that causes garbled output when multiple CPU cores try to call printk() or LOG_*() at the same time.

fabiobaltieri, 2023-11-24:
> That could still cause garbled output, though, couldn't it? I mean, `LOG_MODE_MINIMAL` uses printk itself, and it's the lack of locking in this kind of setup that causes garbled output when multiple CPU cores try to call printk() or LOG_*() at the same time.

Oh because the printk in `EchoApp` and `main` run concurrently? I was thinking that would get rid of the logs from the subsystem so the sample printk would just go in sequence, but the test itself has concurrency yeah it would not work, let's try with the locking option then.

jhedberg, 2023-11-24:
@fabiobaltieri there's already `PRINTK_SYNC` which *should* be set by default with SMP, but isn't - see the discussion at the end of #55071

jhedberg, 2023-11-24:
Created a fix proposal for this, after understanding the underlying cause a bit better: https://github.com/zephyrproject-rtos/zephyr/pull/65739

PR #65739 (2023-11-27): x86: Only enable EFI_CONSOLE if UART console is not enabled
Most x86 build configurations enable the UART console. Since EFI console has also defaulted to enabled, this means that the EFI covers the early part of the boot until UART takes over.

This is all fine, except that enabling EFI console has the effect of disabling PRINTK_SYNC. This in turn has the effect of causing garbled output over UART, which has led to several bug reports on x86 platforms (in particular on up_squared).

Since EFI console should really only be used for early platform bringup and debugging purposes, it's not really [... cut]
commit 084b8e4f (2023-11-27): x86: Only enable EFI_CONSOLE if UART console is not enabled

Most x86 build configurations enable the UART console. Since EFI console
has also defaulted to enabled, this means that the EFI covers the early
part of the boot until UART takes over.

This is all fine, except that enabling EFI console has the effect of
disabling PRINTK_SYNC. This in turn has the effect of causing garbled
output over UART, which has led to several bug reports on x86 platforms
(in particular on up_squared).

Since EFI console should really only be used for early platform bringup
and debugging purposes, it's not reall [... cut]
closed 2023-11-27 by fabiobaltieri