zephyrproject-rtos/zephyr #54865: Blobs: introduce EFR32BG22 BLE/RAIL binary blobs

fkokosinski, 2023-02-15:
## Origin

Three binary blobs are to be introduced:
* libbluetooth_controller_efr32bg22_gcc_release.a
* libbgcommon_efr32xg22_gcc_release.a
* librail_efr32xg22_gcc_release.a

The origin of these binary blobs is the Silicon Labs Gecko SDK repository: https://github.com/SiliconLabs/gecko_sdk.

## Type

- [x] Precompiled library
- [ ] Firmware image

## Module

hal_silabs: https://github.com/zephyrproject-rtos/hal_silabs/.

## Purpose

Precompiled static libraries (binary blobs) supporting EFR32 RF subsystems in EFR32BG22-based boards.

## Pull Request

Zephyr PR: https://github.com/zephyrproject-rtos/zephyr/pull/54853.
HAL PR: https://github.com/zephyrproject-rtos/hal_silabs/pull/21.

## Dependencies

* hal_silabs (Zephyr module): https://github.com/zephyrproject-rtos/hal_silabs.
* mbedtls (Zephyr module): https://github.com/zephyrproject-rtos/mbedtls.

## License

* libbluetooth_controller_efr32bg22_gcc_release.a - MSLA
* libbgcommon_efr32xg22_gcc_release.a - MSLA
* librail_efr32xg22_gcc_release.a - Zlib

MSLA (Silicon Labs Master Software License Agreement) license text will be located in hal_silabs repository under the following path: `zephyr/blobs/license/MSLA.txt`.

Zlib license text will be located in hal_silabs under the following path: `zephyr/blobs/license/Zlib.txt`.

fkokosinski, 2023-03-30:
According to the public [TSC meeting notes](https://docs.google.com/document/d/1cMkXkFN-Yje-7zIGZmWMfkrVCju85UNnKDfjng9vEv0/edit#heading=h.k1pop18i6dhp), the TSC has approved adding binary blobs listed in this issue.

nashif, 2023-04-05:
approved by the TSC