zephyrproject-rtos/zephyr #54886: Bluetooth: controller: llcp: compile controller for different Kconfig settings

kruithofa, 2023-02-16:
**Is your enhancement proposal related to a problem? Please describe.**
There are many different Kconfig settings for the bluetooth controller, and it is impossible to manually keep track of the different permutations. There is a script (tests/bluetooth/bsim/_compile_permutate_kconfigs.sh) which compiles the controller for a wide range of Kconfig settings, but it is not run as part of CI.
Runtime for this script is approx. 2 1/2 hours
There are also the projects in tests/bluetooth/init, which compile for a range of Kconfig settings, but not as exhaustive as the script

**Describe the solution you'd like**
The script should run as part of CI, and probably needs updating for recently added Kconfig settings.
Alternatively the projects in tests/bluetooth/init needs a review (and updating)

**Describe alternatives you've considered**
See above for 2 possible solutions

**Additional context**
N/A

kruithofa, 2023-04-27:
This script should be run as part of a weekly run. @kruithofa to investigate how this can be done. @erbr-ot Please advise if more KConfig options need to be considered.

cvinayak, 2024-02-13:
Not planned in near future.

closed 2024-02-13 by cvinayak