zephyrproject-rtos/zephyr #54893: Bluetooth: list of abbreviations

jori-nordic, 2023-02-16:
We desperately need a table of all the abbreviations used in the Bluetooth stack and what they map to.

This will help newcomers to Bluetooth (and Zephyr) to understand what's happening when reading the code, and hopefully result in more external contributions.

hermabe, 2023-02-16:
The core spec has an overview over abbreviations in Vol 1, Part B that might be of help. We should also try to use those in the stack if we can.

nashif, 2025-06-19:
@jhedberg  can you please take a look at this or assign it to someone else, or close it?

jhedberg, 2025-06-27:
> [@jhedberg](https://github.com/jhedberg) can you please take a look at this or assign it to someone else, or close it?

I'm unable to commit to doing this, and haven't found any other volunteers either, so I'll close it.