zephyrproject-rtos/zephyr #54906: NXP MIMXRT685 I2S Trigger Error

harshadesai, 2023-02-16:
Hi,
Zephyr SDK: v2.6.0

Problem: Not able to play generated sine wave over I2S to amplifier on the MIMXRT685 board.
Getting the below mentioned error when I call I2S_Trigger API.

**Error: [00:06:08.760,144] <err> i2s_mcux_flexcomm: No buffer in input queue to start transmission
           [00:06:08.760,178] <err> i2s_mcux_flexcomm: START trigger failed -35**

henrikbrixandersen, 2023-02-17:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

nashif, 2023-02-21:
please open a new bug with more details using the bug template. There isn't enough information here for us to make any progress.

closed 2023-02-21 by nashif