zephyrproject-rtos/zephyr #54921: drivers: sensors: bmi270 does not wait for data ready bit resulting in many failed i2c transactions

maxmclau, 2023-02-17:
**Describe the bug**
BMI270 driver does not currently support GPIO interrupt driven data ready triggering so you are forced to constantly poll the device at whatever rate it should be producing data. This causes many i2c transactions to fail, especially when sampling at higher data rates.

**Expected behavior**
Drivers should not request data from BMI270 device unless data is ready.

**Impact**
Annoyance when sole device on bus. Showstopper when i2c bus is shared with many peripherals. As noted in [#54917](https://github.com/zephyrproject-rtos/zephyr/issues/54917) this can cause unrecoverable bus arbitration errors.

**Additional context**
I'm currently working on a PR to fix this. Will update when ready.

tautologyclub, 2023-02-28:
See https://github.com/zephyrproject-rtos/zephyr/pull/55275

maxmclau, 2023-04-12:
Closed with https://github.com/zephyrproject-rtos/zephyr/pull/55275

closed 2023-04-12 by maxmclau