zephyrproject-rtos/zephyr #54925: samples: doc: Several samples are missing README.rst file

kartben, 2023-02-17:
Filing this as an enhancement, but feel free to flag as a bug. Several samples in the `samples/` folder are missing a README.
Per our "[Sample definition and criteria](https://docs.zephyrproject.org/latest/samples/sample_definition_and_criteria.html#samples-must-be-documented)":

> Samples must have a README.rst file in the samples folder. Example: samples/subsys/foo/README.rst. clearly explaining the purpose of the sample, its HW requirements, and the expected sample output, if applicable.

See the list below.
Note that there's a few false positives, for samples that are more "modular", but even those probably need to be reviewed as some have some form of README, but not one that's properly mounted in the table of contents on the website (ex. recently added sample for Thrift support, poke @cfriedt). 

```bash
 ~#@❯  find . -name prj.conf -execdir [ ! -e README.rst ] \; -printf %h\\n
./bluetooth/mtu_update/central
./bluetooth/mtu_update/peripheral
./boards/esp32/ethernet
./boards/nrf/nrf53_sync_rtc/net
./drivers/ipm/ipm_esp32/ipm_esp32_net
./drivers/ipm/ipm_mcux/remote
./drivers/lcd_cyclonev_socdk
./drivers/led_apa102c_bitbang
./drivers/lora/receive
./drivers/lora/send
./drivers/mbox/remote
./drivers/spi_fujitsu_fram
./drivers/watchdog
./modules/thrift/hello/client
./modules/thrift/hello/server
./posix/eventfd
./sensor/bmg160
./sensor/fxos8700-hid
./sensor/grove_light
./sensor/grove_temperature
./sensor/max44009
./sensor/sx9500
./sensor/thermometer
./shields/npm6001_ek
./subsys/console/echo
./subsys/fs/format
./subsys/ipc/ipc_service/icmsg/remote
./subsys/ipc/ipc_service/icmsg_me
./subsys/ipc/ipc_service/icmsg_me/remote
./subsys/ipc/ipc_service/static_vrings/remote
./subsys/ipc/openamp/remote
./subsys/ipc/rpmsg_service/remote
./subsys/logging/multidomain
./subsys/logging/multidomain/remote
./subsys/mgmt/osdp/control_panel
./subsys/mgmt/osdp/peripheral_device
./subsys/pm/device_pm
./subsys/pm/latency
./subsys/rtio/sensor_batch_processing
./subsys/shell/devmem_load
./subsys/shell/shell_module
./subsys/testsuite/integration
./subsys/testsuite/pytest
./subsys/tracing
```

Note that some of these samples actually have README or README.txt file so fixing these should be trivial.

```bash
 ~#@❯  find . -name prj.conf -execdir [ -e README ] \; -printf %h\\n
./sensor/max44009
```

```bash
~#@❯  find . -name prj.conf -execdir [ -e README.txt ] \; -printf %h\\n
./sensor/sx9500
./sensor/thermometer
./subsys/tracing
```

Do folks have thoughts/suggestions on how to best proceed? I can work on a PR addressing some of the trivial fixes from the list.
BTW there are some really old samples that should probably be removed altogether, ex. "thermometer" which is IMO essentialy useless/redundant now: https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/sensor/thermometer

henrikbrixandersen, 2023-02-17:
Possible duplicate of #27805.

kartben, 2023-02-17:
> Possible duplicate of #27805.

Damn, and I did search for "readme samples". I am going to close this one and make sure to update the list on the other bug. Thank you!

closed 2023-02-17 by kartben