zephyrproject-rtos/zephyr #54926: tests/kernel/timer/timer_behavior fails on the stm32l073rz nucleo board

FRASTM, 2023-02-17:
**Describe the bug**
Running the tests/kernel/timer/timer_behavior/ on the nucleo_l073rz  board, the test fails on
```
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/tick_timer_train.c:143: 
timer_tick_train_test_one_tick_timer_train: (busy_loops > acceptable_busy_loops is false)
Expected thread to run while 1 tick timers are firing
```

This seems related to the commit e60af7926848dcb7b86f48ce2f2d23c85d0f1e59

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_l073rz tests/kernel/timer/timer_behavior/
2. west flash
3. See error

**Expected behavior**
Test PASSED

**Impact**
CI failure  on this target board (ther stm32 platforms pass the test)

**Logs and console output**
```
*** Booting Zephyr OS build v3.3.0-rc2-75-ge60af7926848 ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift
periodic timer behavior test
periodic timer samples gathered, calculating statistics
timer clock rate 32000000, kernel tick rate 10000, expected period 1000 us, samples 1000, period 32000.000000 cycles, expected period drift 0.000000 us
statistics samples 1000, rollovers 0, mean 1000.075812 us, variance 5.737051 us, stddev 2.395214 us, min 1000.000000 us, max 1075.781250 us
statistics samples 1000, rollovers 0, mean 32002.426000 cycles, variance 5874.740524 cycles, stddev 76.646858 cycles, min 32000 cycles, max 34425 cycles
timer start cycle 630247, end cycle 32632673, total time 1000075.812500, expected time 1000000.000000, expected time drift 0.000000, difference 75.812500
 PASS - test_jitter_drift in 1.210 seconds
===================================================================
TESTSUITE timer_jitter_drift succeeded
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 10000Hz, Expecting 25000 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 3
Faking busy work, remaining timers is 3, timer callbacks 25000
One Tick Timer Train Done, took 10255 ms, busy loop ran 2 times
Timer 0 max delta 13670 cycles or 427 us
Timer 1 max delta 13670 cycles or 427 us
Timer 2 max delta 15168 cycles or 474 us
Timer 3 max delta 13817 cycles or 432 us

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/tick_timer_train.c:143: timer_tick_train_test_one_tick_timer_train: (busy_loops > acceptable_busy_loops is false)
Expected thread to run while 1 tick timers are firing
 FAIL - test_one_tick_timer_train in 10.307 seconds
===================================================================
TESTSUITE timer_tick_train failed.

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [timer_jitter_drift]: pass = 1, fail = 0, skip = 0, total = 1 duration = 1.210 seconds
 - PASS - [timer_jitter_drift.test_jitter_drift] duration = 1.210 seconds

SUITE FAIL -   0.00% [timer_tick_train]: pass = 0, fail = 1, skip = 0, total = 1 duration = 10.307 seconds
 - FAIL - [timer_tick_train.test_one_tick_timer_train] duration = 10.307 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain Zephyr SDK 0.15.0
 - Commit v3.3.0-rc3

FRASTM, 2023-02-17:
In the kernel/timer.c, instead returning immediately on the if (sys_dnode_is_linked(&t->node)), one proposal could : 
```
diff --git a/kernel/timer.c b/kernel/timer.c
index 325224d11b..ad0bdfd50d 100644
--- a/kernel/timer.c
+++ b/kernel/timer.c
@@ -40,8 +40,7 @@ void z_timer_expiration_handler(struct _timeout *t)
 	 * so the function exits immediately.
 	 */
 	if (sys_dnode_is_linked(&t->node)) {
-		k_spin_unlock(&lock, key);
-		return;
+		goto end_thread;
 	}
 
 	/*
@@ -87,6 +86,7 @@ void z_timer_expiration_handler(struct _timeout *t)
 		return;
 	}
 
+end_thread:
 	thread = z_waitq_head(&timer->wait_q);
 
 	if (thread == NULL) {
```

nashif, 2023-02-21:
@teburd  any idea?

andyross, 2023-02-21:
I'm guessing from context that this used to pass but has since regressed due to that commit?  That was to handle a race in k_timer where it was possible for high priority contexts (i.e. higher than the timer interrupt!) to cancel/re-queue a timer between the moment it expired and the entry to its callback, and the solution there was to detect that case and abort (i.e. treat it as having been cancelled).

The suggested patch isn't correct, that will just run whatever the next the next thread waiting on the timer happens to be, which may have nothing to do with the state that caused the wakeup.

I'm at a loss for why this test would be sensitive to this.  It's not running a higher priority interrupt, right?  It should never hit that sys_dnode_is_linked() case.

FRASTM, 2023-02-24:
closing as the https://github.com/zephyrproject-rtos/zephyr/pull/54918 fixed this issue

closed 2023-02-24 by FRASTM