zephyrproject-rtos/zephyr #54932: stm32 ADC init stage before adc_stm32_calib need disable ADC

Cherish-Gww, 2023-02-17:
**Describe the bug**
In stm32 adc init stage befor adc_stm32_calib need disable adc, here is the code:
https://github.com/zephyrproject-rtos/zephyr/blob/0aa4e38bbe806be684992d6782745da3109c9e06/drivers/adc/adc_stm32.c#L1219

 - STM32G0B1RC
 - bootloader use ADC then bootup application it will loop wait calibration done.

**Additional context**
In RM0555 Rev 5 pdf page 347:
Software calibration procedure
1.Ensure that ADEN = 0, ADVREGEN = 1 and DMAEN = 0.
2.Set ADCAL = 1.
3.Wait until ADCAL = 0 (or until EOCAL = 1). This can be handled by interrupt if the
interrupt is enabled by setting the EOCALIE bit in the ADC_IER register
4.The calibration factor can be read from bits 6:0 of ADC_DR or ADC_CALFACT
registers.

and in function start_read at adc_stm32.c
https://github.com/zephyrproject-rtos/zephyr/blob/0aa4e38bbe806be684992d6782745da3109c9e06/drivers/adc/adc_stm32.c#L865
declared can not calibrate the ADC while the ADC is enable, so in stm32 adc init befor adc_stm32_calib, it need disable ADC before adc_stm32_calib function.

FRASTM, 2023-02-22:
In the adc_stm32_init() function, when adc_stm32_calib() is called, the ADC is not yet enabled but still disabled. Not sure that it will change the behavior : 
```
diff --git a/drivers/adc/adc_stm32.c b/drivers/adc/adc_stm32.c
index deda8a40d9..047f9b360c 100644
--- a/drivers/adc/adc_stm32.c
+++ b/drivers/adc/adc_stm32.c
@@ -1216,6 +1216,7 @@ static int adc_stm32_init(const struct device *dev)
 	 * Calibration of F1 and F3 (ADC1_V2_5) series has to be started
 	 * after ADC Module is enabled.
 	 */
+	adc_stm32_disable(adc);
 	adc_stm32_calib(dev);
 #endif
```

erwango, 2023-02-23:
@Cherish-Gww Can you provide some code to reproduce easily this issue ?

erwango, 2023-02-23:
@Cherish-Gww is that similar to the following description ?
> Hi, I have a project which consists of a bootloader and an application (both Zephyr applications), my target is an STM32H7 (stm32h735g_disco board). I've noticed that if the ADC is enabled for both the bootloader and application (CONFIG_ADC=y and CONFIG_ADC_ASYNC=y) then the bootloader runs fine but after it jumps to the application the ADC driver fails to initialise - it gets stuck trying to calibrate (gets stuck in while (LL_ADC_IsCalibrationOnGoing(adc))). Disabling the ADC in the bootloader causes it to initialise successfully in the application. Any ideas why calibration would get stuck in this situation?

Cherish-Gww, 2023-02-28:
> @Cherish-Gww is that similar to the following description ?

Yes

closed 2023-03-08 by carlescufi